espressif/esp-idf #13468: %lld cannot be displayed correctly on esp32c2 (XTAL=26MHz)

nopnop2002, 2024-03-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-210-gcf7b9bed4c

### Espressif SoC revision.

esp32c2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32c2 development board

### Power Supply used.

USB

### What is the expected behavior?

%lld can be displayed correctly on esp32c2 (XTAL=26MHz)

### What is the actual behavior?

%lld cannot be displayed correctly on esp32c2 (XTAL=26MHz)

### Steps to reproduce.

Build this code to esp32c2(26MHz XTAL)

```
#include <stdio.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_timer.h"
#include "rom/ets_sys.h" // ets_get_cpu_frequency()

static const char *TAG = "MAIN";

void app_main()
{
#ifdef CONFIG_IDF_TARGET_ESP32
    ESP_LOGI(TAG, "Core is ESP32@%"PRIu32"Mhz", ets_get_cpu_frequency());
#elif defined CONFIG_IDF_TARGET_ESP32S2
    ESP_LOGI(TAG, "Core is ESP32S2@%"PRIu32"Mhz", ets_get_cpu_frequency());
#elif defined CONFIG_IDF_TARGET_ESP32S3
    ESP_LOGI(TAG, "Core is ESP32S3@%"PRIu32"Mhz", ets_get_cpu_frequency());
#elif defined CONFIG_IDF_TARGET_ESP32C2
    ESP_LOGI(TAG, "Core is ESP32C2@%"PRIu32"Mhz", ets_get_cpu_frequency());
#elif defined CONFIG_IDF_TARGET_ESP32C3
    ESP_LOGI(TAG, "Core is ESP32C3@%"PRIu32"Mhz", ets_get_cpu_frequency());
#elif defined CONFIG_IDF_TARGET_ESP32C6
    ESP_LOGI(TAG, "Core is ESP32C6@%"PRIu32"Mhz", ets_get_cpu_frequency());
#elif defined CONFIG_IDF_TARGET_ESP32H2
    ESP_LOGI(TAG, "Core is ESP32H2@%"PRIu32"Mhz", ets_get_cpu_frequency());
#endif
    int64_t time = esp_timer_get_time();
    ESP_LOGI(TAG, "time=%lld us", time);
}
```

### Debug Logs.

```plain
I (315) MAIN: Core is ESP32@160Mhz
I (315) MAIN: time=34226 us

I (283) MAIN: Core is ESP32S2@160Mhz
I (293) MAIN: time=802396 us

I (305) MAIN: Core is ESP32S3@160Mhz
I (305) MAIN: time=307432 us

I (386) MAIN: Core is ESP32C2@120Mhz(26MHz XTAL)
I (386) MAIN: time=ld us

I (282) MAIN: Core is ESP32C3@160Mhz
I (292) MAIN: time=295127 us

I (283) MAIN: Core is ESP32C6@160Mhz
I (293) MAIN: time=295725 us

I (283) MAIN: Core is ESP32H2@96Mhz
I (293) MAIN: time=296210 us
```

igrr, 2024-03-25:
You probably have CONFIG_NEWLIB_NANO_FORMAT enabled (as it is enabled by default on C2). Please try disabling, then 64-bit formatting should work.

nopnop2002, 2024-03-25:
Thank you for your reply.
I changed it and confirmed it works fine.
[image]
```
I (386) MAIN: Core is ESP32C2@120Mhz
I (386) MAIN: time=396460 us
```

I would like it to be disabled by default in C2 as well.

closed 2024-03-25 by nopnop2002