espressif/esp-idf #13469: Idf v5.2 Zigbee buffer full

kallibr44, 2024-03-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-76-gbf17be96b4

### Espressif SoC revision.

can't find (command gives SyntaxError: Invalid or unexpected token)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32-c6 WIFI6 RE: 1.0 (small board)

### Power Supply used.

USB

### What is the expected behavior?

I'm waiting for a working example of zigbee light

### What is the actual behavior?

i'm tried to run HA_ON_OFF_LIGHT example on version v5.2, but after attempting to connect i got following result in logs and device can't complete connection. 
I need to run this on version 5.2 in order to use the USB console in another project

### Steps to reproduce.

1. Install v5.2 release branch of ESP-IDF
2. Take example HA_on_off_light
3. build & flash esp32-c6
4. attempt to pair with coordinator

### Debug Logs.

```plain
> I (466) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (466) ESP_ZB_ON_OFF_LIGHT: Start network steering
receive buffer full, drop the current frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current ack frame.
receive buffer full, drop the current ack frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current ack frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current ack frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current ack frame.
I (9306) ESP_ZB_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
receive buffer full, drop the current ack frame.
receive buffer full, drop the current ack frame.
receive buffer full, drop the current ack frame.
I (9346) ESP_ZB_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```

kallibr44, 2024-03-26:
upd. I found what i have this problem if i use this parameter in config
[image]

xieqinan, 2024-04-15:
@kallibr44 ,

Noted it, I will try to reproduce it following your steps and provide a response as soon as possible.

Layochi, 2024-04-25:
I'm also having trouble to connect my custom code to my Home Assistant. I get those messages :
```
receive buffer full, drop the current ack frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current ack frame.
receive buffer full, drop the current frame.
receive buffer full, drop the current ack frame.
```

But the example work for me. It's strange. Could you add a more specific error message ?

xieqinan, 2024-04-26:
@Layochi , @kallibr44 ,

Based on recent findings regarding this issue, it appears that the problem might be caused by the complex network environment and insufficient running time allocated to the Zigbee task. Could you provide details on the number of devices operating on the network in your project? Additionally, are there other tasks running alongside the Zigbee task? If so, are their priorities higher than the Zigbee task, and do they consume a significant amount of running time, such as by outputting extensive logs?

Layochi, 2024-04-26:
It’s strange I do not have the bug anymore…

chshu, 2024-11-13:
@kallibr44 @Layochi Please reopen if any further questions.

closed 2024-11-13 by chshu