espressif/esp-idf #13470: esp_console: user pointer for commands

mickeyl, 2024-03-25:
### Is your feature request related to a problem?

I'm doing a lot in C++. It's kind of annoying when I have to resort to static functions only to be C-ABI-compatible when registering a new command for `esp_console`.

### Describe the solution you'd like.

I'd like to have a user pointer in the `esp_console_cmd_t` struct. That way, I could register a new command given a lambda function or a closure.

### Describe alternatives you've considered.

Static C functions are no alternative.

### Additional context.

I'd consider writing this, but I'd like to know what you're thinking about it and whether you would accept a pull request implementing this.

KonstantinKondrashov, 2024-03-27:
Hi @mickeyl!
I would like to ask about the implication first. What is the `user pointer` in the `esp_console_cmd_t` struct?
Could you add a bit more info about your implementation and an example of using it?
Thanks.

igrr, 2024-03-27:
And another question, whether the existing context member of this structure is usable for this purpose:
https://github.com/espressif/esp-idf/blob/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60/components/console/esp_console.h#L213

mickeyl, 2024-03-27:
Ok, this is embarassing. I swear neither this `context` nor the `esp_console_cmd_func_with_context_t` was available in this struct last time I looked. This is _exactly_ what I need. Thanks and sorry for the noise!

closed 2024-03-27 by mickeyl

igrr, 2024-03-27:
No problem, it was a recent addition from https://github.com/espressif/esp-idf/pull/12436, so you are probably right that it wasn't available before!