espressif/esp-idf #13471: Question/Bug about esp_gap_ble_set_channels() in BLE 4.2

meakashrao, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32-WROOM-32E-N8 (revision v3.0 and V3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

I was expecting that after the correct implementation I'd be bale to update the BLE Channel Map for data communication

### What is the actual behavior?

Even after receiving a success back from the API the Channel Map is not updated

### Steps to reproduce.

I normally upload the code to the ESP Module and connect the ESP to Tablet running the Android App and then pass the command to update the Channel Map(only 28,29,30,31 to be active)

This contains Two Sniffer Data packets

a) LL_CHANNEL_MAP_IND when I pass the command to update the channel
b) LL_UNKNOWN_RSP, response that is passed afterwards. 

[BLE_Channel_Selection_Report.txt](https://github.com/espressif/esp-idf/files/14755292/BLE_Channel_Selection_Report.txt)

meakashrao, 2024-03-26:
So, a bit more reading gave me that there should be an LL_CHANNEL_MAP_REQ from the central and then the peripheral will perform this

I have an ESP Running as a client and whenever the Tablet send a BLE Command I was trying to update the channel Map and only use 28,29,30,31 as my data communication channels and the error received was 
Expert Info (Error/Protocol): Incorrect control procedure packet sequencing or direction

[image]

So, I got to know that a Central should initiate the channel, but how?

Please some help.

meakashrao, 2024-03-26:
@xiewenxiang Please can you guide me with the Implementation

meakashrao, 2024-03-28:
I understood that, 
It should be up to the device requesting a BLE connection to request a channel map update task.
[image]

or ### **according to the Flow Chart**
[image]

I am asking for help as to how do I emulate it or how does the android app(on the phone) send LL_CHANNEL_MAP_IND PDU? 
Please let me know what Command the Android App needs to send and how.

meakashrao, 2024-04-01:
Hello, any update/questions?

zhp0406, 2024-04-01:
@meakashrao 
Thank you for reporting the issue. I will check it.

meakashrao, 2024-04-02:
Hello, I tried to initiate the channel map from the peripheral, I have an android app which acts as the master. How do I make sure that Android App/ESP32 Master initiates a BLE Channel Map update and The ESP Peripheral acts on it. 

The API  esp_err_t result = esp_gap_ble_set_channels(channels)

wont be helfull from the ESP32 acting as the Master end. Please suggest something on this

zhp0406, 2024-04-03:
@meakashrao 
hi
Can you provide me with the captured packet files shown in your pictures?

zhp0406, 2024-04-03:
The channel update is initiated by the master side, and the slave cannot initiate it. Based on your description, the ESP32 acts as the slave, and the mobile phone side is the master. You need to call the mobile phone's API to set the channels instead of the slave side initiating the channel update.

zhp0406, 2024-04-03:
The mobile application should provide an API for channel updates, but the specific name of the API is unknown to me. 
In the IDF, the API is named esp_gap_ble_set_channels. You can refer to the description of this API in the BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 4, Part E, page 2382 for more details.
[image]

meakashrao, 2024-04-03:
Ok, thank you for helping me with this API description. 

So, assuming I find the API for mobile, 
a) The mobile will trigger the API which does similar to what has been given above by you (The ESP that I use is BLE 4.2, I will try to find the corresponding API)
b) Then the esp will do esp_gap_ble_set_channels
and it will work?

Can you help me if I can test the same using two ESP's one acting as a Master and the other acting as a slave? How do I do that?

zhp0406, 2024-04-03:
The API esp_gap_ble_set_channels is only valid when called as a master device.

meakashrao, 2024-04-03:
So, even if I find the correct API from the Mobile App which acts as a master, my slave ESP wont update it, or will update it without me doing anything on the ESP side?

zhp0406, 2024-04-03:
Two ESP32 devices, one acting as a master and the other as a slave, is ok.

You can refer to the following examples in esp-idf/examples/bluetooth/bluedroid/ble/gatt_server_service_table and esp-idf/examples/bluetooth/bluedroid/ble/gatt_client.

In esp-idf/examples/bluetooth/bluedroid/ble/gatt_client, the device will initiate the connection as a master, while in esp-idf/examples/bluetooth/bluedroid/ble/gatt_server_service_table, it will act as a slave.

If you want to initiate channel updates, you can do so in esp-idf/examples/bluetooth/bluedroid/ble/gatt_server_service_table after a successful connection is established.

For example, you can call esp_gap_ble_set_channels in ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT.

The parameter "channels" specifies the channels you want to use, with the following description:

/**
* @brief            BLE set channels
*
* @param[in]        channels :   The n th such field (in the range 0 to 36) contains the value for the link layer channel index n.
*                                0 means channel n is bad.
*                                1 means channel n is unknown.
*                                The most significant bits are reserved and shall be set to 0.
*                                At least one channel shall be marked as unknown.
*
* @return           - ESP_OK : success
*                   - ESP_ERR_INVALID_STATE: if bluetooth stack is not yet enabled
*                   - other  : failed
*
*/
esp_err_t esp_gap_ble_set_channels(esp_gap_ble_channels channels);

meakashrao, 2024-04-03:
Thank you for this, I just want one clarification, I am using BLE 4.2 Module, the above API you suggested is for BLE 5.4, so is this possible for BLE 4.2 or not?

zhp0406, 2024-04-03:
Your understanding is correct. Upon receiving the LL_CHANNEL_MAP_IND from the master, the slave automatically updates without requiring any further action from you.

zhp0406, 2024-04-03:
It is also ok in BLE 4.2.

zhp0406, 2024-04-03:
This issue can be closed for now. If you encounter any new problems while using it, you can raise an new issue, and we will respond to you as soon as possible.

meakashrao, 2024-04-03:
OK. Thank you for your guidance.

meakashrao, 2024-04-03:
Hey, I am sorry to reopen this. 

I tried a lot but I am not able to find an Android way to send command that was mentioned above or any HCI command to be honest? Is there some other way?

meakashrao, 2024-04-03:
@zhp0406 sorry for re-opening this.

closed 2024-04-03 by meakashrao