espressif/esp-idf #13472: Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

GregoryCle, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v1.23.0-preview-273-g35b2edfc2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have a circuit with an ESP32C3. This works well with already compiled Micropython code. 

I can't find the reason. I think it comes from my IDF and settings.

### What is the actual behavior?

`Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038061c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xf28
load:0x403cc710,len:0x89c
load:0x403ce710,len:0x2b14
entry 0x403cc710
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4004c48c  RA      : 0x4004c4b0  SP      : 0x3fcde2b0  GP      : 0x3fc97400
TP      : 0x00000000  T0      : 0x4005890e  T1      : 0x00000000  T2      : 0x00000000
S0/FP   : 0x000000ff  S1      : 0x00000050  A0      : 0x000000ff  A1      : 0x3fcde2e0
A2      : 0x0012ad40  A3      : 0x03ff0000  A4      : 0x0000e000  A5      : 0x60ffe01c
A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x0000000c  S3      : 0x3fcde434
S4      : 0x42000000  S5      : 0x3c150000  S6      : 0xffff0000  S7      : 0x3fcd6000
S8      : 0x00050020  S9      : 0x00010020  S10     : 0x00000006  S11     : 0x3fcde3cc
T3      : 0x60008000  T4      : 0x60008000  T5      : 0xffff01ff  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x60ffe01c
MHARTID : 0x00000000

Stack memory:
3fcde2b0: 0x0000000c 0x00000050 0x00000001 0x420007e2 0x400391e6 0x00000000 0x00000000 0x60008000
3fcde2d0: 0x00000001 0x000001e0 0x00000006 0x00000050 0x00000001 0x000001e0 0x00000003 0x000000a0
3fcde2f0: 0x0000000c 0x421430b3 0x00000000 0x40380366 0x4038025c 0x421430b3 0x00000001 0x403cf714
3fcde310: 0x66296ba1 0x3a8457bf 0x30470726 0x00143094 0x3f5cb405 0xc8504df6 0xccf84e19 0x00150000
3fcde330: 0x00000000 0x00000000 0xffffffff 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcdf000
3fcde350: 0x00004308 0x3ff1c14c 0x0000ffff 0xffffffff 0xffffffff 0x3fcde4b0 0xffffffff 0x403cfaf8
3fcde370: 0x80821055 0x80824511 0x00010000 0x001f0000 0x00010000 0x2f0206e9 0x4038025c 0x000000ee
3fcde390: 0x03030005 0x0000c700 0x01000000 0x3c150020 0x0003c3f8 0x3fc96c00 0x00002fdc 0x40380000
3fcde3b0: 0x00000c14 0x42000020 0x00143094 0x40380c14 0x00015fb4 0x50000000 0x00000018 0x00000000
3fcde3d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde410: 0x00000000 0x00000000 0x00000000 0x00010020 0x0004c420 0x0004f404 0x00050020 0x001930bc
3fcde430: 0x001a9078 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde450: 0x00000000 0x00000000 0x00000000 0x001990c0 0x66296ba1 0x3a8457bf 0x30470726 0x769d0c42
[... 1277 lines cut ...]
    "VFS_SELECT_IN_RAM": true,
    "VFS_SEMIHOSTFS_MAX_MOUNT_POINTS": 1,
    "VFS_SUPPORT_DIR": true,
    "VFS_SUPPORT_IO": true,
    "VFS_SUPPORT_SELECT": true,
    "VFS_SUPPORT_TERMIOS": true,
    "VFS_SUPPRESS_SELECT_DEBUG_OUTPUT": true,
    "WIFI_PROV_AUTOSTOP_TIMEOUT": 30,
    "WIFI_PROV_BLE_BONDING": false,
    "WIFI_PROV_BLE_FORCE_ENCRYPTION": false,
    "WIFI_PROV_BLE_SEC_CONN": true,
    "WIFI_PROV_KEEP_BLE_ON_AFTER_PROV": false,
    "WIFI_PROV_SCAN_MAX_ENTRIES": 16,
    "WIFI_PROV_STA_ALL_CHANNEL_SCAN": true,
    "WIFI_PROV_STA_FAST_SCAN": false,
    "WL_SECTOR_SIZE": 4096,
    "WL_SECTOR_SIZE_4096": true,
    "WL_SECTOR_SIZE_512": false,
    "WS_BUFFER_SIZE": 1024,
    "WS_DYNAMIC_BUFFER": false,
    "WS_TRANSPORT": true,
    "XTAL_FREQ": 40,
    "XTAL_FREQ_40": true
}
```

KonstantinKondrashov, 2024-03-26:
Hi @GregoryCle!
Sorry, but you left the description and log that say nothing. Probably you need to raise an issue in the Micropython project.

Regarding the log, you need to use the automatic address decoding that is integrated into the idf monitor tool. It will help you to get the Backtrace. How to use it? just run `idf.py monitor`.
Some useful links that can help in debugging:
https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/api-guides/tools/idf-monitor.html#automatic-address-decoding
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/core_dump.html#automatic-decoding

KonstantinKondrashov, 2024-03-27:
@GregoryCle 
It is usually printed along with a backtrace, at the very end. Maybe you just cut it or that project did not use `--elf-sha256-offset` option for esptool to generate the final bin.

> So there is no error with the configuration that could cause this error?

I can not say it because it is too big. If you did not manually edit it it should be ok. The Menuconfig makes the correct settings. 
Try to decode the backtrace it will give you a place in the code where it happens.

closed 2024-04-02 by KonstantinKondrashov