espressif/esp-idf #13474: MQTT does not connect to emqx broker

AlexMihai1804, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to connect mqtt to emqx serverless broker. this is my code
```
const esp_mqtt_client_config_t mqtt_cfg = {
            .broker = {
                    .address = {
                            .uri = "mqtts://b0axxxxx.ala.eu-central-1.emqxsl.com",
                            .port=8883
                    },
                    .verification = {
                            .certificate = cert,
                    },
            },
            .credentials = {
                    .username = "xxxxx",
                    .authentication = {
                            .password = "xxxx",
                    },
            },
    };
```
```
const char cert[]="-----BEGIN CERTIFICATE-----\n"
                  "MIIDrzCCApegAwIBAgIQCDvgVpBCRrGhdWrJWZHHSjANBgkqhkiG9w0BAQUFADBh\n"
                  "MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3\n"
                  "d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD\n"
                  "QTAeFw0wNjExMTAwMDAwMDBaFw0zMTExMTAwMDAwMDBaMGExCzAJBgNVBAYTAlVT\n"
                  "MRUwEwYDVQQKEwxEaWdpQ2xxxCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j\n"
                  "b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IENBMIIBIjANBgkqhkiG\n"
                  "9w0BAQEFAAOCAQ8AMIIBCgKCAQEA4jvhEXLeqKTTo1eqUKKPC3eQyaKl7hLOllsB\n"
                  "CSDMAZOnTjC3U/dDxGkAV53ijSLdhwZAAIEJzs4bg7/fzTtxRuLWZscFs3YnFo97\n"
                  "nh6Vfe63SKMI2tavegw5BmV/Sl0xxxf4q77uKNd0f3p4mVmFaG5cIzJLv07A6Fpt\n"
                  "43C/dxC//AH2hdmoRBBYMql1GNXRor5H4idq9Joz+EkIYIvUX7Q6hL+hqkpMfT7P\n"
                  "T19sdl6gSzeRntwi5m3OFBqOasv+zbMUZBfHWymeMr/y7vrTC0LUq7dBMtoM1O/4\n"
                  "gdW7jVg/tRvoSSiicNoxBN33shbyTApOB6jtSj1etX+jkMOvJwIDAQABo2MwYTAO\n"
                  "BgNVHQ8BAf8EBAMCAYYwDwYDVR0TAQH/BAUwAwEB/zAdBgNVHQ4EFgQUA95QNVbR\n"
                  "TLtm8KPiGxvDl7I90VUwHwYDVR0jBBgwFoAUA95QNVbRTLtm8KPiGxvDl7I90VUw\n"
                  "DQYJKoZIhvcNAQEFBQADggEBAMucN6pIExIK+t1EnE9SsPTfrgT1eXkIoyQY/Esr\n"
                  "hMAtudXH/vTBH1jLuG2cenTnmCmrEbXjcKChzUyImZOMkXDiqw8cvpOp/2PV5Adg\n"
                  "06O/nVsJ8dWO41P0jmP6P6fbtGbfYmbW0W5BjfIttep3Sp+dWOIrWcBAI+0tKIJF\n"
                  "PnlUkiaY4IBIqDfv8NZ5YBberOgOzW6sRBc4L0na4UU+Krk2U886UAb3LujEV0ls\n"
                  "YSEY1QSteDwsOoBrp+uvFRTp2InBuThs4pFsiv9kuXclVzDAGySj4dzp30d8tbQk\n"
                  "CAUw7C29C79Fv1C5qfPrmAESrciIxpg0X40KPMbp1ZWVbd4=\n"
                  "-----END CERTIFICATE-----\n";
``` (I modified some characters to be safe)
When I put mqtts:// I get this error
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x4037d446  PS      : 0x00060c33  A0      : 0x80381658  A1      : 0x3fcaf610
A2      : 0x23e04c05  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x3c135a94
A6      : 0x00000000  A7      : 0x00000338  A8      : 0x80386318  A9      : 0x3fcaf710
A10     : 0x3fc9a030  A11     : 0x00000640  A12     : 0x00000000  A13     : 0x40376b4c
A14     : 0x00000007  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000009
EXCVADDR: 0x23e04c05  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x4037d443:0x3fcaf610 0x40381655:0x3fcaf630 0x40383801:0x3fcaf660 0x403774c7:0x3fcaf680 0x40377510:0x3fcaf6b0 0x40385375:0x3fcaf6d0 0x40376b51:0x3fcaf6f0 0x420f3a11:0x3fcaf710 0x40386315:0x3fcaf740 0x40386c8c:0x3
fcaf770 0x40387115:0x3fcaf7b0 0x40041502:0x3fcaf7f0 |<-CORRUPTED

ELF file SHA256: e9a804231a4f9e08

And if I remove the mqtts:// It won't crash but it doesn't connect
 mqtt_client: Error parse uri = 
Any ideas how to fix this?

AlexMihai1804, 2024-03-31:
I've made some modifications but I'm still having issues. I embedded the certificate and set the size for the certificate; without it, it causes a core panic. Now, I'm receiving an error related to the certificate. This is my current code. Do you have any suggestions on why it might not be working or help for debugging?
```
#include "esp_log.h"
#include "mqtt_client.h"
#include "nvs_flash.h"

static const char *TAG = "MQTT_EXAMPLE";
esp_mqtt_client_handle_t client;
extern const uint8_t emqxsl_ca_crt_start[] asm("_binary_emqxsl_ca_pem_start");
extern const uint8_t emqxsl_ca_crt_end[] asm("_binary_emqxsl_ca_pem_end");
const size_t emqxsl_ca_crt_size = emqxsl_ca_crt_end - emqxsl_ca_crt_start;

static esp_err_t mqtt_event_handler_cb(esp_mqtt_event_handle_t event) {
    switch (event->event_id) {
        case MQTT_EVENT_CONNECTED:
            ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
            esp_mqtt_client_publish(client, "test", "This is a test message", 0, 1, 0);
            break;
        case MQTT_EVENT_DISCONNECTED:
            ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
            break;
        default:
            break;
    }
    return ESP_OK;
}

static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data) {
    esp_mqtt_event_handle_t event = (esp_mqtt_event_handle_t) event_data;
    mqtt_event_handler_cb(event);
}

void mqtt_app_start(void) {
    const esp_mqtt_client_config_t mqtt_cfg = {
            .broker = {
                    .address = {
                            .uri = "mqtts://b0a3ea3c.ala.eu-central-1.emqxsl.com:8883",
                    },
                    .verification={
                            .certificate=(const char *) emqxsl_ca_crt_start,
                            .certificate_len=emqxsl_ca_crt_size
                    }
            },
            .credentials={
                    .username="test",
                    .authentication={
                            .password="123"
                    }
            }
    };
    client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_register_event(client, MQTT_EVENT_ANY, mqtt_event_handler, client);
    esp_mqtt_client_start(client);
}
```
now i get this 
```
â�›[0;31mE (2543498) esp-tls-mbedtls: mbedtls_x509_crt_parse of CA cert returned -0x2180â�›[0m
â�›[0;31mE (2543498) esp-tls-mbedtls: Failed to set client configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)â�›[0m
â�›[0;31mE (2543498) esp-tls: create_ssl_handle failedâ�›[0m
â�›[0;31mE (2543508) esp-tls: Failed to open new connectionâ�›[0m
â�›[0;31mE (2543508) transport_base: Failed to open a new connectionâ�›[0m
â�›[0;31mE (2543518) mqtt_client: Error transport connectâ�›[0m
â�›[0;32mI (2543528) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTEDâ�›[0m
```

euripedesrocha, 2024-04-04:
@AlexMihai1804 there is an issue parsing your certificate.
I suggest that you take the pem format file in the ssl example and change the contents to be of your instance in emqx.

euripedesrocha, 2024-05-23:
Closing this one, feel free to reopen, adding more information to help identify potential issues in the library.

closed 2024-05-23 by euripedesrocha