espressif/esp-idf #13475: OpenThread joiner not finding commissioner

luisyague, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### **General issue report**
**Issue Summary:**
Hi, we are initiating the development of a production-ready Thread Border Router device using an ESP32-C6 (ESP32-C6-DevKitC-1-N8) and a second device which uses an ESP32-h2 (also using a devkit) with the esp-idf version release/v5.2. In our testing phase, we encountered several issues, with the primary having inconsistent results when starting the joiner, we are also getting some warnings: 
[image]

**ESP32-H2 Behaviour**
I will paste the whole code bellow, as you will see I have also the possibility to use the CLI. The problem is that it is sometimes finding the commissioner, but other times it doesnt. These are the codes both for the joiner and the comissioner

**Result**
```
I (460) app_start: Starting scheduler on CPU0
I (464) main_task: Started on CPU0
I (464) main_task: Calling app_main()
I (474) OPe��I (494) phy: phy_version: 200,0, 1cef4f4, May 22 2023, 11:57:13
I(504) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I(554) OPENTHREAD:[I] Settings------: Read NetworkInfo {rloc:0x7000, extaddr:aeee34d5b6af6736, role:leader, mode:0x0f, version:4, keyseq:0x0, ...
I(564) OPENTHREAD:[I] Settings------: ... pid:0x4ffde8f, mlecntr:0x3ef, maccntr:0x3e8, mliid:65d209060d1a9a2a}
I (574) ot_esp_node: joiner starting joiner
I (574) ot_esp_node: After joiner start
I (554) main_task: Returned from app_main()
I(5714) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:56, chksum:0ee1, ecn:no, to:None, sec:no, error:Abort, prio:net
I(5714) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:88f:5749:6a42:ad9c]:19788
I(5724) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
I (5734) ot_esp_node: Join failed: no network discovered
```

**Source code for the joiner**
```
static void handleJoinerCompleted(otError aError, void* aContext) {
  switch (aError) {
    case OT_ERROR_NONE:
      ESP_LOGI(TAG, "Join succeeded");
      break;
    case OT_ERROR_SECURITY:
      ESP_LOGI(TAG, "Join failed: pskd does not match");
      break;
    case OT_ERROR_NOT_FOUND:
      ESP_LOGI(TAG, "Join failed: no network discovered");
      break;
    case OT_ERROR_RESPONSE_TIMEOUT:
      ESP_LOGI(TAG, "Join failed: timed out");
      break;
    default:
      break;
  }
}
static void otLoopTask(void *aContext) {
  otInstance* instance = esp_openthread_get_instance();

  // Inconfig up
  esp_openthread_lock_acquire(portMAX_DELAY);
  if (otIp6SetEnabled(instance, true) != OT_ERROR_NONE) {
    ESP_LOGE(TAG, "Failed to enable OpenThread IP6 link");
    abort();
  }
  esp_openthread_lock_release();

  ESP_LOGI(TAG, "joiner starting joiner");
  esp_openthread_lock_acquire(portMAX_DELAY);
  otJoinerStart(instance, "J01NME", NULL, NULL, NULL, NULL , NULL, handleJoinerCompleted, (void*) instance);
  ESP_LOGI(TAG, "After joiner start");
  esp_openthread_lock_release();
[... 350 lines cut ...]
  ESP_ERROR_CHECK(mdns_init());
  ESP_ERROR_CHECK(mdns_hostname_set("esp-ot-br"));
}

void app_main(void)
{
  eventfdInit();
  flashInit();

  ESP_ERROR_CHECK(esp_netif_init());
  ESP_ERROR_CHECK(esp_event_loop_create_default());

  ESP_ERROR_CHECK(example_connect());
  ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));
  ESP_ERROR_CHECK(esp_coex_wifi_i154_enable());
  esp_openthread_set_backbone_netif(get_example_netif());
  // esp_ot_wifi_netif_init();
  // esp_openthread_set_backbone_netif(esp_netif_get_handle_from_ifkey("WIFI_STA_DEF"));

  mdnsInit();
  xTaskCreate(otTaskWorker, "ot_br_main", 20480, xTaskGetCurrentTaskHandle(), 5, NULL);
}
```

Any guess of which can be the problem? Thank you very much in advance.

zwx1995esp, 2024-04-18:
> Hello, I am also encountering the same error while using ESP-IDF v5.0.6. My board is ESP32 with the ESP32H2beta2 module.(example: esp-idf/examples/openthread/ot_br)
> 
> > joiner start J01NME
> > Done
> > I(62864) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:56, chksum:f53b, ecn:no, to:None, sec:no, error:Abort, prio:net
> > I(62864) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:e0f6:b9ce:29d5:ccc1]:19788
> > I(62874) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
> > Join failed [Security]
> > W(79404) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > W(108914) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > W(147234) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > W(174134) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > .
> > .
> > .

So did you use the command line or use your own logic? If you use the command line to join a thread network, you can refer to these step:
https://openthread.io/codelabs/esp-openthread-hardware#5

Hinalikp, 2024-04-22:
> So did you use the command line or use your own logic? If you use the command line to join a thread network, you can refer to these step: https://openthread.io/codelabs/esp-openthread-hardware#5

I am using command line and tried same command lines which you refereed. But still I am getting:

> dataset pskc 1f8a6515a237085a3c24a388969538fe
Done
> dataset panid 0x70c2
Done
> dataset commit active
Done
> ifconfig up
Done
I (103034) OPENTHREAD: netif up
> eui64
1097bdf0c5c3fcff
Done
> ifconfig up
Done
> joiner start J01NME
Done
> I(175674) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:56, chksum:5515, ecn:no, to:None, sec:no, error:Abort, prio:net
I(175674) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:d86e:c5b0:f587:9ddb]:19788
I(175684) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
Join failed [Security]

I am using (esp32s3(ot_br)+esp32h2(ot_rcp) ) as thread border router. And as a joiner I am using esp32h2beta2(ot_cli).

zwx1995esp, 2024-04-22:
> > > Hello, I am also encountering the same error while using ESP-IDF v5.0.6. My board is ESP32 with the ESP32H2beta2 module.(example: esp-idf/examples/openthread/ot_br)
> > > > joiner start J01NME
> > > > Done
> > > > I(62864) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:56, chksum:f53b, ecn:no, to:None, sec:no, error:Abort, prio:net
> > > > I(62864) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:e0f6:b9ce:29d5:ccc1]:19788
> > > > I(62874) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
> > > > Join failed [Security]
> > > > W(79404) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > > > W(108914) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > > > W(147234) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > > > W(174134) OPENTHREAD:[W] Mle-----------: Failed to process UDP: InvalidState
> > > > .
> > > > .
> > > > .
> > 
> > 
> > So did you use the command line or use your own logic? If you use the command line to join a thread network, you can refer to these step: https://openthread.io/codelabs/esp-openthread-hardware#5
> 
> I am using command line and tried same command lines which you recommend . But still I am getting:
> 
> > dataset pskc 1f8a6515a237085a3c24a388969538fe
> > Done
> > dataset panid 0x70c2
> > Done
> > dataset commit active
> > Done
> > ifconfig up
> > Done
> > I (103034) OPENTHREAD: netif up
> > eui64
> > 1097bdf0c5c3fcff
> > Done
> > ifconfig up
> > Done
> > joiner start J01NME
> > Done
> > I(175674) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:56, chksum:5515, ecn:no, to:None, sec:no, error:Abort, prio:net
> > I(175674) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:d86e:c5b0:f587:9ddb]:19788
> > I(175684) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
> > Join failed [Security]
> 
> I am using (esp32s3(ot_br)+esp32h2(ot_rcp) ) as thread border router. And as a joiner I am using esp32h2beta2(ot_cli).

So, did you use the latest code? Could you please share the branch you used? Maybe you can have a try with
```
esp-idf: v5.1.3
esp-thread-br: v1.0
```
If you still encounter this issue, please share both the BR and H2 logs.

Hinalikp, 2024-04-22:
Actually for now, i only have esp32h2beta2 device and esp-idf v5.0 only supports it.  These are the logs:
[image]

[image]

[image]

[image]

[image]

[image]

zwx1995esp, 2024-04-22:
Hi， @Hinalikp , so bad to hear that you only had an esp32h2beta2, this was a preview version and we stopped to support it when IDF v5.1 released...

Hinalikp, 2024-04-22:
Okay, Thank you for reply.

zwx1995esp, 2024-04-22:
> Okay, Thank you for reply.

Maybe you can get a MP version of ESP32H2, and play it with the latest IDF. :)

zwx1995esp, 2024-04-22:
Hi @luisyague Sorry for the delay, for your logs on joiner:
[image]
From the logs, we can see there is a previous dataset configured in the boards, you need to erase the NVS flash before you want to commission the device.

Second, from your code, I did not see you set the same panid and pskc with the commissioner. In the commissioning process, the joiner and the commissioner should have the same panid and pskc. Panid is used to join the same network, the pskc is used to create the DTLS handshake.

Could you please check and then have a try?

luisyague, 2024-04-22:
Hi @zwx1995esp, thanks for the reply. They should be the same because they are both configured on the sdkconfig but I will double checkit tomorrow and give you a reply. Thanks!!!

chshu, 2024-07-17:
@luisyague suppose the issue has been resoved by your side, feel free to reopen if you have any follow up questions.

closed 2024-07-17 by chshu