espressif/esp-idf #13476: How to debug code in OpenOCD on the device esp32-c3-devkitm-1

edbek, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Can anyone tell me: is it possible to debug code in OpenOCD on the device esp32-c3-devkitm-1 ?
Or is it impossible to do? Why then produce such a product?

igrr, 2024-03-26:
You can use OpenOCD with ESP32-C3-DevKitm-1, to do this you need to either:

- Connect a JTAG adapter to the JTAG pins of the chip and burn the efuse as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-other-jtag.html). Any FT2232H-based adapter, such as ESP-Prog will work fine.
- Connect a USB cable to GPIOs 18 and 19 as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html) to use the built-in USB_SERIAL_JTAG adapter

> Why then produce such a product?

The question is probably a bit out of scope of ESP-IDF issue tracker, but I can say that this was taken into account for newer chips. ESP32-C6 and ESP32-H2 devboards have the 2nd USB port connected to the built-in USB_SERIAL_JTAG peripheral, so that you only need to plug in a USB cable to debug.

rms43, 2024-08-06:
If you are using Linux as OS, in my case Debian, the GDB debugger worked for me with OpenOCD after I created a udev permission file in /etc/udev/rules.d/70-openocd.rules containing the following parameters.

ACTION!="add|change", GOTO="openocd_rules_end"
SUBSYSTEM=="gpio", MODE="0660", GROUP="plugdev", TAG+="uaccess"
SUBSYSTEM!="usb|tty|hidraw", GOTO="openocd_rules_end"

# Espressif USB JTAG/serial debug units
ATTRS{idVendor}=="303a", ATTRS{idProduct}=="1001", MODE="660", GROUP="plugdev", TAG+="uaccess"
ATTRS{idVendor}=="303a", ATTRS{idProduct}=="1002", MODE="660", GROUP="plugdev", TAG+="uaccess"