espressif/esp-idf #13478: ESP32-S3 Custom SPI Pins

radurotaru19, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone,

I am currently porting my software written in the Arduino framework to the ESP-IDF V5.2.1 (tried V5.1 as well) and I am facing an issue with the SPI setup.
On my custom hardware I have connected to the board an ADC MCP3204 from Microchip which communicates over SPI with the ESP32-S3-WROOM1 module, pinout:

GPIO39 SCK
GPIO40 MISO 
GPIO41 MOSI 
GPIO42 CS Analog to digital converter

I originally wrote my whole code for it using the Arduino Framework & PlatformIO which works absolutely fine, however the SPI is not working as it should on the ESP-IDF code.

I did find a pre-made library for the MCP3204 at: https://github.com/gfurtadoalmeida/esp32-driver-mcp320x. I put an oscilloscope to the SPI pins and I get the expected output on the MOSI line. However, the MISO line seems to not register, it is reading only LOW. When I did play with gpio_reset_pin() and gpio_set_direction(), I made the MISO line to read only HI. 

Question: Is there a good gpio reset procedure in the ESP-IDF that I may be able to do to properly configure my custom pins? In the Arduino framework the function pinMode() does a lot of things in the background. 

Could someone try to replicate the issue with any SPI device connected to the custom pins above? 

CODE FOR REFERENCE:
`#include "driver/spi_master.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "esp32_driver_mcp320x/mcp320x.h"

void app_main(void)
{
    spi_bus_config_t bus_cfg = {
        .mosi_io_num = GPIO_NUM_41,
        .miso_io_num = GPIO_NUM_40,
        .sclk_io_num = GPIO_NUM_39,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .data4_io_num = -1,
        .data5_io_num = -1,
        .data6_io_num = -1,
        .data7_io_num = -1,
        .max_transfer_sz = 3, // 24 bits.
        .flags = SPICOMMON_BUSFLAG_MASTER,
        .isr_cpu_id = INTR_CPU_ID_AUTO,
        .intr_flags = ESP_INTR_FLAG_LEVEL3};

    mcp320x_config_t mcp320x_cfg = {
        .host = SPI3_HOST,
        .device_model = MCP3204_MODEL,
        .clock_speed_hz = 1 * 1000 * 1000, // 1 Mhz.
        .reference_voltage = 5000,         // 5V
        .cs_io_num = GPIO_NUM_42};

    // Bus initialization is up to the developer.
    spi_bus_initialize(mcp320x_cfg.host, &bus_cfg, 0);

    // Add the device to the SPI bus.
    mcp320x_t *mcp320x_handle = mcp320x_install(&mcp320x_cfg);

    // Occupy the SPI bus for multiple transactions.
    mcp320x_acquire(mcp320x_handle, portMAX_DELAY);

    uint16_t voltage = 0;

    for (size_t i = 0; i < 10; i++)
    {
        // Read voltage, sampling 1000 times.
        mcp320x_read_voltage(mcp320x_handle,
                             MCP320X_CHANNEL_0,
                             MCP320X_READ_MODE_SINGLE,
                             1000,
                             &voltage);

        ESP_LOGI("mcp320x", "Voltage: %d mV", voltage);
    }

    // Unoccupy the SPI bus.
    mcp320x_release(mcp320x_handle);

    // Free resources.
    mcp320x_delete(mcp320x_handle);
}`

radurotaru19, 2024-03-26:
Just a side note from my testing. I flash the unit and the first few readings from the ADC are right then it just drops suddenly. It is highly repeatable.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914
I (27) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (27) boot: compile time Mar 26 2024 20:40:30
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0ed44h ( 60740) map
I (110) esp_image: segment 1: paddr=0001ed6c vaddr=3fc94200 size=012ach (  4780) load
I (111) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=1c240h (115264) map
I (137) esp_image: segment 3: paddr=0003c268 vaddr=3fc954ac size=01718h (  5912) load
I (138) esp_image: segment 4: paddr=0003d988 vaddr=40374000 size=101a8h ( 65960) load
I (163) boot: Loaded app from partition at offset 0x10000
I (164) boot: Disabling RNG early entropy source...
I (175) cpu_start: Multicore app
I (185) cpu_start: Pro cpu start user code
I (185) cpu_start: cpu freq: 160000000 Hz
I (185) cpu_start: Application information:
I (188) cpu_start: Project name:     SPITest
I (193) cpu_start: App version:      1
I (197) cpu_start: Compile time:     Mar 26 2024 20:40:08
I (203) cpu_start: ELF file SHA256:  c6a5fd5c0...
I (208) cpu_start: ESP-IDF:          v5.2.1
I (213) cpu_start: Min chip rev:     v0.0
I (218) cpu_start: Max chip rev:     v0.99
I (223) cpu_start: Chip rev:         v0.1
I (228) heap_init: Initializing. RAM available for dynamic allocation:
I (235) heap_init: At 3FC974B0 len 00052260 (328 KiB): RAM
I (241) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (247) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (253) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (260) spi_flash: detected chip: gd
I (264) spi_flash: flash io: dio
W (268) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (281) sleep: Configure to isolate all GPIO pins in sleep state
I (288) sleep: Enable automatic switching of GPIO sleep configuration
I (295) main_task: Started on CPU0
I (305) main_task: Calling app_main()
I (345) mcp320x: Voltage: 688 mV
I (1375) mcp320x: Voltage: 688 mV
I (2405) mcp320x: Voltage: 688 mV
I (3435) mcp320x: Voltage: 0 mV
I (4465) mcp320x: Voltage: 0 mV
I (5495) mcp320x: Voltage: 0 mV
I (6525) mcp320x: Voltage: 0 mV
I (7555) mcp320x: Voltage: 0 mV

vismay5559, 2025-12-19:
i am also facing the same issue in making multiple sensors work on a custom spi bus using esp32 s3, did you solve the problem