espressif/esp-idf #13479: CPU slows down 18 times approx, and the issue survives watchdog actuation

txinto, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.3

### Espressif SoC revision.

Chip is ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

WDT does not cause it rebooting, the execution flow does not slow down nearly x18.

### What is the actual behavior?

After a lot of hours working, with bluetooth active, AWS_OTA, the controller can start rebooting continuously.  The symptoms are "0xGuru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). "

After rebooting, the system keeps rebooting every 9 seconds approximately.

Power cycling the unit recovers.

Comparing a normal boot sequence with the boot sequence of the series of continouous reboots, we can see that from a given point the execution slow down, being 18 times slower approximately.  As a result of that ralentization the watchdog is triggered again.

I think this screen capture is very representative.

[image]

If you look carefully at it, the watchdog did not "clean" the issue, because the dramatic slowdown takes place before app_main() is called, so the issue survives a "soft reset", and it is only cleared by the hard reset.  So the issue is not why it occurs the first time (it is important but it is not the most important one), the real issue is how the silicon can enter in such unwanted status and survive the watchdog action.

### Steps to reproduce.

The first time it can occur randomly after a lot of hours, in our app.  After it occured the first time, you have not to do anything, it will be reproduced every 9 seconds.

I think here the analysis should not be driven by reproducing the issue, but to think what register can be causing this slowdown and which can be unaffected by watchdog resets.  And also to think if this behavior has been seen before.

txinto, 2024-03-26:
I calculated the slow down as 18x by simply dividing the time delta of a representative fast-good boot step with the equivalent boot step in the slow-bad boot sequence.
For instance, in the screen capture, delta in lines 54-55 took 5 milliseconds on the right side, and 104 milliseconds in the left side.   Taking that step as the reference, the slow boot sequence is being 20,8 times slower than the normal one.

KonstantinKondrashov, 2024-03-28:
Hi @txinto!
We are analyzing this issue. Could you show a log related to repeat reboot (rebooting every 9 seconds approximately)?

txinto, 2024-03-28:
`[0;32mI (97[0;31mE (9811) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[0;31mE (9811) task_wdt:  - IDLE (CPU 0)[0m
[0;31mE (9811) task_wdt: Tasks currently running:[0m
[0;31mE (9811) task_wdt: CPU 0: nimble_host[0m
[0;31mE (9811) task_wdt: Print CPU 0 (current core) registers[0m
Core  4 register dump:
MEPC    : 0x4200919e  RA      : 0x4200941e  SP      : 0x40857f90  GP      : 0x4081be20  
TP      : 0x40806428  T0      : 0x40028140  T1     Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
MEPC    : 0x42000f10  RA      : 0x42000f96  SP      : 0x4081f820  GP      : 0x4081be20  
TP      : 0x40806428  T0      : 0x4002289a  T1      : 0xffffffe0  T2      : 0x00000002  
S0/FP   : 0x00000020  S1      : 0x42150afc  A0      : 0x00000020  A1      : 0x4081f81f  
A2      : 0x00000020  A3      : 0x00000000  A4      : 0x6000f000  A5      : 0x0000005b  
A6      : 0x00000001  A7      : 0x0000000a  S2      : 0x00000004  S3      : 0x40857f10  
S4      : 0x00002653  S5      : 0x40858160  S6      : 0x4216f01d  S7      : 0x00000000  
S8      : 0x0000002b  S9      : 0x00000000  SESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x42000f0c
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2e6c
load:0x40875728,len:0x17d8
entry 0x4086c410
[0;32mI (22) boot: ESP-IDF v5.1.3-dirty 2nd stage bootloader[0m
[0;32mI (23) boot: compile time Mar 23 2024 17:40:51[0m
[0;32mI (24) boot: chip revision: v0.0[0m
[0;32mI (26) boot.esp32c6: SPI Speed      : 80MHz[0m
[0;32mI (31) boot.esp32c6: SPI Mode       : DIO[0m
[0;32mI (36) boot.esp32c6: SPI Flash Size : 4MB[0m
[0;33mW (40) boot.esp32c6: PRO CPU has been reset by WDT.[0m
[0;32mI (46) boot: Enabling RNG early entropy source...[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (55) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (84) boot:  3 ota_0            OTA app          00 10 00010000 001e0000[0m
[0;32mI (92) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000[0m
[0;32mI (99) boot:  5 storage          WiFi data        01 02 003d0000 00004000[0m
[0;32mI (107) boot:  6 spiffs_storage   Unknown data     01 82 003d4000 0002c000[0m
[0;32mI (115) boot: End of partition table[0m
[0;32mI (119) esp_image: segment 0: paddr=001f0020 vaddr=42150020 size=52a00h (338432) map[0m
[0;32mI (194) esp_image: segment 1: paddr=00242a28 vaddr=40800000 size=0d5f0h ( 54768) load[0m
[0;32mI (207) esp_image: segment 2: paddr=00250020 vaddr=42000020 size=14d31ch (1364764) map[0m
[0;32mI (476) esp_image: segment 3: paddr=0039d344 vaddr=4080d5f0 size=0e028h ( 57384) load[0m
[0;32mI (490) esp_image: segment 4: paddr=003ab374 vaddr=4081b620 size=03964h ( 14692) load[0m
[0;32mI (501) boot: Loaded app from partition at offset 0x1f0000[0m
[0;32mI (501) boot: Disabling RNG early entropy source...[0m
[0;32mI (513) cpu_start: Unicore app[0m
[0;32mI (513) cpu_start: Pro cpu up.[0m
[0;33mW (522) clk: esp_perip_clk_init() has not been implemented yet[0m
[0;32mI (529) cpu_start: Pro cpu start user code[0m
[0;32mI (529) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (529) cpu_start: Application information:[0m
[0;32mI (532) cpu_start: Project name:     tanknetwifi[0m
[0;32mI (537) cpu_start: App version:      7.3.0.9-diag[0m
[0;32mI (542) cpu_start: Compile time:     Mar 23 2024 22:22:04[0m
[0;32mI (548) cpu_start: ELF file SHA256:  8cfe0fd331cbd443...[0m
[0;32mI (554) cpu_start: ESP-IDF:          v5.1.3-dirty[0m
[0;32mI (560) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (564) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (569) cpu_start: Chip rev:         v0.0[0m
[0;32mI (574) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (581) heap_init: At 4083B460 len 000411B0 (260 KiB): D/IRAM[0m
[0;32mI (588) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM[0m
[0;32mI (594) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM[0m
[0;32mI (602) spi_flash: detected chip: generic[0m
[... 3085 lines cut ...]
W (8211) wifi:(TB)WDEV_PWR_TB_MCS2:19
W (8271) wifi:(TB)WDEV_PWR_TB_MCS3:19
W (8331) wifi:(TB)WDEV_PWR_TB_MCS4:19
W (8401) wifi:(TB)WDEV_PWR_TB_MCS5:19
W (8461) wifi:(TB)WDEV_PWR_TB_MCS6:18
W (8531) wifi:(TB)WDEV_PWR_TB_MCS7:18
W (8591) wifi:(TB)WDEV_PWR_TB_MCS8:17
W (8651) wifi:(TB)WDEV_PWR_TB_MCS9:15
W (8721) wifi:(TB)WDEV_PWR_TB_MCS10:15
W (8781) wifi:(TB)WDEV_PWR_TB_MCS11:15
I (8851) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).

I (8981) wifi:mode : sta (40:4c:ca:44:e8:e0)
I (9061) wifi:enable tsf
[0;32mI (9101) provision_app: Event handler WIFI_EVENT 2[0m
[0;32mI (9211) BLE_INIT: Using main XTAL as clock source[0m
[0;32mI (9331) BLE_INIT: ble controller commit:[217f1bf][0m
[0;32mI (9441) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26[0m
[0;32mI (9571) protocomm_nimble: BLE Host Task Started[0m
[0;32mI (9681) NimBLE: GAP procedure initiated: stop advertising.
[0m
[0;32mI (98[0;31mE (9821) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[0;31mE (9821) task_wdt:  - IDLE (CPU 0)[0m
[0;31mE (9821) task_wdt: Tasks currently running:[0m
`

KonstantinKondrashov, 2024-03-28:
```
MEPC : 0x4200919e RA : 0x4200941e SP : 0x40857fb0 GP : 0x4081be20
TP : 0x40806448 T0 : 0x40028140 T1 Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0).

Core 0 register dump:
MEPC : 0x42000f10 RA : 0x42000f96 SP : 0x4081f820 GP : 0x4081be20
TP : 0x40806448 T0 : 0x4002289a T1 : 0xffffffe0 T2 : 0x00000002
S0/FP : 0x00000020 S1 : 0x42150afc A0 : 0x00000020 A1 : 0x4081f81f
A2 : 0x00000020 A3 : 0x00000000 A4 : 0x6000f000 A5 : 0x0000005b
A6 : 0x00000001 A7 : 0x0000000a S2 : 0x00000004 S3 : 0x40857f30
S4 : 0x0000265d S5 : 0x40858180 S6 : 0x4216f01d S7 : 0x00000000
S8 : 0x0000002b S9 : 0x00000000 SESP-ROM:esp32c6-20220919
```
If you used the idf monitor tool then these addresses convert to a the source code location and function name - 
https://docs.espressif.com/projects/esp-idf/en/release-v3.2/get-started/idf-monitor.html#automatically-decoding-addresses
But if you can not do it now. You  can run the cmd below to decode it manually:

```xtensa-esp32-elf-addr2line -piaf -e build/*.elf 0x42000f10 0x42000f96 ...```

txinto, 2024-03-28:
Ok, I'll do it, which ones should I decode?

At the moment this is what I've got:

```
riscv32-esp-elf-addr2line -piaf -e tanknetwifi.elf 0x42000f10 0x42000f96 
0x42000f10: usb_serial_jtag_ll_write_txfifo at /home/txinto/esp/v5.1.3/esp-idf/components/hal/esp32c6/include/hal/usb_serial_jtag_ll.h:135
 (inlined by) panic_print_char_usb_serial_jtag at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:114
0x42000f96: panic_print_char at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:132
```

KonstantinKondrashov, 2024-03-29:
Could you get for these addresses too `MEPC : 0x4200919e RA : 0x4200941e`?

txinto, 2024-03-29:
```
riscv32-esp-elf-addr2line -piaf -e tanknetwifi.elf 0x4200919e 0x4200941e
0x4200919e: usb_serial_jtag_ll_write_txfifo at /home/txinto/esp/v5.1.3/esp-idf/components/hal/esp32c6/include/hal/usb_serial_jtag_ll.h:135
 (inlined by) usb_serial_jtag_tx_char at /home/txinto/esp/v5.1.3/esp-idf/components/vfs/vfs_usb_serial_jtag.c:122
0x4200941e: usb_serial_jtag_write at /home/txinto/esp/v5.1.3/esp-idf/components/vfs/vfs_usb_serial_jtag.c:157
```

txinto, 2024-03-29:
By the way, I am seeing that the reboot from reset counter 53 to reset counter 54 has different addresses:

```
�[0;31mE (6731) task_wdt: Print CPU 0 (current core) registers�[0m
Core –4 register dump:
MEPC    : 0x4080fbd0  RA      : 0x4080fbc6  SP      : 0x40847420  GP      : 0x4081be20  
TP      : 0x407f53b8  T0      : 0x40030dca  T1      : 0x0Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
MEPC    : 0x42000f10  RA      : 0x42000f96  SP      : 0x4081f820  GP      : 0x4081be20  
TP      : 0x407f53b8  T0      : 0x4002289a  T1      : 0xffffffe0  T2      : 0x00000002  
S0/FP   : 0x00000030  S1      : 0x00000001  A0      : 0x00000030  A1      : 0x4081f81f  
A2      : 0x00000030  A3      : 0x00000000  A4      : 0x6000f000  A5      : 0x0000005b  
A6      : 0x00000001  A7      : 0x0000000a  S2      : 0x00000004  S3      : 0x408473a0  
S4      : 0x00001a4b  S5      : 0x40839c9c  S6      : 0x00000000  S7      : 0x00000000  
```

```
riscv32-esp-elf-addr2line -piaf -e tanknetwifi.elf 0x4080fbd0 0x4080fbc6
0x4080fbd0: vPortYield at /home/txinto/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:404 (discriminator 1)
0x4080fbc6: vPortYield at /home/txinto/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:404 (discriminator 3)
```

Regards.

txinto, 2024-04-02:
Hi again @KonstantinKondrashov , I've seen another device rebooting.  I could connect the monitor to it
```

I (7751) wifi:mode : sta (40:4c:ca:44:f8:28)
I (7831) wifi:enable tsf
I (7871) provision_app: Event handler WIFI_EVENT 2
I (7981) BLE_INIT: Using main XTAL as clock source
I (8101) BLE_INIT: ble controller commit:[217f1bf]
I (8211) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (8341) protocomm_nimble: BLE Host Task Started
I (8441) NimBLE: GAP prE (8491) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (8491) task_wdt:  - IDLE (CPU 0)
E (8491) task_wdt: Tasks currently running:
E (8491) task_wdt: CPU 0: nimble_host
E (8491) task_wdt: Print CPU 0 (current core) registers
Core �0 register dump:
MEPC    : 0x42007a68  RA      : 0x42007914  SP      : 0x40855a10  GP      : 0x4081a380  
0x42007a68: usb_serial_jtag_ll_write_txfifo at /home/txinto/esp/v5.1.3/esp-idf/components/hal/esp32c6/include/hal/usb_serial_jtag_ll.h:135
 (inlined by) usb_serial_jtag_tx_char at /home/txinto/esp/v5.1.3/esp-idf/components/vfs/vfs_usb_serial_jtag.c:122
0x42007914: usb_serial_jtag_write at /home/txinto/esp/v5.1.3/esp-idf/components/vfs/vfs_usb_serial_jtag.c:148 (discriminator 2)

TP      : 0x408070a8  T0      : 0x40028140  T1     Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
0x40028140: __retarget_lock_acquire_recursive in ROM

Core  0 register dump:
MEPC    : 0x42000d48  RA      : 0x42000d1a  SP      : 0x4081d630  GP      : 0x4081a380  
0x42000d48: usb_serial_jtag_ll_write_txfifo at /home/txinto/esp/v5.1.3/esp-idf/components/hal/esp32c6/include/hal/usb_serial_jtag_ll.h:135
 (inlined by) panic_print_char_usb_serial_jtag at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:114
 (inlined by) panic_print_char at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:130
0x42000d1a: usb_serial_jtag_ll_txfifo_writable at /home/txinto/esp/v5.1.3/esp-idf/components/hal/esp32c6/include/hal/usb_serial_jtag_ll.h:157
 (inlined by) panic_print_char_usb_serial_jtag at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:109
 (inlined by) panic_print_char at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:130

TP      : 0x408070a8  T0    0x408070a8: spi_flash_restore_cache at /home/txinto/esp/v5.1.3/esp-idf/components/spi_flash/cache_utils.c:371

  : 0x42130bfc  T1      : 0x42130c38  T2      : 0x42130bf0  
S0/FP   : 0x6000f000  S1      : 0x00000020  A0      : 0x60000000  A1      : 0x4081d63b  
A2      : 0x00000001  A3      : 0x4081d63c  A4      : 0x00000020  A5      : 0x00000049  
A6      : 0x00000004  A7      : 0x42130c44  S2      : 0x40837000  S3      : 0x000001f3  
S4      : 0x42130000  S5      : 0x42131000  S6      : 0x42131000  S7      : 0x00000094  
S8      : 0x0000001c  S9      : 0x00000001  SESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x42000d46
0x42000d46: usb_serial_jtag_ll_write_txfifo at /home/txinto/esp/v5.1.3/esp-idf/components/hal/esp32c6/include/hal/usb_serial_jtag_ll.h:135
 (inlined by) panic_print_char_usb_serial_jtag at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:114
 (inlined by) panic_print_char at /home/txinto/esp/v5.1.3/esp-idf/components/esp_system/panic.c:130

SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xb54
load:0x4086e610,len:0x2bc4
load:0x40875728,len:0x121c
entry 0x4086c410
I (355) cpu_start: Unicore app
None
Waiting for the device to reconnect
I (926) ota_func: Verifying fw ver: 7.3.0.11-forget-hotfix2
I (926) ota_func: Booting fw ver: 7.3.0.11-forget-hotfix2
I (926) main: main: Starting the infinite_app_loop
I (926) main: Starting the function
I (936) func: Frequency 1 log cycles 1000
I (936) provision_app: modbus_mode 0
W (946) PL1_functions: RESET REASON 6
I (946) provision_app: cloud_mode 0
I (946) provision_app: Location string len 4, location = Dev
```

Regards.

KonstantinKondrashov, 2024-04-03:
Thanks for the log @txinto!
Are you using the USB Serial/JTAG port (USJ) for logging? If so then I would ask you to try to switch to UART. If after that you will not observe this issue then we can conclude that the issue is in USJ functionality.

Another thing that can be tested is disconnecting USB from the device (but keeping it's power). We guess that it can stop trigger repeat reboots (seems, the internal USJ FSM is not cleared properly).

txinto, 2024-04-10:
Hi @KonstantinKondrashov , after long-term tests done using a power supply, we suspect the issue could be in the USB power supply.  We have altered the way the USB port is managed to avoid having a programming jumper, and this can be causing electrical issues.   With a blank device (no firmware or bootloader on it, complete flash erased) and connected to the USB port of a windows machine which seem to be more sensitive to issues, we saw the PC screen flickering and the serial port reconnecting after each reboot, and we could not see this behavior on a DevKit board.

KonstantinKondrashov, 2024-04-12:
@txinto 
Is there anything left that we can do to help? If not, could you close the ticket?
Seems it is related only to your custom board.
Thanks.

txinto, 2024-04-12:
Thanks, I closed it.

I think it is a good reference in case this happens again to another person.

Regards.

closed 2024-04-12 by txinto