espressif/esp-idf #13480: FatFS corruption in SDK 5.2.1 [bug]

filzek, 2024-03-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using SDK release v5.2.1, the FatFS with Wearleaveling when concurrence happen in ISR, GPTIMER, Console or UART exchange information and File Read/Write the partition get full corrupted, we will also open a ticket for it.

haberturdeur, 2024-04-02:
Hi, thank you for the report.
Does this happen consistently or is it just sometimes? Could there have been a power fluctuation (it is known to cause FS corruption)?
If it's somewhat consistent, could you provide an example that demonstrates the problem?

filzek, 2024-04-04:
> Hi, thank you for the report. Does this happen consistently or is it just sometimes? Could there have been a power fluctuation (it is known to cause FS corruption)? If it's somewhat consistent, could you provide an example that demonstrates the problem?

We can recreate all the time the bug, we are tracking it right now, the panic is o hard that the debug doesnt even show anything, just crash and reboot.

adokitkat, 2024-04-15:
Hi. Could you please share a zip file with the folder of your project which can be build to reproduce the issue? Please also set log level to verbose in the menuconfig and share the output here as well. Thank you.

AxelLin, 2024-05-06:
@filzek 
Any update for the issue and any feedback for the comment https://github.com/espressif/esp-idf/issues/13480#issuecomment-2056609036 ?

filzek, 2024-05-06:
> Hi. Could you please share a zip file with the folder of your project which can be build to reproduce the issue? Please also set log level to verbose in the menuconfig and share the output here as well. Thank you.

Sure, I understand. We're currently focused on ensuring that SDK version 5.2.1 operates flawlessly. We're close to completing all necessary adjustments and fixes. Once we're done, we will prepare and share a split version of the firmware code, which includes the fatfs corruption issue, for your review. Thank you for your patience and understanding.

filzek, 2024-05-06:
@AxelLin we thing we can do it in about 10 days so far.

AxelLin, 2024-06-05:
> @AxelLin we thing we can do it in about 10 days so far.

@filzek Do you still hit this issue?

filzek, 2024-06-05:
> > @AxelLin we thing we can do it in about 10 days so far.
> 
> @filzek Do you still hit this issue?

In sdk 5.2.2 we are testing now, issue have almost solved but still happen on 5.2.1, now we are testing again with forcing ble, wifi, udp, and all channes requests with massive gpio isrs acting over while saving small data and large data, lets see if it survives

closed 2024-10-31 by filzek