espressif/esp-idf #13482: nimble spp_server example may have a memory leak

nopnop2002, 2024-03-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_spp/spp_server

```
            case UART_DATA:
                if (event.size) {
                    ESP_LOGI(TAG, "event.size=%d", event.size);
                    uint8_t *ntf;
                    ntf = (uint8_t *)malloc(sizeof(uint8_t) * event.size);
                    memset(ntf, 0x00, event.size);
                    uart_read_bytes(UART_NUM_0, ntf, event.size, portMAX_DELAY);
                    ESP_LOGI(TAG, "[%.*s] %d", event.size, ntf, event.size);

                    for (int i = 0; i <= CONFIG_BT_NIMBLE_MAX_CONNECTIONS; i++) {
                        /* Check if client has subscribed to notifications */
                        ESP_LOGI(TAG, "conn_handle_subs[%d]=%d", i, conn_handle_subs[i]);
                        if (conn_handle_subs[i]) {
                            struct os_mbuf *txom;
                            //txom = ble_hs_mbuf_from_flat(ntf, sizeof(ntf)); ---> Changed
                            txom = ble_hs_mbuf_from_flat(ntf, event.size);
                            rc = ble_gatts_notify_custom(i, ble_spp_svc_gatt_read_val_handle,
                                                         txom);
                            if (rc == 0) {
                                MODLOG_DFLT(INFO, "Notification sent successfully");
                            } else {
                                MODLOG_DFLT(INFO, "Error in sending notification rc = %d", rc);
                            }
                        }
                    }
                    free(ntf); ---> Added
                }
                break;
            default:
```

rahult-github, 2024-03-27:
Hi @nopnop2002 , thanks for reporting. Will fix this

nopnop2002, 2024-03-27:
There is one more bug.
```
                            //txom = ble_hs_mbuf_from_flat(ntf, sizeof(ntf)); ---> Changed
                            txom = ble_hs_mbuf_from_flat(ntf, event.size);
```

rahult-github, 2024-04-06:
Merged  via https://github.com/espressif/esp-idf/commit/eafad32dc64c827fc3255e33d1ef97c83de3655f .

closed 2024-04-06 by rahult-github