espressif/esp-idf #13483: Failed to read NVS keys (0x1117) when pre-encrypted binary with secure boot and flash encryption enabled

aectaan, 2024-03-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Chip: ESP32C3
ESP-IDF version: release/v4.4

We have the next production cycle at the factory:
1. read MAC, register the device, generate flash encryption key (unique for each device), burn the key into efuse BLOCK_KEY0
2. Download test firmware (both firmwares are huge and can't fit into flash simultaneously), encrypt it with generated key (bootloader, partition table, app), upload encrypted binaries to device with esptool, burn SPI_BOOT_CRYPT_CNT
3. Perform tests
4. Download prod firmware, encrypt it with the same key, upload it to device via esptool.

In this routine we get `ESP_ERR_NVS_CORRUPT_KEY_PART` for prod firmware. I looked into the docs, they suggests to completely erase flash before flashing firmware with different partition table. Unfortunately esp32c3 ROM doesn't support flash erasing when secure boot is enabled.
If I disable NVS encryption - everything is ok.

We flash both firmwares encrypted because want be able to reflash it just in case, and don't want to wait till MCU performs self-encryption (1 min) because operator may disconnect device from jig and soft-brick device

How to solve this issue? What additional info could I provide?

kayzen9, 2024-03-27:
we write empty (all 0xff) nvs keys partition of size 4k to flash, it shud work . ..

mahavirj, 2024-04-01:
@aectaan Did you try above suggestion? Flashing (flash writes are supported with secure boot) NVS keys partition with all bytes set to 0xff should help to fix the problem.

aectaan, 2024-04-01:
Yes, tried, looks helpful. Repeating this procedure with more devices now, to be 100% sure

closed 2024-04-01 by aectaan