espressif/esp-idf #13484: adc_continuous_stop() causing  assert failed: xTaskPriorityDisinherit tasks.c:5073 (pxTCB == pxCurrentTCBs[ xPortGetCoreID() ])

kamilsss655, 2024-03-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WeMos LOLIN32 Lite

### Power Supply used.

USB

### What is the expected behavior?

I am working on this project https://github.com/kamilsss655/ESPRI and want to:
* output PDM audio on a pin on demand
* when not outputting audio I want to sample audio input with I2S adc_continuous

In order to do this I normally run adc_continuous to sample incoming audio, and when I want to transmit audio I want to stop the `adc_continuous_stop` (as they cannot run at the same time) and use `i2s_channel_init_pdm_tx_mode` to output audio.

Here is `audio.c` file source:

```
#include <driver/i2s.h>
#include <stdint.h>
#include <string.h>
#include <math.h>
#include <driver/i2s_pdm.h>
#include <driver/gpio.h>
#include <esp_check.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/semphr.h>
#include <esp_adc/adc_continuous.h>

#include "audio.h"
#include "settings.h"
#include "helper/rtos.h"

i2s_pin_config_t pin_config;

#define EXAMPLE_ADC_UNIT ADC_UNIT_1
#define _EXAMPLE_ADC_UNIT_STR(unit) #unit
#define EXAMPLE_ADC_UNIT_STR(unit) _EXAMPLE_ADC_UNIT_STR(unit)
#define EXAMPLE_ADC_CONV_MODE ADC_CONV_SINGLE_UNIT_1
#define EXAMPLE_ADC_ATTEN ADC_ATTEN_DB_12
#define EXAMPLE_ADC_BIT_WIDTH SOC_ADC_DIGI_MAX_BITWIDTH
[... 377 lines cut ...]
assert failed: xTaskPriorityDisinherit tasks.c:5073 (pxTCB == pxCurrentTCBs[ xPortGetCoreID() ])

Backtrace: 0x400819aa:0x3ffba1c0 0x4008a9b9:0x3ffba1e0 0x40092aad:0x3ffba200 0x4008d99e:0x3ffba320 0x4008aa17:0x3ffba340 0x4008aee1:0x3ffba360 0x40082c85:0x3ffba3a0 0x40082d35:0x3ffba3d0 0x4011fd26:0x3ffba3f0 0x400eea09:0x3ffba420 0x400d780d:0x3ffba450 0x400d7933:0x3ffba470 0x400d7acb:0x3ffba4f0 0x400d746b:0x3ffba520 0x4008b5a9:0x3ffba540
0x400819aa: panic_abort at /home/nunu/esp/v5.2.1/esp-idf/components/esp_system/panic.c:472
0x4008a9b9: esp_system_abort at /home/nunu/esp/v5.2.1/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x40092aad: __assert_func at /home/nunu/esp/v5.2.1/esp-idf/components/newlib/assert.c:81
0x4008d99e: xTaskPriorityDisinherit at /home/nunu/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5073 (discriminator 1)
0x4008aa17: prvCopyDataToQueue at /home/nunu/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2469
0x4008aee1: xQueueGenericSend at /home/nunu/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:964
0x40082c85: lock_release_generic at /home/nunu/esp/v5.2.1/esp-idf/components/newlib/locks.c:192
0x40082d35: _lock_release at /home/nunu/esp/v5.2.1/esp-idf/components/newlib/locks.c:198
0x4011fd26: adc_lock_release at /home/nunu/esp/v5.2.1/esp-idf/components/esp_hw_support/adc_share_hw_ctrl.c:153
0x400eea09: adc_continuous_stop at /home/nunu/esp/v5.2.1/esp-idf/components/esp_adc/adc_continuous.c:424
0x400d780d: AUDIO_AdcStop at /home/nunu/Documents/projects/personal/esp32/ESPRI/main/hardware/audio.c:118
0x400d7933: AUDIO_Transmit at /home/nunu/Documents/projects/personal/esp32/ESPRI/main/hardware/audio.c:209
0x400d7acb: AUDIO_PlayTone at /home/nunu/Documents/projects/personal/esp32/ESPRI/main/hardware/audio.c:280
0x400d746b: MORSE_CODE_Transmit at /home/nunu/Documents/projects/personal/esp32/ESPRI/main/app/morse_code.c:52
0x4008b5a9: vPortTaskWrapper at /home/nunu/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

This seems to be affected by the fact that the function is initialized from the other file? When I run `adc_continuous_stop` within the `audio.c` file there is no such error.

### Steps to reproduce.

1. To reproduce one can compile the code available here: https://github.com/kamilsss655/ESPRI/tree/audio-input-wip

kamilsss655, 2024-03-28:
So if anyone else encounters this issue, as a workaround I was able to resolve this by calling `
adc_continuous_stop()` and `adc_continuous_start()` from the same task and control it externally by using `EventGroup` in FreeRTOS to signal whether adc should be started or stopped.

If above is not considered bug, then I guess this issue can be closed?

closed 2024-03-28 by kamilsss655

filzek, 2024-06-24:
I think this is a very dangerous problem, facing same issue, in this way if you need to control for a multiple tasking become a really problem until it takes engress, so, this must be solved to be called anywhere you want as what defines the use of it is the HANDLE, so, makes no sense at all.

@igrr Ivanm can you check the problem above as its something that need a solution to be called anywhere?