espressif/esp-idf #13485: How can i connect ESP32-32D on another device via ModBus UDP?

pedrohugo-psc, 2024-03-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Espressif SoC revision.

ESP32-32D

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Devkit_v4

### Power Supply used.

USB

### What is the expected behavior?

I want to connect ESP32-32D on another device via ModBus UDP

### What is the actual behavior?

I receive this error when I try to set mb_mode_type_t to MB_MODE_UDP: ESP_ERR_INVALID_ARG
 

### Steps to reproduce.

1. Create the Modbus TCP Master Example (mb_tcp_master)
2. Go to app_main() 
3. Find comm_info struct
4. Change comm_info.ip_mode from MB_MODE_TCP to MB_MODE_UDP
5. Run the code

### Debug Logs.

```plain
I (610) example_connect: Waiting for IP(s)
I (2610) esp_netif_handlers: example_connect: eth ip: 10.1.25.200, mask: 255.255.255.0, gw: 10.1.25.1
I (2610) example_connect: Got IPv4 event: Interface "example_connect: eth" address: 10.1.25.200
I (2620) example_connect: Connected to example_connect: eth
I (2630) example_connect: - IPv4 address: 10.1.25.200
I (2630) CONNECTION_CONFIG: Leave IP(0) = [10.1.25.100] set manually.
I (2640) CONNECTION_CONFIG: IP(1) is not set in the table.
I (2650) CONNECTION_CONFIG: Configured 1 IP addresse(s).
E (2660) MB_CONTROLLER_MASTER: mbc_tcp_master_setup(124): mb incorrect mode = (0x3).
E (2670) MB_CONTROLLER_MASTER: mbc_master_setup(157): Master setup failure, error=(0x102) (ESP_ERR_INVALID_ARG).
E (2670) MASTER_TEST: master_init(120): mb controller setup fail, returns(0x102).
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40087cdc
0x40087cdc: _esp_error_check_failed at C:/esp/esp-idf/components/esp_system/esp_err.c:42

ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40087cdc
0x40087cdc: _esp_error_check_failed at C:/esp/esp-idf/components/esp_system/esp_err.c:42

file: "./main/modbus_tcp_ledax.c" line 160
func: modbus_tcp_ledax
expression: master_init(&comm_info)

abort() was called at PC 0x40087cdf on core 0
0x40087cdf: _esp_error_check_failed at C:/esp/esp-idf/components/esp_system/esp_err.c:43

Backtrace: 0x40081cce:0x3ffb6ac0 0x40087ce9:0x3ffb6ae0 0x4008e87e:0x3ffb6b00 0x40087cdf:0x3ffb6b70 0x400d81b8:0x3ffb6b90 0x400d7adb:0x3ffb6bd0 0x4011de78:0x3ffb6bf0 0x4008b659:0x3ffb6c10
0x40081cce: panic_abort at C:/esp/esp-idf/components/esp_system/panic.c:408

0x40087ce9: esp_system_abort at C:/esp/esp-idf/components/esp_system/esp_system.c:137

0x4008e87e: abort at C:/esp/esp-idf/components/newlib/abort.c:46

0x40087cdf: _esp_error_check_failed at C:/esp/esp-idf/components/esp_system/esp_err.c:43

0x400d81b8: modbus_tcp_ledax at C:/Users/Ledax/Documents/Pedro-Doc/ESP32/modbus_tcp_sdcard/main/modbus_tcp_ledax.c:160 (discriminator 1)

0x400d7adb: app_main at C:/Users/Ledax/Documents/Pedro-Doc/ESP32/modbus_tcp_sdcard/main/tcp_master.c:18

0x4011de78: main_task at C:/esp/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

0x4008b659: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/port/xtensa/port.c:142
```

Alvin1Zhang, 2024-03-28:
Thanks for reporting.

pedrohugo-psc, 2024-03-28:
> Thanks for reporting.

Hello, Alvin1Zhang, 

What is the prediction for any response to assist me?

alisitsyn, 2024-04-02:
Hello @pedrohugo-psc, 

The error ESP_ERR_INVALID_ARG means the argument is not supported. The UDP communication for the stack is not yet supported. In order to communicate over UDP you need to update the stack code on your side. Let me know if you need more information on how to accomplish this. I also see that you use IDF v4.4, I would recommend to use the [esp-modbus latest version](https://github.com/espressif/esp-modbus) in your project. You need to add the line `set(EXCLUDE_COMPONENTS freemodbus)` to your CMakeLists.txt file as described in the readme.md file.

VitorBFreitas, 2024-10-29:
Hello @alisitsyn ,

Could you provide a brief overview of the steps required to update the stack code to support Modbus over UDP? I'd like to get a sense of the development complexity involved.