espressif/esp-idf #13486: GPTIMER doesnt have a exposed api to know the GPTIMER status

filzek, 2024-03-28:
### Is your feature request related to a problem?

When running a timer in a multiple concurrency environment it has the need to know if the timer is running or not, so, we can decide what to do.

currently the time state is now exposed in the api to get the timer status such as gptimer_fsm_t;

typedef enum {
    GPTIMER_FSM_INIT,        // Timer is initialized, but not enabled
    GPTIMER_FSM_ENABLE,      // Timer is enabled, but is not running
    GPTIMER_FSM_ENABLE_WAIT, // Timer is in the middle of the enable process (Intermediate state)
    GPTIMER_FSM_RUN,         // Timer is in running
    GPTIMER_FSM_RUN_WAIT,    // Timer is in the middle of the run process (Intermediate state)
} gptimer_fsm_t;

so, inside the gptimer object there is _Atomic gptimer_fsm_t fsm;, 

It´s complete important to have a basic function to retrieve the true GPTIMER status, like
esp_err_t gptimer_get_status(gptimer_handle_t timer, int *value){
ESP_RETURN_ON_FALSE_ISR(timer && value, ESP_ERR_INVALID_ARG, TAG, "invalid argument");
*value = timer_hal_capture_and_get_running_state_value(&timer->fsm);
}

This is the idea, but you guys can figure out how to do it simple when timer is running to be dispatch or not.

filzek, 2024-03-28:
or something like

    if (atomic_compare_exchange_strong(&timer->fsm, &expected_fsm, GPTIMER_FSM_RUN)) {
        // timer is running
    } else {
        // timer is not running
    }

suda-morris, 2024-03-28:
But even the driver returned "I'm currently in running state", at the time the function returns, the state may be changed by another task where called the stop function. So the returned state is out-of-date again.

filzek, 2024-03-28:
so, but how could we get the true status of a running timer, if we need to know if its running or not, this mean, if its running it will be going to dispatch the callback sometime.

If we return this state we will know if the timer at least is true running

filzek, 2024-03-28:
> But even the driver returned "I'm currently in running state", at the time the function returns, the state may be changed by another task where called the stop function. So the returned state is out-of-date again.

Sure, so the STOP function should not return any error to stop a not running timer, so a stop shall do the same, you ask to stop but the timer will ran anyway, the design have some flaws to solve yet.

filzek, 2024-03-28:
uint64_t period = 1000000; // Timer period in ticks (e.g., 1 second) used in the timer config
uint64_t current_count;
uint64_t remaining_time;

// Get the current count value
gptimer_get_raw_count(timer_handle, &current_count);

// Calculate the remaining time
remaining_time = period - current_count;

// Convert the remaining time to a more convenient unit if necessary

so, this is just the same problem, once you get it the time could already be dispatched.

the new function could return the current state and the time left to dispatch.

suda-morris, 2024-03-28:
Do you want the start/stop function to be idempotent? By relaxing the FSM check, allow calling stop/start when it's already in stop/start. If so, do you still want a get status function? I'm afraid the "status" thing is not reflected by the hardware, but a software thing. We may introduce more state into the `gptimer_fsm_t`. If we expose such a function, it's easy to introduce breaking change :thinking:

BAL3IA, 2024-05-09:
It would be really helpful to check timer's status when running one_shot triggered events/interrupts

eroto, 2025-01-03:
Agree with @suda-morris an API to get a timer status is basic and common.
I reach this issue because my app also requires the API to know timer status.

closed 2025-02-26 by espressif-bot