espressif/esp-idf #13488: Method to use USB camera with esp32-s3-devkitC-1N8 board

simardeep272, 2024-03-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, hope you are doing well.

I wanted to ask whether there is any method to run the usb-uvc example for the esp32-s3-dekitC-1N8 board. The README does state that an external PSRAM is required to initialize the example, is there any workaround for that, or any other method to connect a camera(usb or spi) to this specific board?

Thanks

roma-jam, 2024-03-28:
Hey @simardeep272,

usb-uvc example (host) requires the external PSRAM to provide enough RAM memory for transfering frames from connected usb device to WiFi server.

There could be some options, like decreasing the resolution, for example.  

Could you please share more details about your task?

simardeep272, 2024-03-29:
Hi @roma-jam , thanks for the reply.

I am currently running the usb-uvc example. Prior to flashing the device, I have made a change in the menuconfig of continuing even if PSRAM is not found and these are the logs(The device fails to find the usb camera attached to the port):

Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1778
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e0c
SHA-256 comparison failed:
Calculated: b6c3e089af2070a5b47799ab7a990146c54fe2acbb1792ffe9ba416d0db900c3
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c9908
I (45) boot: ESP-IDF v5.1.3-412-g9f7eacb089-dirty 2nd stage bootloader
I (45) boot: compile time Mar 26 2024 22:17:43
I (46) boot: Multicore bootloader
I (50) boot: chip revision: v0.2
I (54) boot.esp32s3: Boot SPI Speed : 80MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (63) boot.esp32s3: SPI Flash Size : 8MB
I (68) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (92) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  2 factory          factory app      00 00 00010000 00100000
I (107) boot: End of partition table
I (111) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0e068h ( 57448) map
I (126) esp_image: segment 1: paddr=0001e090 vaddr=3fc92c00 size=01f88h (  8072) load
I (129) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=253bch (152508) map
I (154) esp_image: segment 3: paddr=000453e4 vaddr=3fc94b88 size=00b30h (  2864) load
I (154) esp_image: segment 4: paddr=00045f1c vaddr=40374000 size=0eba0h ( 60320) load
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (186) cpu_start: Multicore app
E (187) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (190) esp_psram: PSRAM enabled but initialization failed. Bailing out.
I (198) cpu_start: Failed to init external RAM; continuing without it.
I (205) cpu_start: Pro cpu up.
I (209) cpu_start: Starting app cpu, entry point is 0x403754b0
0x403754b0: call_start_cpu1 at /home/../esp/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (232) cpu_start: Pro cpu start user code
I (232) cpu_start: cpu freq: 160000000 Hz
I (232) cpu_start: Application information:
I (235) cpu_start: Project name:     usb-uvc
I (240) cpu_start: App version:      v5.1.3-412-g9f7eacb089-dirty
I (247) cpu_start: Compile time:     Mar 29 2024 12:39:31
I (253) cpu_start: ELF file SHA256:  3637f16f7523a210...
I (259) cpu_start: ESP-IDF:          v5.1.3-412-g9f7eacb089-dirty
I (266) cpu_start: Min chip rev:     v0.0
I (270) cpu_start: Max chip rev:     v0.99 
I (275) cpu_start: Chip rev:         v0.2
I (280) heap_init: Initializing. RAM available for dynamic allocation:
I (288) heap_init: At 3FC95F80 len 00053790 (333 KiB): DRAM
I (293) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (300) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (306) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (315) spi_flash: detected chip: gd
I (317) spi_flash: flash io: dio
I (321) sleep: Configure to isolate all GPIO pins in sleep state
I (328) sleep: Enable automatic switching of GPIO sleep configuration
I (335) app_start: Starting scheduler on CPU0
I (0) app_start: Starting scheduler on CPU1
I (340) main_task: Started on CPU0
I (350) main_task: Calling app_main()
I (350) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (390) example: Waiting for USB UVC device connection ...

I am currently connecting my usb camera(Logitech C270) with a micro-usb to usb-3.0 adapter.

Should I try using a 4-pin converter board and use GPIO 19,20 for the same?

roma-jam, 2024-03-29:
Hi @simardeep272 ,

there is a specific with this board.
Please, refer to the comment here: https://github.com/espressif/esp-idf/issues/13327#issuecomment-1983808858

Regarding the PSRAM: it seems that it fails during the initialization.
Please, try to change quad to octal in menuconfig and rebuld the project.

simardeep272, 2024-03-29:
I have tried testing with Octal Mode PSRAM, but am still getting the same error that PSRAM chip not found or not supported. If I am not wrong, isn't that the expected error because devkitC-1N8 doesn't have PSRAM.

Thanks for suggesting the connector device, will definitely try this

roma-jam, 2024-03-29:
@simardeep272 
That is right, devboard 1N8 has only 8MB of flash and no ram.

[Ordering information](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html#ordering-information)

simardeep272, 2024-04-01:
Hi thanks for the suggestion, I used the method that you mentioned, attached a DIY converter to the GPIO pins and I was able to detect the camera. But there is one issue, after detecting the camera I am not able to capture a frame. These are the logs:

DEVICE CONFIGURATION (046d:0825/[none]) ---
Status: idle
VideoControl:
	bcdUVC: 0x0100
VideoStreaming(1):
	bEndpointAddress: 129
	Formats:
	UncompressedFormat(1)
		  bits per pixel: 16
		  GUID: 5955593200001000800000aa00389b71 (YUY2)
		  default frame: 1
		  aspect ratio: 0x0
		  interlace flags: 00
		  copy protect: 00
			FrameDescriptor(1)
			  capabilities: 01
			  size: 160x120
			  bit rate: 1536000-4608000
			  max frame size: 38400
			  default interval: 1/15
			  interval[0]: 1/15
			  interval[1]: 1/10
			  interval[2]: 1/7
			  interval[3]: 1/5
			FrameDescriptor(2)
			  capabilities: 01
			  size: 176x144
			  bit rate: 2027520-6082560
			  max frame size: 50688
			  default interval: 1/15
			  interval[0]: 1/15
			  interval[1]: 1/10
			  interval[2]: 1/7
			  interval[3]: 1/5
	MJPEGFormat(2)
		  bits per pixel: 0
		  GUID: 4d4a5047000000000000000000000000 (MJPG)
		  default frame: 1
		  aspect ratio: 0x0
		  interlace flags: 00
		  copy protect: 00
			FrameDescriptor(1)
			  capabilities: 01
			  size: 640x480
			  bit rate: 24576000-147456000
			  max frame size: 614400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(2)
			  capabilities: 01
			  size: 160x120
			  bit rate: 1536000-9216000
			  max frame size: 38400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(3)
			  capabilities: 01
			  size: 176x144
			  bit rate: 2027520-12165120
			  max frame size: 50688
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(4)
			  capabilities: 01
			  size: 320x176
			  bit rate: 4505600-27033600
			  max frame size: 112640
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(5)
			  capabilities: 01
			  size: 320x240
			  bit rate: 6144000-36864000
			  max frame size: 153600
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(6)
			  capabilities: 01
			  size: 352x288
			  bit rate: 8110080-48660480
			  max frame size: 202752
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(7)
			  capabilities: 01
			  size: 432x240
			  bit rate: 8294400-49766400
			  max frame size: 207360
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(8)
			  capabilities: 01
			  size: 544x288
			  bit rate: 12533760-75202560
			  max frame size: 313344
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(9)
			  capabilities: 01
			  size: 640x360
			  bit rate: 18432000-110592000
			  max frame size: 460800
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/24
			  interval[2]: 1/20
			  interval[3]: 1/15
			  interval[4]: 1/10
			  interval[5]: 1/7
			  interval[6]: 1/5
			FrameDescriptor(10)
			  capabilities: 01
			  size: 752x416
			  bit rate: 25026560-75079680
			  max frame size: 625664
			  default interval: 1/15
			  interval[0]: 1/15
			  interval[1]: 1/10
			  interval[2]: 1/7
			  interval[3]: 1/5
			FrameDescriptor(11)
			  capabilities: 01
			  size: 800x448
			  bit rate: 28672000-86016000
			  max frame size: 716800
			  default interval: 1/15
			  interval[0]: 1/15
			  interval[1]: 1/10
			  interval[2]: 1/7
			  interval[3]: 1/5
			FrameDescriptor(12)
			  capabilities: 01
			  size: 800x600
			  bit rate: 38400000-115200000
			  max frame size: 960000
			  default interval: 1/15
			  interval[0]: 1/15
			  interval[1]: 1/10
			  interval[2]: 1/7
			  interval[3]: 1/5
END DEVICE CONFIGURATION
I (172670) example: Negotiate streaming profile 100x100, 30 fps ...
E (173670) example: Negotiation failed. Try again (2) ...
I (173670) example: Negotiate streaming profile 100x100, 30 fps ...
E (174670) example: Negotiation failed. Try again (1) ...
I (174670) example: Negotiate streaming profile 100x100, 30 fps ...

roma-jam, 2024-04-01:
@simardeep272 , 
There is not enough RAM to get the frame buffer for this resolution. 

Try 160x120 with interval 1/5.

simardeep272, 2024-04-01:
Hi, thanks a lot for your assistance, I was able to run the camera on the resolution you mentioned.

roma-jam, 2024-04-01:
@simardeep272 
No problem. 
You can try to choose another parameters, just pay attention to the max_frame_size value and select the resolution based on the ram usage in your project. 

Happy that I can help. 
Can we close the issue now?

simardeep272, 2024-04-01:
Sure, thanks again for your help

simardeep272, 2024-04-03:
Hi, sorry to reopen the issue, I tested the esp32-s3-devkitC-1 with extended PSRAM as you suggested, I am current using N32R8V,

while running the code with the manual streaming parameters and MJPEG mode, I am getting this error:

E (21250) libusb adapter: Failed to allocate transfer with length: 16384

roma-jam, 2024-04-03:
Hey @simardeep272, 

Can you share the full debug log from the start till the allocation error?

simardeep272, 2024-04-04:
Hi @roma-jam ,

These are the logs:

I (33) boot: ESP-IDF v5.2-dirty 2nd stage bootloader
I (33) boot: compile time Apr  3 2024 16:04:32
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v0.1
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : SLOW READ
I (51) boot.esp32s3: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=26ec8h (159432) map
I (144) esp_image: segment 1: paddr=00036ef0 vaddr=3fc99000 size=04918h ( 18712) load
I (150) esp_image: segment 2: paddr=0003b810 vaddr=40374000 size=04808h ( 18440) load
I (155) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=961bch (614844) map
I (305) esp_image: segment 4: paddr=000d61e4 vaddr=40378808 size=107ach ( 67500) load
I (332) boot: Loaded app from partition at offset 0x10000
I (332) boot: Disabling RNG early entropy source...
I (344) cpu_start: Multicore app
I (353) cpu_start: Pro cpu start user code
I (353) cpu_start: cpu freq: 160000000 Hz
I (353) cpu_start: Application information:
I (356) cpu_start: Project name:     usb-uvc
I (361) cpu_start: App version:      v5.2-dirty
I (366) cpu_start: Compile time:     Apr  4 2024 10:56:59
I (373) cpu_start: ELF file SHA256:  bad50bb66...
I (378) cpu_start: ESP-IDF:          v5.2-dirty
I (383) cpu_start: Min chip rev:     v0.0
I (388) cpu_start: Max chip rev:     v0.99 
I (393) cpu_start: Chip rev:         v0.1
I (397) heap_init: Initializing. RAM available for dynamic allocation:
I (405) heap_init: At 3FCA1B50 len 00047BC0 (286 KiB): RAM
I (411) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (417) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (423) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (430) spi_flash: detected chip: mxic (opi)
I (434) spi_flash: flash io: opi_str
W (438) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (452) sleep: Configure to isolate all GPIO pins in sleep state
I (459) sleep: Enable automatic switching of GPIO sleep configuration
I (466) main_task: Started on CPU0
I (476) main_task: Calling app_main()
I (476) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (526) example_connect: Start example_connect.
I (526) pp: pp rom version: e7ae62f
I (526) net80211: net80211 rom version: e7ae62f
I (536) wifi:wifi driver task: 3fcb0608, prio:23, stack:6656, core=0
I (536) wifi:wifi firmware version: cc1dd81
I (536) wifi:wifi certification version: v7.0
I (546) wifi:config NVS flash: enabled
I (546) wifi:config nano formating: disabled
I (546) wifi:Init data frame dynamic rx buffer num: 8
I (556) wifi:Init static rx mgmt buffer num: 5
I (556) wifi:Init management short buffer num: 32
I (566) wifi:Init dynamic tx buffer num: 32
I (566) wifi:Init static tx FG buffer num: 2
I (566) wifi:Init static rx buffer size: 1600
I (576) wifi:Init static rx buffer num: 8
I (576) wifi:Init dynamic rx buffer num: 8
I (586) wifi_init: rx ba win: 8
I (586) wifi_init: tcpip mbox: 32
I (586) wifi_init: udp mbox: 6
I (596) wifi_init: tcp mbox: 6
[... 216 lines cut ...]
			  size: 800x600
			  bit rate: 38400000-115200000
			  max frame size: 960000
			  default interval: 1/15
			  interval[0]: 1/15
			  interval[1]: 1/10
			  interval[2]: 1/7
			  interval[3]: 1/5
END DEVICE CONFIGURATION
I (14926) example: Negotiate streaming profile 160x120, 15 fps ...
I (14936) example: Negotiation complete.
bmHint: 0001
bFormatIndex: 2
bFrameIndex: 2
dwFrameInterval: 666666
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 61
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 38400
dwMaxPayloadTransferSize: 512
bInterfaceNumber: 1
E (14956) libusb adapter: Failed to allocate transfer with length: 16384
I (14966) example: Streaming...

nopnop2002, 2024-04-05:
@simardeep272 

There are only a limited number of USB cameras that can be used in the USB-UVC example.

This might help.

https://github.com/nopnop2002/esp-idf-video-streaming

simardeep272, 2024-04-08:
Hi @nopnop2002 , I am using the suggested camera only: Logitech C270

roma-jam, 2024-04-08:
Hi @simardeep272, 

I see that based on the log there is another memory distribution (then it was with previous board), but I don‘t see any log from esp_psram. 

Just in case: is the feature enabled in menuconfig?

simardeep272, 2024-04-08:
I don't see an option to enable the psram in menuconfig. The component config is not showing ESP PSRAM as an option

nopnop2002, 2024-04-08:
@simardeep272 

>I am using the suggested camera only: Logitech C270

C270 and Microsoft LifeCam only work with UncompressedFormat VideoStreaming format.

Official samples do not support this format.

simardeep272, 2024-04-08:
@nopnop2002 , right.
What do you think might be causing this error then?, because I can see the indicator on my camera being lit up, so the camera is working. Also, I have tested the PSRAM and memory in the devkit, it is being detected.

nopnop2002, 2024-04-08:
Logitech C270 is recognized as follows.

Official examples do not support UncompressedFormat.

```
Device found
DEVICE CONFIGURATION (046d:0825/2) ---
Status: idle
VideoControl:
        bcdUVC: 0x0100
VideoStreaming(1):
        bEndpointAddress: 129
        Formats:
        UncompressedFormat(1)
                  bits per pixel: 16
                  GUID: 5955593200001000800000aa00389b71 (YUY2)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 01
                          size: 160x120
                          bit rate: 3072000-3072000
                          max frame size: 38400
                          default interval: 1/10
                          interval[0]: 1/10
END DEVICE CONFIGURATION
```

simardeep272, 2024-04-08:
Oh, but I am able to get the device configuration, the issue seems to be maybe in the memory allocation for the frame

nopnop2002, 2024-04-08:
@simardeep272 

Device id of My camera is (046d:0825/2)
Device id of My camera is (046d:0825/7)

I would like to know the part number of your camera tag.
This is my camera tag.
[image]

roma-jam, 2024-04-08:
@simardeep272 , 

> I don't see an option to enable the psram in menuconfig. The component config is not showing ESP PSRAM as an option

esp_psram is a component and to enable it it should be in PRIV_REQUIRES list in CMakeLists.txt. 

Could you please add `esp_psram` to `PRIV_REQUIRES`, to have something like: 

` PRIV_REQUIRES protocol_examples_common nvs_flash usb mdns esp_ringbuf esp_psram esp_timer esp_wifi driver)`

Then relaunch `idf.py menuconfig`?

roma-jam, 2024-04-10:
Hi @simardeep272 ,

is there any news? 
Was you able to activate esp_psram configuration?

simardeep272, 2024-04-12:
Hi @roma-jam,
Apologies for the delay, I enabled the esp_psram config and it worked for me. Thanks a lot for your assistance. Also, is there any way/documentation to understand the frame configurations

closed 2024-04-12 by simardeep272

MuhammadZia-ud-Din, 2024-12-17:
> Hi @roma-jam , thanks for the reply.
>
> I am currently running the usb-uvc example. Prior to flashing the device, I have made a change in the menuconfig of continuing even if PSRAM is not found and [...]

Hi did you find the solution I am going through same issue