espressif/esp-idf #13489: [TLSv1.3] ESP32S3 failing to connect when using DS peripheral

cbaechler, 2024-03-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-234-gfcb76f0671

### Espressif SoC revision.

ESP32-S3 (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Device shall connect to MQTT with TLS v1.3 enabled and using DS peripheral. This test was done with the provided example in https://github.com/espressif/esp-idf/tree/release/v5.2/examples/protocols/mqtt/ssl_ds. 

### What is the actual behavior?

Device connects in the following cases: 
- TLS v1.2 (but TLS v1.3 disabled)
- TLS v1.3 but not using the private key from DS (using `target_add_binary_data` in CMakeLists.txt to embed it)

However, when enabling TLS v1.3 and using the key provided from DS, the device refuses to connect. 

### Steps to reproduce.

1. Follow the steps to generate the client certificate for https://test.mosquitto.org/ssl/ and create the partition. 

```
$ openssl genrsa -out client.key
$ openssl req -out client.csr -key client.key -new
$ py configure_esp_secure_cert.py -p COM27 --device-cert client.crt --private-key client.key --priv_key_algo RSA 2048 --target_chip esp32s3 --configure_ds --skip_flash
Using the same hmac key burned in efuse BLOCK_KEY1
dev_cert tlv: total length = 1312
ds_data tlv: total length = 1632
ds_context tlv: total length = 32
Total length of tlv data = 2976
```

2. Run the example. The device will connect to the broker successfully.
3. Enable TLS v1.3 support in `menuconfig` and run again. The device will not connect with this configuration. 

See the following log: 

```
I (10479) example_common: Connected to example_netif_sta
I (10489) example_common: - IPv4 address: 192.168.141.98,
I (10489) example_common: - IPv6 address: fe80:0000:0000:0000:7edf:a1ff:fee1:a5e0, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (10499) mqtts_example: [APP] Free memory: 287960 bytes
I (10509) mqtts_example: Other event id:7
I (10519) main_task: Returned from app_main()
E (11629) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x6E00
I (11629) esp-tls-mbedtls: Certificate verified.
E (11629) esp-tls: Failed to open new connection
E (11639) transport_base: Failed to open a new connection
E (11639) mqtt_client: Error transport connect
I (11649) mqtts_example: MQTT_EVENT_ERROR
I (11649) mqtts_example: MQTT_EVENT_DISCONNECTED
```

4. Apply following modification and run again (still with TLS v1.3 enabled): 

```diff
diff --git a/examples/protocols/mqtt/ssl_ds/CMakeLists.txt b/examples/protocols/mqtt/ssl_ds/CMakeLists.txt
index ce04bb08b7..5ba376bdcd 100644
--- a/examples/protocols/mqtt/ssl_ds/CMakeLists.txt
+++ b/examples/protocols/mqtt/ssl_ds/CMakeLists.txt
@@ -14,3 +14,4 @@ partition_table_get_partition_info(offset "--partition-name ${partition}" "offse
 esptool_py_flash_target_image(flash "${partition}" "${offset}" "${image_file}")

 target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/mosquitto.org.crt" TEXT)
+target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "client.key" TEXT)
diff --git a/examples/protocols/mqtt/ssl_ds/main/app_main.c b/examples/protocols/mqtt/ssl_ds/main/app_main.c
index 758995b8b8..c582873a1e 100644
--- a/examples/protocols/mqtt/ssl_ds/main/app_main.c
+++ b/examples/protocols/mqtt/ssl_ds/main/app_main.c
@@ -35,6 +35,8 @@ static const char *TAG = "mqtts_example";
 extern const uint8_t server_cert_pem_start[] asm("_binary_mosquitto_org_crt_start");
 extern const uint8_t server_cert_pem_end[] asm("_binary_mosquitto_org_crt_end");

+extern const uint8_t client_key_start[] asm("_binary_client_key_start");
+
 /*
  * @brief Event handler registered to receive MQTT events
  *
@@ -118,8 +120,8 @@ static void mqtt_app_start(void)
         .credentials = {
             .authentication = {
                 .certificate = (const char *)device_cert,
-                .key = NULL,
-                .ds_data = (void *)ds_data
+                .key =  (const char *)client_key_start,
+                .ds_data = NULL,
             },
         },
     };
```

Given the last modification, this hints me towards that something in the configuration of the DS peripheral is different based on whether TLS v1.3 is enabled or not. 

### More Information.

FYI @KonssnoK

KonssnoK, 2024-04-05:
ok thanks!
we put this on hold on our side and wait for updates then.

Harshal5, 2024-04-11:
Hello @KonssnoK,

`mbedtls` does not provide RSA-PSS ALT implementation support similar to that of RSA_ALT ([reference](https://github.com/Mbed-TLS/mbedtls/blob/development/include/mbedtls/pk.h#L79-L80)). We had some discussion about this [here](https://github.com/Mbed-TLS/mbedtls/issues/9016).

As mentioned in the above issue, `mbedtls` won't be supporting this using the **ALT** method but suggested using mbedtls' PSA Crypto implementation. We are currently exploring the migration to mbedtls' PSA Crypto in ESP-IDF and thus getting it fully operational might need some time.

Thus, to support RSA-PSS signatures using ALT implementation in `mbedtls`, you would need to maintain a patch in the `mbedtls` submodule. The patch adds support for RSA-PSS signature generation through the existing RSA_ALT wraps.
Patch to be applied in the `mbedtls` submodule: 
[0001-feat-mbedtls-rsa-Add-support-for-RSA-PSS-signatures-.patch](https://github.com/espressif/esp-idf/files/14945435/0001-feat-mbedtls-rsa-Add-support-for-RSA-PSS-signatures-.patch)

```
cd $IDF_PATH/components/mbedtls/mbedtls
git apply 0001-feat-mbedtls-rsa-Add-support-for-RSA-PSS-signatures-.patch
```

You will also need to apply a patch in ESP-IDF, that adds support for using the DS peripheral for the above RSA-PSS operation.
[0001-feat-mbedtls-ds-Add-support-for-generating-RSA-PSS-s.patch](https://github.com/espressif/esp-idf/files/14945458/0001-feat-mbedtls-ds-Add-support-for-generating-RSA-PSS-s.patch)

```
cd $IDF_PATH
git apply 0001-feat-mbedtls-ds-Add-support-for-generating-RSA-PSS-s.patch
```

Could you confirm if these patches solve your issue?

cbaechler, 2024-04-11:
Thanks @Harshal5 for your explanation and the patches. 

I confirm applying the patches solves our issue. 

Is my assumption correct that once you add the support for the PSA crypto implementation in ESP-IDF, these patches would be redundant and can be removed again? 

Thank you!

closed 2024-04-12 by Harshal5

AxelLin, 2024-04-29:
@Harshal5 

This issue is marked as "Resolution: Done" and closed, so I'm confused if this is already fixed in esp-idf?

Harshal5, 2024-05-07:
Hello @AxelLin,

As I mentioned above,

> mbedtls does not provide RSA-PSS ALT implementation support similar to that of RSA_ALT ([reference](https://github.com/Mbed-TLS/mbedtls/blob/development/include/mbedtls/pk.h#L79-L80)). We had some discussion about this https://github.com/Mbed-TLS/mbedtls/issues/9016.

to support this feature we would need to maintain a patch in the `mbedtls` submodule. But a cleaner solution would be using the PSA crypto implementation of `mbedtls`, which we are currently exploring. Once we add support for the PSA crypto implementation, this feature should be supported even without the patch. Thus we won't be merging this in ESP-IDF for now.