espressif/esp-idf #13494: SMTP Client with TLS 1.3 mbedtls_ssl_read failed with error -0x7b00

zhichunlee, 2024-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2 release version

### Espressif SoC revision.

ESP32(revision v1.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32_DevKitc_V4

### Power Supply used.

USB

### What is the expected behavior?

I used smtp client example connect to gmail server with TLS v1.3  

### What is the actual behavior?

The mbedtls_ssl_handshake() returned -0x7b00, and mbedtls_ssl_read failed with error -0x7b00 form Debug Logs.

### Steps to reproduce.

1. The root cert used as example.
2. SERVER_USES_STARTSSL was enable same as example.
3. TLS 1.3 and others setting was configurated as follows:
CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
CONFIG_MBEDTLS_SSL_TLS1_3_COMPATIBILITY_MODE=y
CONFIG_MBEDTLS_SSL_TLS1_3_KEXM_PSK=y
CONFIG_MBEDTLS_SSL_TLS1_3_KEXM_EPHEMERAL=y
CONFIG_MBEDTLS_SSL_TLS1_3_KEXM_PSK_EPHEMERAL=y

CONFIG_MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH=y
CONFIG_MBEDTLS_X509_TRUSTED_CERT_CALLBACK=y
CONFIG_MBEDTLS_SSL_CONTEXT_SERIALIZATION=y
CONFIG_MBEDTLS_SSL_KEEP_PEER_CERTIFICATE=y
CONFIG_MBEDTLS_PKCS7_C=y
CONFIG_MBEDTLS_SSL_CID_PADDING_GRANULARITY=16

CONFIG_MBEDTLS_SSL_RENEGOTIATION=y
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=y
# CONFIG_MBEDTLS_SSL_PROTO_GMTSSL1_1 is not set
# CONFIG_MBEDTLS_SSL_PROTO_DTLS is not set
CONFIG_MBEDTLS_SSL_ALPN=y
CONFIG_MBEDTLS_CLIENT_SSL_SESSION_TICKETS=y
CONFIG_MBEDTLS_SERVER_SSL_SESSION_TICKETS=y

4. Built and run the example.
5. The connection and certificate verifying was OK, but at authentication after event MBEDTLS_SSL_TLS1_3_NEW_SESSION_TICKET a new session ticket was untreated. Is there any problem with my settings? or a software bug?
[... 381 lines cut ...]
I (12537) mbedtls: ssl_tls13_client.c:2921 => parse new session ticket

I (12547) mbedtls: ssl_msg.c:4134 => read record

I (12557) mbedtls: ssl_msg.c:4202 reuse previously read message

I (12557) mbedtls: ssl_msg.c:4206 <= read record

I (12577) mbedtls: ssl_tls13_client.c:2938 <= parse new session ticket

I (12577) mbedtls: ssl_tls.c:3942 <= handshake

W (12577) mbedtls: ssl_msg.c:5730 mbedtls_ssl_handshake() returned -31488 (-0x7b00)

E (12587) smtp_example: mbedtls_ssl_read failed with error -0x7b00
I (12597) mbedtls: ssl_tls.c:4841 => free

I (12597) mbedtls: ssl_tls.c:4903 <= free

E (12607) smtp_example: Last error was: -0x7b00 - SSL - * Received NewSessionTicket Post Handshake Message. This error code is experimental and may b
```

### More Information.

I also tried to run the example in SSL/TLS mode and got the same result before authentication.

mahavirj, 2024-04-02:
@zhichunlee 

Have you made any changes of your own in this example? If yes, can you please share the patch here?

Regarding the error you are facing, please try following change once:

```
diff --git examples/protocols/smtp_client/main/smtp_client_example_main.c examples/protocols/smtp_client/main/smtp_client_example_main.c
index f271c03db0..e02cc12d4d 100644
--- examples/protocols/smtp_client/main/smtp_client_example_main.c
+++ examples/protocols/smtp_client/main/smtp_client_example_main.c
@@ -28,6 +28,7 @@
 #include "mbedtls/error.h"
 #include <mbedtls/base64.h>
 #include <sys/param.h>
+#include "sdkconfig.h"
 
 
 /* Constants that are configurable in menuconfig */
@@ -144,6 +145,18 @@ static int write_ssl_and_get_response(mbedtls_ssl_context *ssl, unsigned char *b
         len = DATA_SIZE - 1;
         memset(data, 0, DATA_SIZE);
         ret = mbedtls_ssl_read(ssl, data, len);
+#if CONFIG_MBEDTLS_CLIENT_SSL_SESSION_TICKETS
+        // If a post-handshake message is received, connection state is changed to `MBEDTLS_SSL     _TLS1_3_NEW_SESSION_TICKET`
+        // Call mbedtls_ssl_read() till state is `MBEDTLS_SSL_TLS1_3_NEW_SESSION_TICKET` or ret     urn code is `MBEDTLS_ERR_SSL_RECEIVED_NEW_SESSION_TICKET`
+        // to process session tickets in TLS 1.3 connection
+        if (mbedtls_ssl_get_version_number(ssl) == MBEDTLS_SSL_VERSION_TLS1_3) {
+            while (ret == MBEDTLS_ERR_SSL_RECEIVED_NEW_SESSION_TICKET || ssl->MBEDTLS_PRIVATE(state) == MBEDTLS_SSL_TLS1_3_NEW_SESSION_TICKET) {
+                ESP_LOGD(TAG, "got session ticket in TLS 1.3 connection, retry read");
+                ret = mbedtls_ssl_read(ssl, data, len);
+            }
+        }
+#endif // CONFIG_MBEDTLS_CLIENT_SSL_SESSION_TICKETS
+
 
         if (ret == MBEDTLS_ERR_SSL_WANT_READ || ret == MBEDTLS_ERR_SSL_WANT_WRITE) {
             continue;
```

zhichunlee, 2024-04-03:
Hi mahavirj
Thank you for your confirmation. Based on previous issue#13410, I added psa_crypto_init() before connecting to the mail server.
After adding the above processing, the problem has been resolved and the email server has been successfully connected.
But a new error occurred while sending the email Content as follows.
I (5801) smtp_example: Connected.
I (5801) smtp_example: Performing the SSL/TLS handshake...
I (6601) smtp_example: Verifying peer X.509 certificate...
I (6601) smtp_example: Certificate verified.
I (6601) smtp_example: Cipher suite is TLS1-3-AES-256-GCM-SHA384
I (6611) smtp_example: handshake OK, get respone... 
I (7041) smtp_example: Writing EHLO to server...
I (7101) smtp_example: Authentication...
I (7101) smtp_example: Write AUTH LOGIN
I (7161) smtp_example: Write USER NAME
I (7231) smtp_example: Write PASSWORD
I (7441) smtp_example: Write MAIL FROM
I (7581) smtp_example: Write RCPT
I (7681) smtp_example: Write DATA
I (7951) smtp_example: Write Content
E (8231) smtp_example: Last error was: -0xfffffdd6 - UNKNOWN ERROR CODE (0200) : BASE64 - Output buffer too small

Why did TLS1.3 occur when TLS1.2 did not have this error? Increasing base64_buffer size did not improve either.

Thinks.