espressif/esp-idf #13496: WiFi scan doesn't find any APs after ESP-IDF update from 4.4.2 to 5.2.1

gniemirowski-invis, 2024-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Scan finds available networks.

### What is the actual behavior?

Scan doesn't find any networks.

### Steps to reproduce.

ESP_ERROR_CHECK(esp_wifi_scan_start(NULL, false));

event handler:
            case WIFI_EVENT_SCAN_DONE:
            {
                wifi_event_sta_scan_done_t* event = (wifi_event_sta_scan_done_t*) event_data;
                ESP_LOGI(TAG, "scan finished, status: %s", event->status ? "failure" : "success");
                uint16_t number = SCAN_LIST_SIZE;
                ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));
                ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));
                printf("scanning finished, found %u\n", ap_count);
                for (int i = 0; i < ap_count; i++) {
                    printf("%s\n", ap_info[i].ssid);
                }
                scanInProgress = false;
                break;
            }

### Debug Logs.

```plain
On IDF 5.2.1
I (42409) wifi: scan finished, status: success
scanning finished, found 0

On IDF 4.4.2
I (31527) wifi: scan finished, status: success
scanning finished, found 13
intermarium_n
intermarium
UPC247576703
Orange_Swiatlowod_84B0
DIRECT-6b-HP M15 LaserJet
UPCF8E1FE7
UPC8072952
Marcel2,4
Tuta
Penthouse
[air purifier]_E30AJT1003085R
PLAY_Swiatlowodowy_2FAF
UPC247720463
```

shreyassheth1, 2024-04-02:
Hey @gniemirowski-invis 
The sequence of the APIs should be to get the number of AP(get_ap_num) first and then deciding upon how many AP records you want to fetch call get_ap_records.
case WIFI_EVENT_SCAN_DONE:
{
wifi_event_sta_scan_done_t* event = (wifi_event_sta_scan_done_t*) event_data;
ESP_LOGI(TAG, "scan finished, status: %s", event->status ? "failure" : "success");
uint16_t number = SCAN_LIST_SIZE;
`ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));`
`ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&number, ap_info));`
printf("scanning finished, found %u\n", ap_count);
for (int i = 0; i < ap_count; i++) {
printf("%s\n", ap_info[i].ssid);
}
so the code should look like the above.
Please note that the AP records in the Wi-Fi driver will be freed once [esp_wifi_scan_get_ap_records()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv428esp_wifi_scan_get_ap_recordsP8uint16_tP16wifi_ap_record_t) is called.
Hence calling `esp_wifi_scan_get_ap_num(&ap_count)` after calling `esp_wifi_scan_get_ap_records(&number, ap_info)` will make ap_count as 0

AxelLin, 2024-04-02:
> Please note that the AP records in the Wi-Fi driver will be freed once [...]

So why does it work in 4.4.2?

shreyassheth1, 2024-04-04:
Hey @AxelLin this was changed in commit [b495f543](https://github.com/espressif/esp-idf/commit/b495f5432931b6394d527faffe67355ccec5332c)

grzegorzniemirowski, 2024-04-04:
@shreyassheth1 Thank you for reply. It's working now.

Sherry616, 2024-04-08:
Thanks for reporting, we will close this issue. Feel free to reopen if have more updates.

closed 2024-04-08 by Sherry616

cbicari, 2026-02-18:
this also worked for me. had the same issue. went from this : 

functional in ESP v4 :
```
esp_err_t err = esp_wifi_scan_start(NULL, true);
esp_err_t error = (esp_wifi_scan_get_ap_records(&number, ap_info));
esp_err_t mistake = esp_wifi_scan_get_ap_num(&ap_count);
```

in ESP v5 :
```
esp_err_t err = esp_wifi_scan_start(NULL, true);
esp_err_t mistake = esp_wifi_scan_get_ap_num(&ap_count);
esp_err_t error = (esp_wifi_scan_get_ap_records(&number, ap_info));
```