espressif/esp-idf #13497: simple_ota_example fails after upgrade to idf v5.1.2 from v 5.1.1

ghoti57, 2024-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

esp32-s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board based on esp32-s3 N16R8

### Power Supply used.

USB

### What is the expected behavior?

Perform OTA update from github

### What is the actual behavior?

Fails with following output
```
`I (7475) wifi:<ba-add>idx:1 (ifx:0, 9c:a2:f4:43:a2:2c), tid:0, ssn:0, winSize:64
I (8445) esp-x509-crt-bundle: Certificate validated
I (9685) esp-x509-crt-bundle: Certificate validated
W (10425) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (10555) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (10675) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (10795) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (10915) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (11045) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (11165) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (11285) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (11415) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
W (11535) HTTP_CLIENT: This request requires authentication, but does not provide header information for that
E (11655) esp_https_ota: Reached max_authorization_retries (401)
E (11655) esp_https_ota: Failed to establish HTTP connection
E (11665) simple_ota_example: Firmware upgrade failed

````

### Steps to reproduce.

1. Download simple_ota_example 
2. edit simple_ota_example.c to increase buffer_size_tx
```
    esp_http_client_config_t config = {
        .url = CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL,
#ifdef CONFIG_EXAMPLE_USE_CERT_BUNDLE
        .crt_bundle_attach = esp_crt_bundle_attach,
#else
        .cert_pem = (char *)server_cert_pem_start,
#endif /* CONFIG_EXAMPLE_USE_CERT_BUNDLE */
        .event_handler = _http_event_handler,
        .keep_alive_enable = true,
#ifdef CONFIG_EXAMPLE_FIRMWARE_UPGRADE_BIND_IF
        .if_name = &ifr,
#endif
		.buffer_size_tx=2048
    };
```
3. Edit sdkconfig.defaults
```
# Minimum FLASH size
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"

# Use Built in USB CDC Serial
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
CONFIG_ESP_CONSOLE_UART_NUM=-1

# Certificate bundle configuration
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y
```
4.  `idf.py set-target esp32-s3`

5.  `idf.py menuconfig` to configure wifi

6. using idf 5.1.1
`idf.py fullclean build -p COMxx flash monitor`

hmalpani, 2024-04-01:
Hello @ghoti57
Can you try reverting the commit: 9525da090800d84e9a4eaf801cf44b940622ed78 and let me know if that helps you resolve the issue.
Use command `git revert 9525da090800d84e9a4eaf801cf44b940622ed78`

ghoti57, 2024-04-01:
Hi,  reverting that commit allows the OTA to work in 5.1.2

How do I progress to 5.2.1 which exhibits the same problem?

hmalpani, 2024-04-01:
Hello @ghoti57 
We have fixed the issue in release/v5.2 as well. The fix should be available in GitHub soon. Until then, you can try to revert 04ac8e43dbaf37e42452fef034606ae3f04e644b on v5.2.1.

ghoti57, 2024-04-02:
Reverting that commit also resolves the problem for me in v5.2.1

Do you know when the fix is likely to appear in the release branches?

hmalpani, 2024-04-02:
We have backported the fix to release branches as well. The fix should be available soon.

closed 2024-04-02 by ghoti57