espressif/esp-idf #13499: UART get/set sclk functions

llamasoft, 2024-03-29:
### Is your feature request related to a problem?

It's possible to individually get/set the struct values passed into `uart_param_config` except for the `source_clk` element.  For example, `data_bits`, `parity`, and `stop_bits` can be fetched and updated with the functions mentioned in the [Multiple Steps](https://docs.espressif.com/projects/esp-idf/en/v5.2.1/esp32c3/api-reference/peripherals/uart.html#multiple-steps) portion of the documentation.

However, there are no functions like `uart_set_sclk` or `uart_get_sclk` to get or set the UART clock source.

### Describe the solution you'd like.

I propose adding the following methods:

- `esp_err_t uart_get_sclk(uart_port_t uart_num, uart_sclk_t *sclk)`
- `esp_err_t uart_set_sclk(uart_port_t uart_num, uart_sclk_t sclk)`

### Describe alternatives you've considered.

It's possible to get the UART sclk using the HAL function `uart_hal_get_sclk` but it requires a `uart_hal_context_t` struct.  Unfortunately, those are statically defined in [uart.c](https://github.com/espressif/esp-idf/blob/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60/components/esp_driver_uart/src/uart.c#L145-L151) with no way to access them.  The only alternative is to use the `UART_LL_GET_HW` and `uart_ll_get_sclk` low-level functions instead.

### Additional context.

Some clock sources like `UART_SCLK_RTC` require [additional steps](https://github.com/espressif/esp-idf/blob/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60/components/esp_driver_uart/src/uart.c#L800-L804) when enabling or disabling them.  Moving this logic into `uart_set_sclk` would mean less work when dealing with low-level UART functions.

For context, I'm currently working on creating a UART autobaud library.  One of the main challenges is that the `uart_ll_get_*_pulse_cnt` functions return values in clock cycle units.  It's not possible to use this into a baud rate calculation without knowing the UART source clock's frequency.  While this can be done with low-level functions, I'd prefer to use higher-level functions to ensure portability and prevent upgrade issues.  For example, upgrade issues like the [return type of `uart_ll_get_sclk`](https://github.com/espressif/esp-idf/blame/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60/components/hal/esp32c3/include/hal/uart_ll.h#L192) changing from `uart_sclk_t` to `soc_module_clk_t`. 😅

songruo, 2024-04-10:
Hi @llamasoft,

According to what you are trying to do, I think it is better if we offer an API that can directly know the time passed between two pulses. How do you think?

llamasoft, 2024-05-13:
I was mostly pointing out how all settings controlled by `uart_param_config` have corresponding `get` and `set` methods _except_ for the `source_clk` member.  It means that the only way to set the value is with `uart_param_config` which will necessarily overwrite the other settings.  Once set, there's no way to retrieve the value without using low-level functions.

(The use for autobaud is tangential.  Autobaud is going to require low-level functions anyways so I don't mind using other low-level functions to get `source_clk`.)