espressif/esp-idf #13502: MQTT接收数据失败

CYCHEN97, 2024-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.6-dirty

### Espressif SoC revision.

esp32-s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

通过巴法云平台，接入小爱同学，实现语音控制

### What is the actual behavior?

接收不到数据

### Steps to reproduce.

https://blog.csdn.net/lum250/article/details/128667774?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522171172294716800215065796%2522%252C%2522scm%2522%253A%252220140713.130102334.pc%255Fall.%2522%257D&request_id=171172294716800215065796&biz_id=0&utm_medium=distribute.pc_search_result.none-task-blog-2~all~first_rank_ecpm_v1~rank_v31_ecpm-2-128667774-null-null.142^v100^pc_search_result_base6&utm_term=esp32%20idf%E6%8E%A5%E5%85%A5%E5%B0%8F%E7%88%B1%E5%90%8C%E5%AD%A6&spm=1018.2226.3001.4187
按这个链接的步骤来的，使用idf4.4.6、idf5.1.1都是一样的问题

### Debug Logs.

```plain
[10:45:39.437]收←◆[0;32mI (924017) MQTT_EXAMPLE: Other event id:7[0m

[10:45:39.476]收←◆[0;32mI (924067) MQTT_EXAMPLE: MQTT_EVENT_CONNECTED[0m
[0;32mI (924067) MQTT_EXAMPLE: sent subscribe successful, msg_id=13567[0m
[0;31mE (924097) MQTT_CLIENT: Failed to acquire suback data[0m
[0;31mE (924097) MQTT_CLIENT: Failed to deliver suback message id=13567[0m
[0;32mI (924097) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED[0m

[10:45:49.527]收←◆[0;32mI (934107) MQTT_EXAMPLE: Other event id:7[0m

[10:45:49.584]收←◆[0;32mI (934157) MQTT_EXAMPLE: MQTT_EVENT_CONNECTED[0m
[0;32mI (934167) MQTT_EXAMPLE: sent subscribe successful, msg_id=4633[0m
[0;31mE (934187) MQTT_CLIENT: Failed to acquire suback data[0m
[0;31mE (934187) MQTT_CLIENT: Failed to deliver suback message id=4633[0m
[0;32mI (934187) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED[0m
```

CYCHEN97, 2024-03-30:
@ESP-YJM 大佬，帮忙看看这个问题，谢谢

ESP-YJM, 2024-04-01:
@CYCHEN97 你这个问题应该是服务器返回的 sub ack 包里不带 return code 导致的。目前 IDF 不支持不带 return code 的情况。如果服务器无法修改，可以使用如下修改方案。将下方的函数替换原来 IDF 里的同名函数。
[image]
```
char *mqtt_get_suback_data(uint8_t *buffer, size_t *length)
{
    // SUBACK payload length = total length - (fixed header (2 bytes) + variable header (2 bytes))
    // This requires the remaining length to be encoded in 1 byte.
    if (*length > 4) {
        *length -= 4;
        return (char *)(buffer + 4);
    }
    *length = 1;
    buffer[3] = 1;
    return (char *)(buffer + 3);
}

```

euripedesrocha, 2024-04-05:
@CYCHEN97 This looks like an issue in the broker side, could you provide something like a wireshark capture, so I can try to reproduce the issue?

euripedesrocha, 2024-05-23:
Closing since I couldn't reproduce. Feel free to reopen, adding the information to reproduce the issue.

closed 2024-05-23 by euripedesrocha