espressif/esp-idf #13507: Using QSPI for in-package PSRAM

hopkinskong, 2024-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

I am currently using ESP32-S3R8V which contains an in-package 8MB Octal SPI PSRAM. Since I wish to free up 4 GPIOs for other uses, is it possible to drive the in-package PSRAM with QSPI mode? (My Flash is also in QSPI mode.)

If it is not possible in the boot stage / bootloader, can I do that in application after my ESP32S3 has boot up? 

Thanks.

andylinpersonal, 2024-03-31:
The in-package PSRAM in S3R8(V) from AP Memory does not support any mode narrower than 8-line at all. Details can be seen in OPI PSRAM’s datasheet.

hopkinskong, 2024-04-01:
Thanks. How about using the PSRAM as a temporary storage only? For example, I will fill the internal RAM first, then dump the data onto the PSRAM, then I disable the PSRAM via some ESP-IDF API (or making sure the PSRAM's CS will never go LOW), then fill the internal RAM again, re-enable the PSRAM and copy data from internal RAM to PSRAM, and so-on?

nopnop2002, 2024-04-03:
>then I disable the PSRAM via some ESP-IDF API
>re-enable the PSRAM

I'm also looking for an API that works like this, but I can't find it.