espressif/esp-idf #13508: SoftAP Not Detected by Mobile Devices on ESP32-S3, Works Fine on ESP32

periclesbgf, 2024-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

**Description:**

I initiated the SoftAP example provided with the ESP-IDF and tested it on two devices: an ESP32 and an ESP32-S3. On the ESP32, the SoftAP operates as expected: the SSID is visible on my mobile device, and I am able to connect without any issues. However, when replicating the test in the exact same manner on the ESP32-S3, the SSID of the SoftAP does not appear in the list of available networks on my mobile device, preventing any attempt to connect.

**Environment:**

Device Model: ESP32-S3 vs. ESP32
ESP-IDF Version: 5.3 (master)
SoftAP Settings: see the logs
Development Tools: ESP-IDF in vscode

**Steps to Reproduce:**

Load the SoftAP example onto the ESP32-S3 using the example's default settings.
Attempt to view the SoftAP's SSID on a mobile device.
Repeat steps 1 and 2 with an ESP32 device, noting that in this case, the SSID is visible and connection is possible.

**Expected Result:**

The SoftAP's SSID should be visible and accessible for connection on both devices, ESP32 and ESP32-S3.

**Actual Result:**

The SoftAP's SSID is visible and connectable on the ESP32 but does not appear in the list of available networks on the ESP32-S3.

`/*  WiFi softAP Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_mac.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_WIFI_CHANNEL   CONFIG_ESP_WIFI_CHANNEL
#define EXAMPLE_MAX_STA_CONN       CONFIG_ESP_MAX_STA_CONN

static const char *TAG = "wifi softAP";

static void wifi_event_handler(void* arg, esp_event_base_t event_base,
                                    int32_t event_id, void* event_data)
{
[... 158 lines cut ...]
I (489) wifi:Init dynamic tx buffer num: 32
I (489) wifi:Init static tx FG buffer num: 2
I (499) wifi:Init static rx buffer size: 1600
I (499) wifi:Init static rx buffer num: 10
I (499) wifi:Init dynamic rx buffer num: 32
I (509) wifi_init: rx ba win: 6
I (509) wifi_init: accept mbox: 6
I (509) wifi_init: tcpip mbox: 32
I (519) wifi_init: udp mbox: 6
I (519) wifi_init: tcp mbox: 6
I (529) wifi_init: tcp tx win: 5760
I (529) wifi_init: tcp rx win: 5760
I (529) wifi_init: tcp mss: 1440
I (539) wifi_init: WiFi IRAM OP enabled
I (539) wifi_init: WiFi RX IRAM OP enabled
I (549) phy_init: phy_version 640,cd64a1a,Jan 24 2024,17:28:12
I (589) wifi:mode : softAP (dc:da:0c:2a:13:29)
I (649) wifi:Total power save buffer number: 16
I (649) wifi:Init max length of beacon: 752/752
I (649) wifi:Init max length of beacon: 752/752
I (649) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
I (649) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (10669) wifi softAP: SoftAP SSID:myssid Password:mypassword Channel:1
I (20669) wifi softAP: SoftAP SSID:myssid Password:mypassword Channel:1
I (30669) wifi softAP: SoftAP SSID:myssid Password:mypassword Channel:1`

nishanth-radja, 2024-04-04:
@periclesbgf can we have a sniffer capture on channel 1,to see if AP is beaconing.

AxelLin, 2024-04-23:
@periclesbgf Any update for comment https://github.com/espressif/esp-idf/issues/13508#issuecomment-2036189516 ?

periclesbgf, 2024-04-24:
I just got another S3 board and tested in the same conditions, and it worked. I think maybe it is a hardware problem.

AxelLin, 2024-05-08:
@nishanth-radja 

I hit this issue (no AP found) when test esp32-camera on esp32s3-wroom-1 using esp-idf v5.2.1-485-g9b3ae86417 .
The esp32s3 is running as an AP, but my laptop and mobile phone cannot find the esp32s3's AP (SSID:ICAM_49_40)

[esp32s3_no_ap_found2.pcapng.txt](https://github.com/espressif/esp-idf/files/15244438/esp32s3_no_ap_found2.pcapng.txt)
[esp32s3_no_ap_found3.pcapng.txt](https://github.com/espressif/esp-idf/files/15244439/esp32s3_no_ap_found3.pcapng.txt)

AxelLin, 2024-05-08:
@nishanth-radja 
Also attached the log when the esp32s3's AP (SSID:ICAM_49_40) is found for comparison.
It seems if I remove the camera then I have a much better chance to find the esp32s3's AP.

[esp32s-ap1.pcapng.txt](https://github.com/espressif/esp-idf/files/15245661/esp32s-ap1.pcapng.txt)
[esp32s-ap2.pcapng.txt](https://github.com/espressif/esp-idf/files/15245663/esp32s-ap2.pcapng.txt)

AxelLin, 2024-05-08:
Below is the log when I found the AP and try to join it, it seems join success but then disassoc immediately.

````

V (37544) wifi:updated connection retries to 0
I (56845) wifi:removing station <20:1e:88:5e:2b:3f> after unsuccessful auth/assoc, AID = 0
W (56846) wifi:rm mis
I (56846) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<0,0>, prof:6
V (73768) wifi:updated connection retries to 0
D (82379) wifi:ap recv assoc/reassoc request
I (82380) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<0,0>, prof:6
D (82380) wifi:bss=0x3c1ae9c4, resp=16 join
I (82382) wifi:station: 20:1e:88:5e:2b:3f join, AID=1, bgn, 20
D (82388) wpa:   New STA
D (82391) wpa: add sm, index=0 bitmap=1
D (82394) wpa: init wpa sm=0x3fca8a94
D (82398) wpa: rsn_ie - hexdump(len=28):
D (82402) wpa: 30 1a 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 
D (82408) wpa: 00 0f ac 02 80 00 00 00 00 0f ac 06 
D (82413) wifi:join success bss=0x3c1ae9c4, ap send assoc response
D (82419) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state INITIALIZE

D (82426) wpa: wpa_auth_set_key : key idx  0 alg 0 vlan_id 0 key_len 0 key
D (82433) wpa: ELOOP: wpa_remove_ptk:1252 called to remove timer handler=0x4204ab80, removed count=0
0x4204ab80: wpa_rekey_ptk at /home/axel/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:262

D (82442) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK_GROUP entering state IDLE

D (82449) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state AUTHENTICATION

D (82456) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state AUTHENTICATION2

D (82463) wpa: WPA: Re-initialize GMK/Counter on first station
D (82469) wpa: GMK - hexdump(len=32):
D (82473) wpa: 3a 3c 8d 39 60 0c cc 4a 3d 6c 0a d9 ab a3 54 23 
D (82479) wpa: ee fe fc 55 46 3d 79 20 45 79 af 8d 42 86 04 12 
D (82486) wpa: Key Counter - hexdump(len=32):
D (82490) wpa: f6 e0 20 d5 5a c6 fe 13 29 1d 14 e2 a9 50 16 1b 
D (82495) wpa: 2e 50 11 8f 12 a3 ee 26 f1 38 d9 d6 49 ee 4d 92 
D (82502) wpa: GTK - hexdump(len=16):
D (82505) wpa: f0 85 56 bd d3 11 a4 af 75 98 69 e2 d4 68 83 c8 
D (82512) wpa: IGTK - hexdump(len=16):
D (82515) wpa: 76 ef 76 ad 9b 51 8e 38 1b a2 31 93 2a 7e 48 82 
D (82521) wpa: wpa_auth_set_key : key idx  1 alg 3 vlan_id 0 key_len 16 key
D (82528) wpa: wpa_auth_set_key : igtk idx 4
D (82532) wpa: WPA: Assign ANonce - hexdump(len=32):
D (82537) wpa: 63 03 8a df 8b ef f2 b5 8f b4 bb ba 8a f8 48 b6 
D (82543) wpa: 3c 92 42 09 04 b6 58 21 4b 83 fd 09 e4 99 eb 5d 
D (82549) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state INITPSK

D (82556) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state PTKSTART

D (82562) wpa: wpa_auth=0x3fcd03bc sm=0x3fca8a94  kdersc=0x0 kde=0x0 nounce=0x3fca8ae4 kde_len=0 keyidx=0 encr=0 force=0
D (82574) wpa: WPA: Send EAPOL(version=2 secure=0 mic=0 ack=1 install=0 pairwise=1 kde_len=0 keyidx=0 encr=0)
D (82584) wpa: ELOOP: wpa_send_eapol:1212 called to remove timer handler=0x420491c8, removed count=0
0x420491c8: resend_eapol_handle at /home/axel/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:1184

D (82593) wpa: ELOOP: Added one timer from wpa_send_eapol:1213 to call 0x420491c8, current order=0
0x420491c8: resend_eapol_handle at /home/axel/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:1184

V (82602) wifi:tx process: alloc static tx buf, pb=0x0 l=113 hdr=0 data=113
D (82609) wifi:ap recv assoc/reassoc request
D (82612) wifi:Ignore immediate assoc req as processing other assoc req for 20:1e:88:5e:2b:3f
D (82621) wifi:ap recv assoc/reassoc request
D (82625) wifi:Ignore immediate assoc req as processing other assoc req for 20:1e:88:5e:2b:3f
D (83593) wpa: ELOOP: Running timer fn:0x420491c8 scheduled by wpa_send_eapol:1213 
0x420491c8: resend_eapol_handle at /home/axel/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:1184

D (83594) wpa: resend eapol1
D (83594) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state PTKSTART
[... 154 lines cut ...]
D (117692) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state DISCONNECT

D (117699) wpa: wpa_sta_disconnect STA 20:1e:88:5e:2b:3f
D (117704) wifi:Removing node for station 20:1e:88:5e:2b:3f
D (117710) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state DISCONNECTED

D (117717) wpa: WPA: 20:1e:88:5e:2b:3f WPA_PTK entering state INITIALIZE

D (117724) wpa: wpa_auth_set_key : key idx  0 alg 0 vlan_id 0 key_len 0 key
D (117731) wpa: ELOOP: wpa_remove_ptk:1252 called to remove timer handler=0x4204ab80, removed count=0
0x4204ab80: wpa_rekey_ptk at /home/axel/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:262

D (123733) wifi:ap recv deauth/disassoc from sta 20:1e:88:5e:2b:3f, reason:3
I (123733) wifi:station: 20:1e:88:5e:2b:3f leave, AID = 1, bss_flags is 33721442, bss:0x3c1ae9c4
I (123737) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<0,0>, prof:6
D (123743) wpa: ELOOP: wpa_auth_sta_deinit:508 called to remove timer handler=0x420491c8, removed count=0
0x420491c8: resend_eapol_handle at /home/axel/esp/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:1184

D (123753) wpa: del sm, index=0 bitmap=0
D (123757) wpa: wpa_free_sta_sm: free eapol=0x0

D (123772) wifi:no bcast bss, send deauth: reason not authed(6)
V (125150) wifi:updated connection retries to 0

````

AxelLin, 2024-05-14:
@nishanth-radja 

I found this symtom (ESP32S3 AP not visible) only hapeens at certain environment.
I'm wondering what can cause such issue as I test on the same ESP32S3 with the same laptop
and the esp2s3 is very closed to my laptop (Just put on the same desk).

Any feedback about the above sniffer log and debug print above?
Is there anything unusual in the no AP found log (https://github.com/espressif/esp-idf/issues/13508#issuecomment-2099804524) ?

nishanth-radja, 2024-05-15:
@AxelLin Looks like a RF related issue,  Even periclesbgf described it to be a RF related issue.

AxelLin, 2024-05-31:
> @AxelLin Looks like a RF related issue, Even periclesbgf described it to be a RF related issue.

This issue causes very bad user experience because it sometimes works sometimes does not work.
The status switches to "In Progress" so I'm wondering if that means you know how to address this issue?

FrancoisB-HEX, 2024-11-20:
My project is also crippled by this issue. I have a relatively simple project that sets up a Wifi AP, and in specific locations the AP is either not discovered at all my client devices (mobile phones/Windows PCs), or clients are unable to connect to the AP, or there is no IP throughput if the clients do get connected. In other locations it works almost perfectly (client devices can discover, connect, and have good IP throughput).

I am running `v5.3.1-395` on `ESP32-S3-WROOM-1-N16` modules

AxelLin, 2024-11-20:
@FrancoisB-HEX
Any chance to capture the sniffer log for analysis?
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/wireshark-user-guide.html

FrancoisB-HEX, 2024-11-20:
> @FrancoisB-HEX Any chance to capture the sniffer log for analysis? https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/wireshark-user-guide.html

Unfortunately my work machine's wifi interface does not support Monitor mode no. 
`Network monitor mode                        : Not supported`

FrancoisB-HEX, 2024-11-21:
> My project is also crippled by this issue. I have a relatively simple project that sets up a Wifi AP, and in specific locations the AP is either not discovered at all my client devices (mobile phones/Windows PCs), or clients are unable to connect to the AP, or there is no IP throughput if the clients do get connected. In other locations it works almost perfectly (client devices can discover, connect, and have good IP throughput).
> 
> I am running ESP-IDF `v5.3.1-395` on `ESP32-S3-WROOM-1-N16` modules

Spending a few hours on this I was able to isolate the issue on my board and firmware: I have a 40MHz CLOCK output on `GPIO_NUM_21` to a peripheral that is driven by the LEDC driver using `LEDC_TIMER_0`, `LEDC_CHANNEL_0`, `LEDC_USE_XTAL_CLK`. If I disable this LEDC clock, the Wifi AP works flawlessly, but with it enabled I sporadically have all of the issues mentioned. 
The PCB design means it is very unlikely to be an EMC issue (the CLOCK line runs in between 2 ground layers with 2 vias on route to the peripheral with a short and straght track, and setting up another identical board with the clock signal being generated physically next to the one that has perfectly stable Wifi AP (and no clock) has no influence on it). 

Can anyone explain why setting up a LEDC PWM output as mentioned would influence Wifi AP on ESP32-S3?

mattmarion, 2026-08-21:
I had the same issue on a board I just got.  I could get it to very sporadically show up at 15db for some reason and then noticed that if I touch the antenna with my finger, it's good to go.  Even better, I put a little piece of electric tape on it (cover antenna and wrap around the edge by it) and it's working perfectly.

https://www.waveshare.com/wiki/ESP32-S3-LCD-1.47B

That board.  The antenna is the little red thing between two of the feet on the "top" of the board (opposite the usbc)

So if you are just doing dev work on it and just need the antenna working, try putting some electric tape on it.