espressif/esp-idf #13509: I2C asynchronous transfers used when queue depth set to non zero value.

NeroProtagonist, 2024-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-234-gfcb76f0671

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Huzzah ESP32 Feather

### Power Supply used.

USB

### What is the expected behavior?

I expect I2C transactions to be executed synchronously unless a callback is registered, as per documentation:
```
Although I2C is a synchronous communication protocol, we also support asynchronous behavior by registering above callback. In this way, I2C APIs will be non-blocking interface. But note that on the same bus, only one device can adopt asynchronous operation.
```

### What is the actual behavior?

I2C transactions are performed asynchronously if trans_queue_depth is set to a non-zero value, even though no callback is registered.

This results in at least i2c_master_receive returning before the transaction has finished.

### Steps to reproduce.

1. Create master bus initialized with trans_queue_depth equal to positive value (4 in my case).
2. Do not register master callback.
3. Perform I2C read operation.
4. Observe asynchronous code path is taken and i2c_master_receive returns before transaction finishes.
...

### More Information.

In my case I was reading data into a buffer residing on the stack and this was obviously disastrous when my stack frame popped and the I2C ISR was then writing to a location of another variable in another stack frame.

Having had a quick peek through other I2C issues for the new driver, I wouldn't be suprised if some of them can be attributed to this bug (e.g. stack corruption/crashing, apparent timing issues).

mythbuster5, 2025-01-13:
i think some of the issue has been solved. Actually sync or async transation is controlled by queue size in this case. If queue size is 0 then you don't need to register a callback

NeroProtagonist, 2025-01-13:
The documentation states:

> [i2c_master_bus_config_t::trans_queue_depth](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv4N23i2c_master_bus_config_t17trans_queue_depthE) sets the depth of internal transfer queue. Only valid in asynchronous transaction.

That sounds to me that it doesn't matter what it is set to in synchronous mode, which is not actually true.

Documentation also says:

> Although I2C is a synchronous communication protocol, asynchronous behavior is supported by registering above callbacks. In this way, I2C APIs will be non-blocking interface. 

That sounds to me like synchronous operation will occur if callback is not set, but that is not true since it is, as you say, controlled by queue depth parameter and not callback registration.

It would be great if the documentation for these two parts could be updated., e.g.

"Queue depth sets the depth of internal transfer queue to enable asynchronous operation. This must be set to 0 to enable synchronous operation".
"Although I2C is a synchronous communication protocol, asynchronous behavior is supported by setting a non-zero queue depth and registering above callbacks. In this way, I2C APIs will be non-blocking interface. "

Although I personally think it is better to control sync/async behaviour by registering/not registering a callback, and ignore the queue depth when synchronous operation is used.

DushkaF, 2025-04-05:
@mythbuster5 Hi! I encountered the same problem. Can you explain, is it a bug or is it intended? I just need to first transfer data in synchronous mode, and then switch to asynchronous. Should I change the queue size for this or just assign a callback function?