espressif/esp-idf #13510: Driver RMT

gabrielgvicente, 2024-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2-dirty

### Espressif SoC revision.

ESP32-WROOM-32E

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Personalized board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I need to read a remote control and obtain signals above 250us.

### What is the actual behavior?

 but I am not successful when applying the RMT receiver filter, I receive signals below 50uS.

### Steps to reproduce.

    rmt_receive_config_t receive_config = {
    .signal_range_min_ns = 250000,
    .signal_range_max_ns = 1250000
};
rmt_rx_channel_config_t rx_channel_cfg = {
        // .clk_src = RMT_CLK_SRC_DEFAULT,
        .clk_src = RMT_CLK_SRC_REF_TICK,
        .resolution_hz = 1000000,
        .mem_block_symbols = 64,
        .gpio_num = rf_pin,
        .flags.invert_in = true,
        .flags.with_dma = false,
    };

suda-morris, 2024-04-01:
@gabrielgvicente Thanks for the report, I can confirm the issue. 🐛 For now, I would suspect the hardware maybe still using the APB (which is 80MHz) as the clock source for doing the filter, thus even you set 250us, the real bound may be 250/80 = 3.125 us, it only filter out 3us pulses...

We will need some time to get some feedback from the digital team.

gabrielgvicente, 2024-04-01:
Hello Suda-Morris, thank you for your reply, I will wait for the team's response. This filter is very important for the functioning of my application

gabrielgvicente, 2024-04-16:
Even so, the maximum filter I can apply will be 3uS?

closed 2024-04-17 by espressif-bot

dzungpv, 2025-04-28:
@suda-morris this is still be true at current IDF version?