espressif/esp-idf #13511: GCOV (Coverage) different exceptions on esp32 c3

m0rt4l1n, 2024-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1 (a322e6bdad4b6675d4597fb2722eea2851ba88cb)

### Espressif SoC revision.

ESP32-C3 (revision v0.4)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

When enabling gcov using
```
CONFIG_APPTRACE_DEST_JTAG=y
CONFIG_APPTRACE_GCOV_ENABLE=y
```
in sdkconfig, adding `--coverage` flag to some files and adding the `app_trace` component the program runs normally.

(This already fails)

After that, connecting openocd, running `reset` and after running the program (in this case unit tests) `esp gcov` the coverage is dumped

### What is the actual behavior?

I am getting `Illegal Instruction`, `Breakpoint` or `Interrupt wdt timeout on CPU0` exceptions. (e.g. `Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.`)

Using idf.py's monitor the Illegal Instruction exceptions (which I am getting the most) it looks something like this
```
I (218) sleep: Configure to isolate all GPIO pins in sleep state
I (218) sleep: Enable automatic switching of GPIO sleep configuration
I (219) esp_apptrace: Apptrace initialized on CPU0. Tracing control block @ 0x3fc99c38.
I (219) esp_core_dump_uart: Init core dump to UART
I (219) main_task: Started on CPU0
I (219) main_task: Calling app_main()
I (219) nvs: NVS Partition opened
./src/main.cpp:107:wifi_settings_test::test_init:PASS
./src/main.cpp:108:wifi_settings_test::test_add_wifi:PASS
./src/main.cpp:109:wifi_settings_test::test_remove_wifi:PASS
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420137cc  RA      : 0x40382e10  SP      : 0x3fc91b10  GP      : 0x3fc8dc00  
0x4038061e: esp_restart_noos at /Users/mark/esp/esp-idf5/components/esp_system/port/soc/esp32c3/system_internal.c:111

Stack dump detected
[... 279 lines cut ...]
.iram0.bss 0x4038ca00 0x0 RW  
.coredump.tasks.data 0x3fc9cb44 0x150 RW 
.coredump.tasks.data 0x3fc90830 0x910 RW 
.coredump.tasks.data 0x3fc9c970 0x1c0 RW 
.coredump.tasks.data 0x3fc9d79c 0x150 RW 
.coredump.tasks.data 0x3fc9d6f0 0xa0 RW 
.coredump.tasks.data 0x3fc9a854 0x150 RW 
.coredump.tasks.data 0x3fc9a740 0x100 RW 
.coredump.tasks.data 0x3fc9b9ac 0x150 RW 
.coredump.tasks.data 0x3fc9b8c0 0xe0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump checksum='ddede511'
I (1452) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...
```
```

### More Information.

Been trying a lot of different thing to try and fix this but no success until now

Note that I am using C++

Lapshin, 2024-04-01:
@m0rt4l1n , sorry for the issue you faced. Could you please confirm that this patch helps to fix it?

```diff
diff --git a/components/app_trace/gcov/gcov_rtio.c b/components/app_trace/gcov/gcov_rtio.c
index 887440d780..fad4e59651 100644
--- a/components/app_trace/gcov/gcov_rtio.c
+++ b/components/app_trace/gcov/gcov_rtio.c
@@ -16,6 +16,7 @@
 #include "esp_freertos_hooks.h"
 #include "esp_private/dbg_stubs.h"
 #include "esp_ipc.h"
+#include "esp_attr.h"
 #include "hal/wdt_hal.h"
 #if CONFIG_IDF_TARGET_ESP32
 #include "esp32/rom/libc_stubs.h"
@@ -80,6 +81,7 @@ void gcov_create_task(void *arg)
                (void *)&s_gcov_task_running, configMAX_PRIORITIES - 1, NULL, 0);
 }
 
+static IRAM_ATTR
 void gcov_create_task_tick_hook(void)
 {
     extern esp_err_t esp_ipc_start_gcov_from_isr(uint32_t cpu_id, esp_ipc_func_t func, void* arg);
```

m0rt4l1n, 2024-04-01:
Thank you @Lapshin! This definitely fixes the IllegalInstruction.

I still sometimes get the `Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).` exceptions, though it seems like increasing this timeout to 1000 ms fixes this 99% of the time. (That 1% ist still a bit annoying but I can live with that)

I got the Breakpoint exception a few times while just running `idf.py clean set-target esp32c3 build flash monitor` but I can't really reproduce this behavior.
But I haven't got this Exception while using pytest and a bit of code to pull the coverage after each test, yet, so this is still success for me

Lapshin, 2024-04-01:
@m0rt4l1n , thank you for the feedback! The fix will be added soon for the all affected branches

Lapshin, 2024-04-01:
@m0rt4l1n, regarding `wdt timeout` exceptions you may consider setting `CONFIG_ESP_INT_WDT=n` and `CONFIG_ESP_TASK_WDT_EN=n` for disabling it in your tests. If you need to generate only code coverage. Otherwise, disabling this in a "release" firmware is going to be a bad idea.

Regarding the Breakpoint exceptions, I have no strict advices to say.\
Perhaps it was generated due to another exception (like `wdt timeout`) when openocd set the chip into debug mode. But something went wrong after...
In case you find out how to reproduce this issue, please open a new issue.
CC @erhankur

m0rt4l1n, 2024-04-01:
Thank you!

I just stumbled across the breakpoint exception when I just tried to run my tests in CI, looks like this:
[image]

Both addresses resolve to `/root/esp/esp-idf/components/riscv/include/riscv/semihosting.h:77`

I'll try to reproduce it with a smaller program but still thanks!

closed 2024-04-04 by espressif-bot

mickeyl, 2024-04-29:
@m0rt4l1n Congratulations, you seem to have a gcov setup for profiling in place. Judging from the few hits on github and the web in general it looks like hardly anyone is interested in that. Could you perhaps share some of the key bits to get this going?

All I want to do is to get a list of the hottest code paths and how long they took in order to give me some hard data for optimization...