espressif/esp-idf #13512: MQTT crt file crash

AlexMihai1804, 2024-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

Connect without any crash

### What is the actual behavior?

I'm trying to connect to MQTT. When I use .crt certificates, I get a crash. Without them, it works, but it's not secure. I've tried with HiveMQ and EMQX, and I get the same crash. I've used an ESP32 and an ESP32S3.

### Steps to reproduce.

```
#include "esp_log.h"
#include "mqtt_client.h"
#include "nvs_flash.h"

static const char *TAG = "MQTT";
esp_mqtt_client_handle_t client;
extern const uint8_t hivemq_crt_start[] asm("_binary_hivemq_crt_start");
extern const uint8_t hivemq_crt_end[] asm("_binary_hivemq_crt_end");
const size_t hivemq_crt_size = hivemq_crt_end - hivemq_crt_start;

static esp_err_t mqtt_event_handler_cb(esp_mqtt_event_handle_t event) {
    switch (event->event_id) {
        case MQTT_EVENT_CONNECTED:
            ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
            esp_mqtt_client_publish(client, "test", "This is a test message", 0, 1, 0);
            break;
        case MQTT_EVENT_DISCONNECTED:
            ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
            break;
        default:
            break;
    }
    return ESP_OK;
}

static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data) {
    esp_mqtt_event_handle_t event = (esp_mqtt_event_handle_t) event_data;
    mqtt_event_handler_cb(event);
}

void mqtt_app_start(void) {
    const esp_mqtt_client_config_t mqtt_cfg = {
            .broker = {
                    .address = {
                            .uri = "mqtts://9a8b0eedab19429098cf30728dc442f6.s1.eu.hivemq.cloud:8883",
                    },
                    .verification={
                            .certificate=(const char *) hivemq_crt_start
                    }
            },
            .credentials={
                    .username="USER",
                    .authentication={
                            .password="PASS"
                    }
            }
    };
    client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_register_event(client, MQTT_EVENT_ANY, mqtt_event_handler, client);
    esp_mqtt_client_start(client);
}
```
i connect to wifi and sync sntp in another file
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4038238d  PS      : 0x00060433  A0      : 0x803846a3  A1      : 0x3fcad100
A2      : 0x03411003  A3      : 0xffffffff  A4      : 0x0000cdcd  A5      : 0x00060423
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80382369  A9      : 0x3fcad100
A10     : 0x00060423  A11     : 0x00000000  A12     : 0x00060420  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x011c0000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x03411003  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x4038238a:0x3fcad100 0x403846a0:0x3fcad130 0x40376e8b:0x3fcad150 0x40376ed4:0x3fcad180 0x40386211:0x3fcad1a0 0x4205dfa8:0x3fcad1c0 0x4205f143:0x3fcad1e0 0x420634aa:0x3fcad210 0x4206e150:0x3fcad260 0x4206ecc2:0x3
fcad290 0x4205cad1:0x3fcad2b0
```

dvosully, 2024-04-01:
Text format certificates must be nul terminated.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mqtt.html#verification

Either append a nul to your CRT file, copy it into RAM and nul terminate it, or provide a der format certificate.

AlexMihai1804, 2024-04-01:
I tried adding this before the struct initialization
```
const size_t cert_size_with_null = hivemq_crt_size + 1;
    char *cert_buffer = (char*)malloc(cert_size_with_null);
    if (cert_buffer != NULL) {
        memcpy(cert_buffer, hivemq_crt_start, hivemq_crt_size);
        cert_buffer[hivemq_crt_size] = '\0';
    }
```
and I get the same crash.
I tried converting to der(I'm not sure if it's correct) and I still get a crash. I tried 1 week to get it working and no luck.

dvosully, 2024-04-06:
I don't have time to test your example at the moment sorry unless you can provide a minimal example, but I have certificate authentication on AWS working on idf 5.2.1. It seems you want to use mqtts:// with username/password authentication for HiveMQ - I had that working successfully a while ago and the current branch is still working for me for mqtts:// cert/key authentication. I can't see anything else obvious with your configuration.

Your backtrace isn't super helpful, if you use the idf.py monitor it should convert the backtrace to filenames, function names and lines which would make it much easier to troubleshoot.

euripedesrocha, 2024-05-06:
@AlexMihai1804 could you provide a reproducer? 
I would suggest you to take the SSL example and change the certificate used to verify that your certificate format is correct.

euripedesrocha, 2024-07-15:
Closing. Feel free to reopen with added information.

closed 2024-07-15 by euripedesrocha