espressif/esp-idf #13513: fgetc does not work on esp32s3 on ttyACM0

nopnop2002, 2024-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-210-gcf7b9bed4c

### Espressif SoC revision.

ESP32S3 Development board

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 Development board

### Power Supply used.

USB

### What is the expected behavior?

fgetc does work on esp32s3 on ttyACM0.

### What is the actual behavior?

fgetc does not work on esp32s3 on ttyACM0.

### Steps to reproduce.

Build this.
```
#include <stdio.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"

static const char *TAG = "MAIN";

void app_main()
{
    uint16_t c;
    while (1) {
        c = fgetc(stdin);
        if (c == 0xffff) {
            vTaskDelay(1);
            continue;
        }
        ESP_LOGI(TAG, "c=%x", c);
        vTaskDelay(1);
    }
}
```

### Debug Logs.

```plain
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
```
```

### More Information.

This may be related.
https://github.com/espressif/esp-idf/issues/9974

ESP-Marius, 2024-04-01:
@nopnop2002 did you use `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG`? As mentioned in `ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG` (the default setup) the seconday console cannot be used for input.

nopnop2002, 2024-04-01:
thank you for your reply.

I can now use fgets() by adding the following definition to sdkconfig.

```
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
```

However, no matter where I look in the menu, I cannot find the above item.

ESP-Marius, 2024-04-02:
It is located at `Component config → ESP System Settings → Channel for console output`.

If you have any issues finding kconfig options you can press `/` inside menuconfig to search for items.

closed 2024-04-02 by ESP-Marius