espressif/esp-idf #13514: [ble mesh] before publish callback event

chegewara, 2024-03-31:
### Additional context.

Hi,
im still beginner on the ble mesh field, but i am wondering. Is it possible to implement publish callback event which will be fired before publish so we could update model state just before publish?

In most cases its not important, because most models state is known and can be updated when state is changed.
I found 3 use cases where it could be useful to have such event:
- most obvious may be vendor model (optional)
- then location server model, assuming we are talking about global location and GPS
- another model may be time server model.

I am currently thinking about time server. In theory i could update its state and publish buffer every some time, like every 1 second or more often, but its not efficient and not accurate. With location model it is less important, because GPS location updates are less often, even if node is moving.

Custom model is just concept, and probably its state can update itself whenever its needed. 

Thanks

chegewara, 2024-04-02:
Yes, i know, and i am using it now. 
But, like i mentioned, it updates after publish, so i cant update state (current time) before publishing it.

There is few hacks i could do, like mentioned update time every some time (100ms) or so, but this still is just a hack.

It is nothing really important, just an idea.

Thanks

forx157, 2024-04-03:
@chegewara 
Ok, I see the feature you want, I'll make a note of it and discuss it with the teams. Thanks for the suggestion.