espressif/esp-idf #13515: Memory placed into .noinit section occupies space in flash/binary image

dvosully, 2024-04-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-188-ge759833c44

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expect that variables placed in .noinit RAM sections do not occupy space in the binary image

### What is the actual behavior?

When a new variable is placed in .noinit RAM the size of the binary image is increased by the size of the variable

### Steps to reproduce.

1. Build the hello_world example and run idf.py size. Observe the size of the sections.
2. Add a large data variable to hello_world_main.c and ensure it is linked eg.
uint8_t data[50*1024];
void app_main(void) { data[1] = 5; [...]
3. Build and observe the size of the sections - it is unchanged. This is expected - BSS does not occupy space in flash.
4. Change the type of the large data variable to .noinit
__NOINIT_ATTR uint8_t data[50*1024];
5. Build and observe the size of the sections - the total image size has increased by 50kB

### Build or installation Logs.

```plain
C:\Espressif\development\hello_world>idf.py size-components
Executing action: size-components
Running ninja in directory C:\Espressif\development\hello_world\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Espressif\development\hello_world\build\esp-idf\espto.../partition-table.bin C:/Espressif/development/hello_world/build/hello_world.bin"
hello_world.bin binary size 0x2b620 bytes. Smallest app partition is 0x100000 bytes. 0xd49e0 bytes (83%) free.
[1/1] cmd.exe /C "cd /D C:\Espressif\development\hello_world\build\bootloader\es...der 0x1000 C:/Espressif/development/hello_world/build/bootloader/bootloader.bin"
Bootloader binary size 0x6860 bytes. 0x7a0 bytes (7%) free.
Running ninja in directory C:\Espressif\development\hello_world\build
Executing "ninja size-components"...
Total sizes:
Used static DRAM:   62156 bytes ( 118580 remain, 34.4% used)
      .data size:    8716 bytes
      .bss  size:    2240 bytes
 DRAM other size:   51200 bytes
Used static IRAM:   51106 bytes (  79966 remain, 39.0% used)
      .text size:   50079 bytes
   .vectors size:    1027 bytes
Used Flash size :  117759 bytes
           .text:   79339 bytes
         .rodata:   38164 bytes
Total image size:  228781 bytes (.bin may be padded larger)
Per-archive contributions to ELF file:
            Archive File DRAM .data & 0.bss .noinit IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .appdesc flash_total
               libmain.a          0       0   51200           0           0        51200         304         0          0       51504
                  libc.a        260     316       0           0           0          576       42290      3194          0       45744
         libesp_system.a        480     283       0        3975           0         4738        7609     25999          0       38063
           libfreertos.a       3100     741       0       13778           0        17619         372       982          0       18232
     libesp_hw_support.a         80      80       0        5150           0         5310        6209       867          0       12306
                libhal.a       1995       8       0        5773           0         7776        2018        98          0        9884
               libheap.a          4       8       0        5328           0         5340        2957      1544          0        9833
          libspi_flash.a       1112      24       0        6904           0         8040         735       401          0        9152
                libvfs.a        308      72       0           0           0          380        6743       490          0        7541
             libesp_mm.a         12     124       0         812           0          948        2631       214          0        3669
             libxtensa.a       1044       0       0        1789         424         3257         119        36          0        3412
             libnewlib.a        152     200       0        1418           0         1770        1190        95          0        2855
             libdriver.a         56      12       0           0           0           68        2119       256          0        2431
          libesp_timer.a         32      24       0         936           0          992        1199        80          0        2247
 libbootloader_support.a          0       0       0        1793           0         1793          94        40          0        1927
         libesp_common.a          0       0       0           0           0            0          51      1742          0        1793
                libsoc.a          0       0       0          37           0           37           0      1444          0        1481
        libesp_ringbuf.a          0       0       0        1024           0         1024           0        97          0        1121
      libesp_partition.a          0       8       0           0           0            8         946        37          0         983
                liblog.a          8     272       0         229           0          509         546        21          0         804
             libxt_hal.a          0       0       0         443           0          443           0        32          0         475
     libesp_app_format.a          0      10       0           0           0           10         184         4        256         444
            libpthread.a          8      12       0           0           0           20         284         0          0         292
         libapp_update.a          0       4       0           0           0            4         147        30          0         177
                libgcc.a          0       0       0           0           0            0           0       160          0         160
            libesp_rom.a          0       0       0         102           0          102           0         0          0         102
                   (exe)          0       0       0           0           3            3           3        12          0          18
              libefuse.a          0       0       0           0           0            0          17         0          0          17
                libcxx.a          0       0       0           0           0            0           5         0          0           5
            libesp_phy.a          0       0       0           0           0            0           5         0          0           5
   libnvs_sec_provider.a          0       0       0           0           0            0           5         0          0           5
         libmbedcrypto.a          0       0       0           0           0            0           0         0          0           0
[0/1] cmd.exe /C "cd /D C:\Espressif\development\hello_world\build && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_SIZE_TOOL=C:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe;-m;esp_idf_size -D IDF_SIZE_MODE=--archives -D MAP_FILE=C:/Espressif/development/hello_world/build/hello_world.map -D OUTPUT_JSON= -P C:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/run_size_tool.cmake"
```

### More Information.

Variables placed in .noinit RAM (internal or external) should not have initialisation data occupying space in flash. This seems like a small thing until you have 1MB of noinit RAM for logs, you are running out of flash space and your OTA updates have blown out to 1MB larger than they should be.

ESP-Marius, 2024-04-01:
@dvosully I think you are right that there is a bug here, but not the one you suspect. 

If you just look at the output from `idf.py size` you should see that the total flash size actually do not increase. (section is correctly marked as NOLOAD in the linkerscript)

The issue seems to be in `esp-idf size` when we sum up the `flash total`. As we can see [here](https://github.com/espressif/esp-idf-size/blob/2910a565c8513953381aa644c7ca880a1ccbce99/esp_idf_size/core.py#L964C116-L964C129) we exclude `bss` but not `noinit`. We'll take a look and fix this.

dvosully, 2024-04-01:
Thanks for the rapid response @ESP-Marius, I see you are correct. When I look at the generated binary on disk its size is not increased.

fhrbata, 2024-04-01:
Hello @dvosully ,

yes, I believe @ESP-Marius  is right and this  seems like a  bug in `esp-idf-size`. IIUC the problem is that `.noinit` section is accounted into total image size [here](https://github.com/espressif/esp-idf-size/blob/2910a565c8513953381aa644c7ca880a1ccbce99/esp_idf_size/core.py#L688). Thank you for this report, I will look into this and provide a fix. Note, as a workaround, you can try to `set ESP_IDF_SIZE_NG=1` in the `cmd` shell.  With this env. var. set, the refactored `esp-idf-size`(>=`1.0.0`)  version should be used by default and it  actually considers only `PROGBITS` sections when counting the total image size if the elf file is available.

Thank you

closed 2024-04-04 by espressif-bot