espressif/esp-idf #13516: ASSERT_PARAM(5 13), in llm_util.c at line 215

aboyuanna, 2024-04-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

不会出现此错误。

### What is the actual behavior?

运行一段时间后出现此错误。

### Steps to reproduce.

我的项目上整合了BLE GATTS、BLE MESH、BT A2DP SINK、WIFI、MQTT这几项功能。我用手机连接BT播放音乐，过几个小时就会出现此错误。

### Build or installation Logs.

```plain
ASSERT_PARAM(5 13), in llm_util.c at line 215
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40091d50: f01d020c 00004136 f01d0000
0x40091d50: btdm_sleep_check_duration at C:/Users/HJYVM10/.espressif/frameworks/v5.1.3/esp-idf/components/bt/controller/esp32/bt.c:911

Core  0 register dump:
PC      : 0x40091d57  PS      : 0x00060b30  A0      : 0x8008531a  A1      : 0x3ffced30
0x40091d57: r_assert at C:/Users/HJYVM10/.espressif/frameworks/v5.1.3/esp-idf/components/bt/controller/esp32/bt.c:1848

A2      : 0x00000000  A3      : 0x00000005  A4      : 0x0000000d  A5      : 0x3f43def9
A6      : 0x000000d7  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffceca0
A10     : 0x00000000  A11     : 0x3ffcecc3  A12     : 0x3ffcec6f  A13     : 0x00000035
A14     : 0x00000000  A15     : 0x3ffcec74  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffe
0x400012c5: strcmp in ROM
0x400012d5: strcmp in ROM

Backtrace: 0x40091d54:0x3ffced30 0x40085317:0x3ffced50 0x4019aa99:0x3ffced70 0x401977d3:0x3ffcedc0 0x401947ae:0x3ffcee10 0x40019d11:0x3ffcee50 0x40055b4d:0x3ffcee70 0x40189147:0x3ffcee90 0x401897d2:0x3ffceeb0 0x4009518a:0x3ffceee0
0x40091d54: r_assert at C:/Users/HJYVM10/.espressif/frameworks/v5.1.3/esp-idf/components/bt/controller/esp32/bt.c:1847
0x40085317: r_assert_param at ??:?
0x4019aa99: r_llm_pdu_defer at ??:?
0x401977d3: r_lld_pdu_check at ??:?
0x401947ae: r_lld_evt_deffered_elt_handler at ??:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x40189147: r_rw_schedule at ??:?
0x401897d2: btdm_controller_task at ??:?
0x4009518a: vPortTaskWrapper at C:/Users/HJYVM10/.espressif/frameworks/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: f86599902c1deb5d
```

### More Information.

[Uploading sdkconfig.txt…]()

weiyuhannnn, 2024-04-01:
Hi @aboyuanna Thanks for reporting. May I know which Espressif chip do your use in your project? Thanks.

aboyuanna, 2024-04-01:
> Hi @aboyuanna Thanks for reporting. May I know which Espressif chip do your use in your project? Thanks.

Thank you for your reply.My code runs on the development board ESP32_LyraT v4.3, which is equipped with an ESP32-WROWER-E N8R8 module.

zhp0406, 2024-04-01:
We have fixed this issue, but it hasn't been synchronized to GitHub yet. Could you provide me with the controller commit? I'll give you a patch for the fix. The controller commit can be found in the startup logs. Could you please provide me with the complete startup log? Thank you.

aboyuanna, 2024-04-01:
> We have fixed this issue, but it hasn't been synchronized to GitHub yet. Could you provide me with the controller commit? I'll give you a patch for the fix. The controller commit can be found in the startup logs. Could you please provide me with the complete startup log? Thank you.

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1    
load:0x3fff0030,len:10228
load:0x40078000,len:17548
load:0x40080400,len:4    
0x40080400: _init at ??:?

load:0x40080404,len:4624 
entry 0x40080694
I (29) boot: ESP-IDF v5.1.3-dirty 2nd stage bootloader
I (29) boot: compile time Mar 29 2024 09:18:12
W (30) boot: Unicore bootloader
D (34) bootloader_flash: non-XMC chip detected by SFDP Read (68), skip.
D (41) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (47) boot: chip revision: v3.1
D (51) qio_mode: Probing for QIO mode enable...
D (56) qio_mode: Raw SPI flash chip id 0x684017
D (60) qio_mode: Manufacturer ID 0x68 chip ID 0x4017
I (65) qio_mode: Enabling default flash chip QIO
D (71) qio_mode: Initial flash chip status 0x2
D (75) qio_mode: QIO mode already enabled in flash
D (80) qio_mode: Enabling QIO mode...
D (84) boot.esp32: magic e9
D (87) boot.esp32: segments 04
D (90) boot.esp32: spi_mode 02
D (93) boot.esp32: spi_speed 0f
D (96) boot.esp32: spi_size 03
I (99) boot.esp32: SPI Speed      : 80MHz
I (104) boot.esp32: SPI Mode       : QIO
I (109) boot.esp32: SPI Flash Size : 8MB
D (113) boot: Enabling RTCWDT(9000 ms)
I (117) boot: Enabling RNG early entropy source...
D (123) bootloader_flash: rodata starts from paddr=0x0000f000, size=0xc00, will be mapped to vaddr=0x3f400000
D (132) boot: mapped partition table 0xf000 at 0x3f40f000
D (138) flash_parts: partition table verified, 9 entries
I (143) boot: Partition Table:
I (147) boot: ## Label            Usage          Type ST Offset   Length
D (154) boot: load partition table entry 0x3f40f000
D (159) boot: type=0 subtype=10
I (163) boot:  0 ota_0            OTA app          00 10 00010000 002b0000
D (170) boot: load partition table entry 0x3f40f020
D (175) boot: type=0 subtype=11
I (178) boot:  1 ota_1            OTA app          00 11 002c0000 002b0000
D (186) boot: load partition table entry 0x3f40f040
D (191) boot: type=1 subtype=2
I (194) boot:  2 nvs              WiFi data        01 02 00570000 00080000
D (201) boot: load partition table entry 0x3f40f060
D (206) boot: type=1 subtype=2
I (210) boot:  3 factory_nvs      WiFi data        01 02 005f0000 0000c000
D (217) boot: load partition table entry 0x3f40f080
D (222) boot: type=1 subtype=4
I (225) boot:  4 nvs_keys         NVS keys         01 04 005fc000 00001000
D (233) boot: load partition table entry 0x3f40f0a0
D (238) boot: type=1 subtype=0
I (241) boot:  5 otadata          OTA data         01 00 005fd000 00002000
D (248) boot: load partition table entry 0x3f40f0c0
D (253) boot: type=1 subtype=1
I (256) boot:  6 phy_init         RF data          01 01 005ff000 00001000
D (264) boot: load partition table entry 0x3f40f0e0
D (269) boot: type=1 subtype=81
I (272) boot:  7 storage          Unknown data     01 81 00600000 00200000
I (280) boot: End of partition table
D (284) boot: OTA data offset 0x5fd000
D (288) bootloader_flash: rodata starts from paddr=0x005fd000, size=0x2000, will be mapped to vaddr=0x3f400000
[... 221 lines cut ...]
I (08:00:04.148) ble_slave: 3a 61 32 61 33 3a 62 33 66 66 3a 66 65 38 35 3a 
I (08:00:04.156) ble_slave: 66 39 36 38 22 2c 22 62 73 73 69 64 22 3a 22 61 
I (08:00:04.163) ble_slave: 34 3a 33 39 3a 62 33 3a 64 37 3a 33 37 3a 32 61 
I (08:00:04.171) ble_slave: 22 2c 22 72 73 73 69 22 3a 2d 33 35 7d 
I (9013) wifi:<ba-add>idx:1 (ifx:0, a4:39:b3:d7:37:2a), tid:0, ssn:0, winSize:64
I (16:50:40.721) time: time synchronization: 2024-04-01 16:50:40
W (24268) BLE_MESH: Already stopped, index 1
W (24268) BLE_MESH: Already stopped, index 2
I (16:50:40.734) node: send cmd. srcAddr 0x0004, dstAddr 0xffff
I (16:50:40.742) node: 84 00 01 28 32 10 01 03 e8 07 01
I (16:50:42.120) mqtt: mqtt server:mqtt://msg.mob-mob.com  port:32777
I (16:50:42.121) mqtt: user:975567252@qq.com
I (16:50:42.122) mqtt: bb0a32a000c42dd783df950f5b8de77e
I (16:50:42.129) mqtt: Other event id:7
I (16:50:42.134) time: timer mqtt_msginr create!
I (16:50:42.137) time: timer pubinr_min create!
I (16:50:42.143) time: timer mqtt_reconnect create!
I (16:50:42.148) main: internal Free memory: 97451 bytes   psram Free memory: 3808844 bytes
I (16:50:42.158) main_task: Returned from app_main()
I (16:50:42.309) mqtt: MQTT_EVENT_CONNECTED
I (16:50:42.312) mqtt: sent publish successful, msg_id=37371
I (16:50:42.313) mqtt: sent subscribe successful, msg_id=52501
I (16:51:18.622) eventLoop: internal Free memory: 102063 bytes   psram Free memory: 3808484 bytes   internal Min memory: 97379 bytes 
I (16:52:18.622) eventLoop: internal Free memory: 102063 bytes   psram Free memory: 3808484 bytes   internal Min memory: 97379 bytes
I (16:53:18.622) eventLoop: internal Free memory: 102063 bytes   psram Free memory: 3808484 bytes   internal Min memory: 97379 bytes

zhp0406, 2024-04-01:
please replace controller lib file in esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a with the following lib:
[60f4bc1.zip](https://github.com/espressif/esp-idf/files/14822194/60f4bc1.zip)

AxelLin, 2024-04-27:
> We have fixed this issue, but it hasn't been synchronized to GitHub yet. 

@zhp0406 Is this fixed now? Which commit fixed it?

aboyuanna, 2024-04-28:
After a period of testing, this error did not occur again.Thank you for your help.@zhp0406

closed 2024-04-28 by aboyuanna

zhp0406, 2024-04-28:
master:9c6869fbf8fffe13bc9aa794fdc12c24c867cf50
Backports: v5.2 51f7ddc223db977eeea014d6327d2502c62cc554
Backports: v5.1 will coming soon...
Backports: v5.0 c9c946811c079d302a2df234976519c34ddfccb8
Backports: v4.4 fe1299e5dafc48c651f6044d9c534f9edf6e45a1