espressif/esp-idf #13518: USB CDC could be not flush

yizi15, 2024-04-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1 MCN8R8

### Power Supply used.

External 5V

### What is the expected behavior?

USB CDC can work in full duplex at 5M baud rate

### What is the actual behavior?

USB CDC can only receive under unknown conditions, and flush always returns a timeout（500ms）. tinyusb_cdcacm_write_flush(TINYUSB_CDC_ACM_0, 500);

### Steps to reproduce.

1. Use ble client and server at the same time, and perform operations according to USB commands at the same time
2. the function tinyusb_cdcacm_write_queue and tinyusb_cdcacm_write_flush is only called at a thread

### More Information.

usb cdc works most of the time，and flush fail cant recover

roma-jam, 2024-04-02:
Hi @yizi15 , 
Thanks for the reporting the problem, 

To excluding the possibility of the ble interfere, is the problem with cdc still present when you disable the ble client and server? 

To have more information regarding the problem is that possible: 
1. To share part of the code (especially the part where the problem is, I mean tinyusb acm cdc receiving part)
2. Share some information about the commands on cdc line (length, frequency e.t.c)
3. Attach the device output log, where the problem is (timeout)

yizi15, 2024-04-02:
@roma-jam the code without ble is also flush fail, 
the log is like this
W (306135) tusb_cdc_acm: Flush failed
(306135) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(306135) ql_usb_flush_inter: usb ready 1
(306135) ql_usb_flush_inter: tud_cdc_n_write_available 5979
W (306636) tusb_cdc_acm: Flush failed
[usb_test.zip](https://github.com/espressif/esp-idf/files/14834556/usb_test.zip)

yizi15, 2024-04-02:
> @roma-jam the code without ble is also flush fail, the log is like this W (306135) tusb_cdc_acm: Flush failed (306135) ql_usb_flush_inter: ESP_ERR_TIMEOUT (306135) ql_usb_flush_inter: usb ready 1 [...]

in devKit ，the probability of this problem occurring is high, especially after plugging and unplugging the USB cable.

yizi15, 2024-04-07:
[image]
The application not have eht vbus monitor, I will add the gpio and try agin

yizi15, 2024-04-07:
> ![image](https://private-user-images.githubusercontent.com/63935496/320254407-aade860a-8e4a-40e1-906c-ba1d0fcd583d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

add the R1(56K),R2(75K), and code
 ```       
 tinyusb_config_t tusb_cfg = {.self_powered=true,.vbus_monitor_io=17}; 
// the configuration using default values
  ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));
``` 
not work

roma-jam, 2024-04-08:
Hi @yizi15,

Which version of espressif TinyUSB do you use? It should be >= 0.15.0.5 (more info here: https://github.com/espressif/esp-idf/issues/12360#issuecomment-1943769455) 

Based on your last input, it seems that the problem is more connected to the „data transfer problem after several plug-unplug actions with usb cdc device“, not simultanious using usb and ble. 

Does that seem rigth? 
I will check it asap and return.

yizi15, 2024-04-08:
> Hi @yizi15,
> 
> Which version of espressif TinyUSB do you use? It should be >= 0.15.0.5 (more info here: [#12360 (comment)](https://github.com/espressif/esp-idf/issues/12360#issuecomment-1943769455))
> 
> Based on your last input, it seems that the problem is more connected to the „data transfer problem after several plug-unplug actions with usb cdc device“, not simultanious using usb and ble.
> 
> Does that seem rigth? I will check it asap and return.

version of esp_tinyusb is `version: 1.4.3`, version of tinyusb is `0.15.1`

yizi15, 2024-04-08:
@roma-jam ， The version of tinyusb is 0.15.1 is version.yml but `version: 0.15.0~4` in idf_component.yml.
 I modify the  esp_tinyusb dependances to 
```
dependencies:
  idf:
    version: '>=5.0'
  tinyusb:
    public: true
    version: '=0.14.3'
```
 The version of tinyusb is 0.0.0 in version.yml,  version: 0.14.3`in idf_component.yml.
After serval  plug-unplug actions with usb cdc device, the log
```
(59895) UsbRecvCb: at msg AT+box_info?
8ebcdc3015
, res no error
W (62694) tusb_cdc_acm: Flush failed
(62694) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(62694) ql_usb_flush_inter: usb ready 0
(62694) ql_usb_flush_inter: tud_cdc_n_write_available 5851
(62694) ql_usb_flush_inter: cpu runtime 62694
free_heap 8146224 min_heap 8135228
```

roma-jam, 2024-04-08:
@yizi15, 

> version of tinyusb is 0.15.1 

Versions, without issue with plug/unplug - is either `0.14.2` or higher than `^0.15.0.5`. (here is [info](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/versioning.html) about versioning) 
First thing that need to be done - is to use the correct version.

yizi15, 2024-04-08:
I delete before component and  modify the main/idf_component.yml
```
# dependencies:
#   espressif/led_strip: "^2.0.0"

## IDF Component Manager Manifest File
dependencies:
  espressif/esp_tinyusb: "^1.4.2"
  espressif/tinyusb: '0.14.2'
```
This problem will still occur, but the probability has decreased
```
I (177623) usb: tud_umount_cb
W (178883) tusb_cdc_acm: Flush failed
(178883) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(178883) ql_usb_flush_inter: usb ready 0
(178883) ql_usb_flush_inter: tud_cdc_n_write_available 5851
(178883) ql_usb_flush_inter: cpu runtime 178883
free_heap 8142784 min_heap 8139120

(178883) ql_usb_flush_inter: inter free_heap 313419 min_heap 310015

I (180847) usb: tud_mount_cb
I (180854) msg_up_usb: Line state changed! dtr:0, rst:0
I (181638) msg_up_usb: Line state changed! dtr:0, rst:0
(181639) UsbRecvCb: at msg AT+sync_time=18ebceda384
, res undefined cmd
(181639) UsbRecvCb: at msg AT+start_acq
8ebceda384
, res no error
(181640) UsbRecvCb: at msg AT+start_acq
8ebceda384
, res no error
I (183883) usb: tud_umount_cb
W (185120) tusb_cdc_acm: Flush failed
(185120) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(185120) ql_usb_flush_inter: usb ready 0
(185120) ql_usb_flush_inter: tud_cdc_n_write_available 5851
(185120) ql_usb_flush_inter: cpu runtime 185120
free_heap 8145124 min_heap 8139120

(185120) ql_usb_flush_inter: inter free_heap 313419 min_heap 310015

I (185906) usb: tud_mount_cb
I (185911) msg_up_usb: Line state changed! dtr:0, rst:0
I (186917) msg_up_usb: Line state changed! dtr:0, rst:0
(186918) UsbRecvCb: at msg AT+sync_time=18ebcedb823
, res undefined cmd
(186918) UsbRecvCb: at msg AT+start_acq
8ebcedb823
, res no error
(186918) UsbRecvCb: at msg AT+start_acq
8ebcedb823
, res no error
(186919) UsbRecvCb: at msg AT+box_info?
8ebcedb823
, res no error
I (188018) usb: tud_umount_cb
W (188957) tusb_cdc_acm: Flush failed
(188957) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(188957) ql_usb_flush_inter: usb ready 0
(188957) ql_usb_flush_inter: tud_cdc_n_write_available 5851
(188957) ql_usb_flush_inter: cpu runtime 188957
free_heap 8144500 min_heap 8139120

(188957) ql_usb_flush_inter: inter free_heap 313419 min_heap 310015

I (189506) usb: tud_mount_cb
I (189512) msg_up_usb: Line state changed! dtr:0, rst:0
I (190032) msg_up_usb: Line state changed! dtr:0, rst:0
(190033) UsbRecvCb: at msg AT+sync_time=18ebcedc44e
, res undefined cmd
(190033) UsbRecvCb: at msg AT+start_acq
8ebcedc44e
, res no error
(190033) UsbRecvCb: at msg AT+start_acq
8ebcedc44e
, res no error
I (191872) usb: tud_umount_cb
I (192795) usb: tud_mount_cb
I (192801) msg_up_usb: Line state changed! dtr:0, rst:0
I (192871) msg_up_usb: Line state changed! dtr:0, rst:0
(192872) UsbRecvCb: at msg AT+sync_time=18ebcedcf64
, res undefined cmd
(192872) UsbRecvCb: at msg AT+start_acq
8ebcedcf64
, res no error
(192872) UsbRecvCb: at msg AT+start_acq
8ebcedcf64
, res no error
I (194252) usb: tud_umount_cb
I (195672) usb: tud_mount_cb
I (195678) msg_up_usb: Line state changed! dtr:0, rst:0
W (196104) tusb_cdc_acm: Flush failed
(196104) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(196104) ql_usb_flush_inter: usb ready 1
(196104) ql_usb_flush_inter: tud_cdc_n_write_available 5915
(196104) ql_usb_flush_inter: cpu runtime 196104
free_heap 8141224 min_heap 8137560

(196104) ql_usb_flush_inter: inter free_heap 313419 min_heap 310015

I (196281) msg_up_usb: Line state changed! dtr:0, rst:0
(196282) UsbRecvCb: at msg AT+sync_time=18ebceddcb6
, res undefined cmd
(196282) UsbRecvCb: at msg AT+start_acq
8ebceddcb6
, res no error
(196282) UsbRecvCb: at msg AT+start_acq
8ebceddcb6
, res no error
W (196783) tusb_cdc_acm: Flush failed
(196783) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(196783) ql_usb_flush_inter: usb ready 1
(197472) ql_usb_flush_inter: usb ready 1
(197472) ql_usb_flush_inter: tud_cdc_n_write_available 5939
(197472) ql_usb_flush_inter: cpu runtime 197472
free_heap 8146048 min_heap 8137560

(197472) ql_usb_flush_inter: inter free_heap 313419 min_heap 310015

(207017) UsbRecvCb: at msg AT+box_info?
8ebceddcb6
, res no error
W (207517) tusb_cdc_acm: Flush failed
(207517) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(207517) ql_usb_flush_inter: usb ready 1
(207517) ql_usb_flush_inter: tud_cdc_n_write_available 5925
(207517) ql_usb_flush_inter: cpu runtime 207517
free_heap 8146048 min_heap 8137560

(207517) ql_usb_flush_inter: inter free_heap 313419 min_heap 310015

I (210366) usb: tud_umount_cb
```

roma-jam, 2024-04-08:
@yizi15, 

Great, thanks for sharing the log. (It is better to use the latest version though).

Now, the logic. in the code there is a `usb_cdc_line_state_changed_callback`, which detects the line state change on usb cdc device. (Which means, that when the host (PC Host) launches the software, it changes the line state). So, the correct way of determine the possibility to send/receive data will be: 
1. Use `tud_mount_cb`, to get the event of connecting the device to the PC host.
2. Use  `usb_cdc_line_state_changed_callback` to determine readiness of host software to transmit of receive data. 

Based in both events, the `ql_usb_is_ready` could change the state. 

In all other cases there could be a problems on both sides, so it is better to use that strategy. 
If changes on the CDC line correspond to the PC software enabling the receive, user can start transfer data

yizi15, 2024-04-08:
```
W (196783) tusb_cdc_acm: Flush failed
(196783) ql_usb_flush_inter: ESP_ERR_TIMEOUT
(196783) ql_usb_flush_inter: usb ready 1
(197472) ql_usb_flush_inter: usb ready 1
(197472) ql_usb_flush_inter: tud_cdc_n_write_available 5939
(197472) ql_usb_flush_inter: cpu runtime 197472
free_heap 8146048 min_heap 8137560
```
the log is abnormal, the ` usb ready 1` print twice, and interval 90ms,  but in the code , it should be once. 
in my log code , a bug of 
```
        vTaskSuspendAll();
        buf = (char*) malloc(256);
        xTaskResumeAll();
``` found but not lead to print twice. I remove the Suspend and flush fail again.

yizi15, 2024-04-08:
@roma-jam  Version specifications for "espressif/tinyusb" are invalid. when 
```
dependencies:
  espressif/esp_tinyusb: "^1.4.2"
  espressif/tinyusb: '^0.15.0.5'
```

roma-jam, 2024-04-08:
@yizi15, 

> tud_cdc_n_write_available 5915

What is the actual length of the data that should be transmitted? 

For versioning you may use the following format: 
```
dependencies:
  espressif/esp_tinyusb: "^1.4.2"
  espressif/tinyusb: "^0.15.0~5"
```
More information about versioning is [here](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/versioning.html).

roma-jam, 2024-04-08:
@yizi15, 

> What is the actual length of the data that should be transmitted?

Based on the example you provided, the data length should be 149 bytes.

Please, review the logic of the event, when the data transfer start is triggered. 
If the host software is not ready (that means we omitted the `usb_cdc_line_state_changed_callback`) or we start the data transfer sooner than USB device stack is ready (that means we omitted the `tud_mount_cb`) it may affect the transfer stage.

yizi15, 2024-04-08:
@roma-jam my code could start the data transfer after the `tud_umount_cb`, after flush or write fail, the data is not trans before the next recvive data event。

yizi15, 2024-04-08:
@roma-jam 
  What is the actual length of the data that should be transmitted?
the normal data len is 149 per 20ms , but the host will send heartbeat msg per 10s, the ack msg is about 100

yizi15, 2024-04-08:
@roma-jam at the version `0.15.0~7`, without `in the umount_cb set the ready_flag false`, after serval plug/unplug, the log is 
```
I (189230) usb: tud_umount_cb
I (190752) usb: tud_suspend_cb 0
I (193702) usb: tud_umount_cb
I (207942) usb: tud_umount_cb
I (209620) usb: tud_suspend_cb 0
I (211674) usb: tud_umount_cb
I (229112) usb: tud_suspend_cb 0
I (244711) usb: tud_umount_cb
I (247643) usb: tud_suspend_cb 0
I (252593) usb: tud_umount_cb
I (252639) usb: tud_umount_cb
I (256187) usb: tud_suspend_cb 0
I (263032) usb: tud_umount_cb
I (276896) usb: tud_suspend_cb 0
I (278048) usb: tud_umount_cb
I (280608) usb: tud_umount_cb
I (282115) usb: tud_suspend_cb 0
I (286620) usb: tud_umount_cb

```
only the tud_umount_cb and pc cant open the serial again

roma-jam, 2024-04-17:
Hi @yizi15, 

Sorry for the delay. 

May I ask you: 
1. With the current configuration, try to use tinyusb verison 0.15.0~5. Then rebuild and verify the last issue you reported. 
2. If the issue will be still there, enable TinyUSB debug (level 3) and please share the debug output. 

Thanks.

roma-jam, 2024-04-18:
Hi @yizi15 ,

We have release new version of espressif/tinyusb (v.0.15.0~8) and it could (there is a possibility your issue is connected https://github.com/espressif/tinyusb/pull/23, but I didn't get an answer on my previous question)

Could you try to verify your problem with several plug/unplug with new version? 

Thanks

yizi15, 2024-04-19:
@roma-jam the `only the tud_umount_cb and pc cant open the serial again`  no apear, the previous  problem (usb only can receive )  appear, I will  try enable TinyUSB debug (level 3) at  0.15.0~8

yizi15, 2024-04-19:
@roma-jam 
[usb_debug.log](https://github.com/espressif/esp-idf/files/15032279/usb_debug.log)

yizi15, 2024-04-19:
```

[11:44:45.827]收←◆ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1758
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e04
entry 0x403c9908
[0;32mI (27) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader[0m
[0;32mI (27) boot: compile time Apr  2 2024 17:38:53[0m
[0;32mI (27) boot: Multicore bootloader[0m
[0;32mI (30) boot: chip revision: v0.2[0m
[0;32mI (34) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (39) boot.esp32s3: SPI Mode       : DIO[0m
[0;32mI (44) boot.esp32s3: SPI Flash Size : 8MB[0m
[0;32mI (48) boot: Enabling RNG early entropy source...[0m
[0;32mI (54) boot: Partition Table:[0m
[0;32mI (57) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (87) boot:  3 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (95) boot:  4 ota_0            OTA app          00 10 00210000 00200000[0m
[0;32mI (102) boot:  5 ota_1            OTA app          00 11 00410000 00200000[0m
[0;32mI (110) boot:  6 coredump         Unknown data     01 03 00610000 00010000[0m
[0;32mI (117) boot: End of partition table[0m
[0;32mI (121) boot: Defaulting to factory image[0m
[0;32mI (126) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0e3a8h ( 58280) map[0m
[0;32mI (145) esp_image: segment 1: paddr=0001e3d0 vaddr=3fc96e00 size=01c48h (  7240) load[0m
[0;32mI (152) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2e4f8h (189688) map[0m
[0;32mI (186) esp_image: segment 3: paddr=0004e520 vaddr=3fc98a48 size=049e0h ( 18912) load[0m
[0;32mI (191) esp_image: segment 4: paddr=00052f08 vaddr=40378000 size=0ed7ch ( 60796) load[0m
[0;32mI (213) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (218) boot: Disabling RNG early entropy source...[0m
[0;32mI (224) cpu_start: Multicore app[0m
[0;32mI (224) octal_psram: ECC is enabled[0m
[0;32mI (225) octal_psram: vendor id    : 0x0d (AP)[0m
[0;32mI (226) octal_psram: dev id       : 0x02 (generation 3)[0m
[0;32mI (232) octal_psram: density      : 0x03 (64 Mbit)[0m
[0;32mI (237) octal_psram: good-die     : 0x01 (Pass)[0m
[0;32mI (243) octal_psram: Latency      : 0x01 (Fixed)[0m
[0;32mI (248) octal_psram: VCC          : 0x01 (3V)[0m
[0;32mI (253) octal_psram: SRF          : 0x
[11:44:46.116]收←◆01 (Fast Refresh)[0m
[0;32mI (259) octal_psram: BurstType    : 0x00 ( Wrap)[0m
[0;32mI (264) octal_psram: BurstLen     : 0x03 (1024 Byte)[0m
[0;32mI (270) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)[0m
[0;32mI (276) octal_psram: DriveStr
[11:44:46.148]收←◆ength: 0x00 (1/1)[0m
[0;32mI (281) esp_psram: Found 8MB PSRAM device[0m
[0;32mI (286) esp_psram: Speed: 40MHz[0m
[0;32mI (290) cpu_start: Pro cpu up.[0m
[0;32mI (293) cpu_start: Starting app cpu, entry point is 0x40379520[0m
[0;32mI (0) cpu_start: App cpu up.[0m

[11:44:46.885]收←◆[0;32mI (1056) esp_psram: SPI SRAM memory test OK[0m

[11:44:46.988]收←◆[0;32mI (1065) cpu_start: Pro cpu start user code[0m
[0;32mI (1070) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (1071) cpu_start: Application information:[0m
[0;32mI (1074) cpu_star
[11:44:47.009]收←◆t: Project name:     ESP32_USB_TEST[0m
[0;32mI (1079) cpu_start: App version:      0500240305000200000000000000[0m
[0;32mI (1086) cpu_start: ELF file SHA256:  5a2b9347455f6ffd...[0m
[0;32mI (1092) cpu_start: ESP-IDF:          v5.1.2-dirty[0m
[0;32mI (1098) cpu_start: Min chip rev:     v0.0[0m
[... 1586 lines cut ...]
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 11 bytes ...
USBD Xfer Complete on EP 80 with 11 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

[11:45:15.044]收←◆USBD Xfer Complete on EP 02 with 14 bytes
  CDC xfer callback
  Queue EP 02 with 64 bytes ...
  Queue EP 82 with 14 bytes ...
(29217) UsbRecvCb: at msg AT+start_acq
, res no error

[11:45:15.547]收←◆[0;33mW (29718) tusb_cdc_acm: Flush failed[0m
[0;32mI (29718) ql_usb_flush_inter: ESP_ERR_TIMEOUT[0m
[0;32mI (29718) ql_usb_flush_inter: usb ready 1[0m
[0;32mI (29720) ql_usb_flush_inter: tud_cdc_n_write_available 5993[0m
[0;32mI (29727) ql_usb_flush_inter: cpu runtime 29727
free_heap 8146928 min_heap 8137508
[0m
[0;32mI (29735) ql_usb_flush_inter: inter free_heap 314407 min_heap 309963
[0m
```

yizi15, 2024-04-19:
```
const gpio_config_t vbus_gpio_config = {
                .pin_bit_mask = BIT64(17),
                .mode = GPIO_MODE_INPUT,
                .intr_type = GPIO_INTR_DISABLE,
                .pull_up_en = false,
                .pull_down_en = false,
        };
        ESP_ERROR_CHECK(gpio_config(&vbus_gpio_config));
``` 
previous  dont have vbus gpio init, and the init no work.

yizi15, 2024-04-19:
```
USBD Setup Received 80 06 00 02 00 00 4B 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 11 bytes ...
USBD Xfer Complete on EP 80 with 11 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Xfer Complete on EP 02 with 14 bytes
  CDC xfer callback
  Queue EP 02 with 64 bytes ...
  Queue EP 82 with 14 bytes ...
(USBD Xfer Complete on EP 82 with 14 bytes
  CDC xfer callback
  Queue EP 82 with 7 bytes ...
30877) UUSBD Xfer Complete son EP 82 with 7 bytes
  CDC xfer callback
  Queue EP 82 with 64 bytes ...
bRecvCb: at msg AT+start_acq
, res no error
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x42010b55  PS      : 0x00060034  A0      : 0x82011515  A1      : 0x3fc98e80
0x42010b55: handle_epin_irq at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1245

A2      : 0x00000002  A3      : 0x60080940  A4      : 0x00002090  A5      : 0x60080000  
A6      : 0x00000002  A7      : 0x00000001  A8      : 0x0003fffc  A9      : 0x00000010
A10     : 0x00ff0000  A11     : 0xff000000  A12     : 0x8037b47d  A13     : 0x3fcb1200
A14     : 0x3fcb1220  A15     : 0x0000000c  SAR     : 0x0000001e  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000
0x40056fc5: memcpy in ROM

0x40056fe7: memcpy in ROM

Core  0 was running in ISR context:
EPC1    : 0x40383aa3  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x42010b55
0x40383aa3: uart_hal_write_txfifo at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/hal/uart_hal_iram.c:35

0x42010b55: handle_epin_irq at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1245

Backtrace: 0x42010b52:0x3fc98e80 0x42011512:0x3fc98ed0 0x4201151e:0x3fc98f00 0x4037b011:0x3fc98f20 0x4200f86b:0x3fcb15c0 0x4200e233:0x3fcb15f0 0x4200d26b:0x3fcb1610 0x4037b752:0x3fcb1630 0x4037b8d1:0x3fcb1660 0x4200b472:0x3fcb1680 0x40381805:0x3fcb16b0
0x42010b52: handle_epin_irq at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1245

0x42011512: dcd_int_handler at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1418

0x4201151e: dcd_int_handler_wrap at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dwc2_esp32.h:100

0x4037b011: _xt_lowint1 at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240

0x4200f86b: usbd_edpt_xfer at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__tinyusb/src/device/usbd.c:1270

0x4200e233: tud_cdc_n_write_flush at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__tinyusb/src/class/cdc/cdc_device.c:201

0x4200d26b: tinyusb_cdcacm_write_flush at D:/ESP32_Work/project/X7_BoardV2/usb_test/managed_components/espressif__esp_tinyusb/tusb_cdc_acm.c:271

0x4037b752: ql_usb_flush_inter at D:/ESP32_Work/project/X7_BoardV2/usb_test/main/src/ql_usb.c:93

0x4037b8d1: ql_usb_flush at D:/ESP32_Work/project/X7_BoardV2/usb_test/main/src/ql_usb.c:108

0x4200b472: msg_usb_send_task_entry at D:/ESP32_Work/project/X7_BoardV2/usb_test/main/src/msg_config.c:53

0x40381805: vPortTaskWrapper at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

Core  1 register dump:
PC      : 0x4037d1f2  PS      : 0x00060934  A0      : 0x8200359d  A1      : 0x3fca5350
0x4037d1f2: esp_cpu_wait_for_intr at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/esp_hw_support/cpu.c:121

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fca2010  A5      : 0x3fca1ff0
A6      : 0x4037cf40  A7      : 0x00000001  A8      : 0x82014e0e  A9      : 0x3fca5310  
0x4037cf40: ipc_task at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/esp_system/esp_ipc.c:48

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3fca1ff0  A13     : 0x3fca1fc0  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4037d1ef:0x3fca5350 0x4200359a:0x3fca5370 0x40380195:0x3fca5390 0x40381805:0x3fca53b0
0x4037d1ef: xt_utils_wait_for_intr at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/esp_hw_support/cpu.c:111

0x4200359a: esp_vApplicationIdleHook at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/esp_system/freertos_hooks.c:59

0x40380195: prvIdleTask at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/tasks.c:4327 (discriminator 1)

0x40381805: vPortTaskWrapper at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 2ac95957f41080d9

I (13489) esp_core_dump_flash: Save core dump to flash...
I (13496) esp_core_dump_flash: Erase flash 16384 bytes @ 0x610000
I (13623) esp_core_dump_flash: Write end offset 0x3fe4, check sum length 4
I (13623) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379938
0x40379938: esp_restart_noos_dig at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1758
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e04
entry 0x403c9908
```
therer is a dump log, my env cant parser the dump of esp32
[build.zip](https://github.com/espressif/esp-idf/files/15033430/build.zip)

roma-jam, 2025-01-20:
Hi @yizi15, 

sorry for such a long delay in reply. Is the problem still relevant? 

During the past time, we have updated the `espressif/tinyusb` component to v0.17.0.2 and `espressif/esp_tinyusb` component to version v1.7.0. 
You can update dependencies in you project by running `idf.py update-dependencies` and verify the error presence in the newer versions. 

Thanks.

yizi15, 2025-01-20:
> Hi [@yizi15](https://github.com/yizi15),
>
> sorry for such a long delay in reply. Is the problem still relevant?
>
> During the past time, we have updated the `espressif/tinyusb` component to v0.17.0.2 [...]

During the past time,  The problem also occurs on rp2040(tinyusb), zynq7000(zynq example)。A short Reset is acceptable for us project（only to reset the system, i cant find way to reset usb only）.

roma-jam, 2025-01-20:
@yizi15 

Ok, I see. 
Are there any related issue ticket in the tinyusb upstream regarding you problem?

yizi15, 2025-01-21:
@roma-jam 
Another issue is host could not found or open the usb device(esp32). After a device reset is ok.  It is difficult to reproduce。the usb init always return ESP_OK.

closed 2025-01-22 by yizi15