espressif/esp-idf #13520: hcif mode change consumes 800Tick

nopnop2002, 2024-04-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I use classic bluetooth spp.
```
$ idf.py --version
ESP-IDF v5.2.1-210-gcf7b9bed4c-dirty
```

This is the logging when only executing esp_spp_write().
All processing is executed quickly.
```
W (110209) BT_BTC: btc_transfer_context btc_task_post ret=0
I (110209) SPP: ESP_SPP_WRITE_EVT
W (110209) BT_BTC: btc_transfer_context btc_task_post ret=0

W (111349) BT_BTC: btc_transfer_context btc_task_post ret=0
I (111349) SPP: ESP_SPP_WRITE_EVT
W (111349) BT_BTC: btc_transfer_context btc_task_post ret=0
```

After execute ESP_SPP_DATA_IND_EVT, the log output will obviously change.
__hcif mode change consumes 800Tick.__
```
W (435159) BT_BTC: btc_transfer_context btc_task_post ret=0
I (435159) SPP: ESP_SPP_WRITE_EVT
W (435169) BT_BTC: btc_transfer_context btc_task_post ret=0
W (435959) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x0
W (435959) BT_BTC: btc_transfer_context btc_task_post ret=0
I (435959) SPP: ESP_BT_GAP_MODE_CHG_EVT mode:0 bda:[b8:c3:85:42:11:44]

W (257289) BT_BTC: btc_transfer_context btc_task_post ret=0
I (257289) SPP: ESP_SPP_WRITE_EVT
W (257289) BT_BTC: btc_transfer_context btc_task_post ret=0
```

xiongweichao, 2024-04-07:
Hi @nopnop2002 ,

If both devices have no data to send, they will enter sniff mode after a few seconds. This is to reduce the number of empty packets (lower power consumption).

If there is data to be sent, sniff mode will exit and enter active mode. When there is a lot of data to be sent, there will be a huge delay if you do not exit sniff mode.

Thanks

nopnop2002, 2024-04-07:
@xiongweichao 
Thank you for your reply.

Is there a way to exit sniff mode?

xiongweichao, 2024-04-07:
@nopnop2002  May I ask if you don't want to use sniff mode?

xiongweichao, 2024-04-08:
> After execute ESP_SPP_DATA_IND_EVT noticeably changes the log output.
> 
> Will this put BLE into smiff mode?

No, sniff mode has been exited.

> BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x0

`mode = 0` means that both devices have entered active mode, `mode = 2` means that they have entered sniff mode.

xiongweichao, 2024-04-15:
> No, sniff mode has been exited.
>
>
> `mode = 0` means that both devices have entered active mode, `mode = 2` means that they have entered sniff mode.

@nopnop2002  I don't know if you understand it yet. If you have any questions, please tell me. Thanks.

closed 2025-03-10 by Alvin1Zhang