espressif/esp-idf #13528: ESP IDF 如何確認外部靜態函示庫，能夠直接使用或是需要重新編譯

AM0219350, 2024-04-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

我實作下方連結提到的專案，但是在UWB模組上，有更新的library需要更新，但Qorvo uwb官方僅提供 a.的檔案給 cortex-M4及M33，我預期是能夠直接使用官方提供的靜態函示庫

UWB Project:
https://github.com/Makerfabs/Makerfabs-ESP32-UWB-DW3000

### What is the actual behavior?

我使用下方連結的API，並將檔案加入到我的專案中，但是在build時，出現relocations in generic ELF (EM: 40)，ESP32是否能夠直接使用 cortex-M4或M33的靜態函示庫？

UWB模組：
https://www.qorvo.com/products/p/DWM3000#documents

[image]

### Steps to reproduce.

1.  實作上面提到的UWB Project
2.  下載UWB模組新的library，Software > DWM3000 API Software and API Guide > Release_XR6.0C > Software > DW3XXX_API > Sources > DW3XXX_API_rev9p3 打開第二個壓縮檔
3. API > Shared > dwt_uwb_driver ，將 dwt_uwb_driver 資料夾移植到ESP專案中
4. Build 後產生錯誤
...
[DWM3000 API Software and API Guide](https://www.qorvo.com/products/d/da007992)

louielauanu, 2024-04-10:
> ### Answers checklist.
> * [x]  I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
> * [x]  I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
> * [x]  I have searched the issue tracker for a similar issue and not found a similar issue.
> 
> ### IDF version.
> esp-idf-v4.4.3
> 
> ### Operating System used.
> Windows
> 
> ### How did you build your project?
> Command line with idf.py
> 
> ### If you are using Windows, please specify command line type.
> PowerShell
> 
> ### What is the expected behavior?
> 我實作下方連結提到的專案，但是在UWB模組上，有更新的library需要更新，但Qorvo uwb官方僅提供 a.的檔案給 cortex-M4及M33，我預期是能夠直接使用官方提供的靜態函示庫
> 
> UWB Project: https://github.com/Makerfabs/Makerfabs-ESP32-UWB-DW3000
> 
> ### What is the actual behavior?
> 我使用下方連結的API，並將檔案加入到我的專案中，但是在build時，出現relocations in generic ELF (EM: 40)，ESP32是否能夠直接使用 cortex-M4或M33的靜態函示庫？
> 
> UWB模組： https://www.qorvo.com/products/p/DWM3000#documents
> 
> ![螢幕擷取畫面 2024-04-02 175345](https://private-user-images.githubusercontent.com/12125404/318731805-64272172-6877-478f-b0dd-c8fbed8f32ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI3Mzg5OTEsIm5iZiI6MTcxMjczODY5MSwicGF0aCI6Ii8xMjEyNTQwNC8zMTg3MzE4MDUtNjQyNzIxNzItNjg3Ny00NzhmLWIwZGQtYzhmYmVkOGYzMmFjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEwVDA4NDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZmMDhiYmU5MGViMjkwZDM2OWRkNTY2NzhlYWQ3YTQ0OTQxZjBmYzlhYWZlNDFhMjMyYTg3MGFjNzM1YmI3OTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.irnfi_ifhILcNXFhO7M3fk5CBGG5XoUjkGTVGLjjsBY)
> 
> ### Steps to reproduce.
> 1. 實作上面提到的UWB Project
> 2. 下載UWB模組新的library，Software > DWM3000 API Software and API Guide > Release_XR6.0C > Software > DW3XXX_API > Sources > DW3XXX_API_rev9p3 打開第二個壓縮檔
> 3. API > Shared > dwt_uwb_driver ，將 dwt_uwb_driver 資料夾移植到ESP專案中
> 4. Build 後產生錯誤
>    ...
>    [DWM3000 API Software and API Guide](https://www.qorvo.com/products/d/da007992)
> 
> ### Build or installation Logs.
> _No response_
> 
> ### More Information.
> _No response_

Hi Ken,

在嵌入式系统中，静态库文件与机器型号和指令集架构密切相关。ESP32-S3采用xtensa架构，所以针对ARM cortex-M的库文件并不适用于此芯片。如果有源文件，请尝试重新编译针对xtensa架构的版本。

AM0219350, 2024-04-12:
> Hi Ken,
>
> 在嵌入式系统中，静态库文件与机器型号和指令集架构密切相关。ESP32-S3采用xtensa架构，所以针对ARM cortex-M的库文件并不适用于此芯片。如果有源文件，请尝试重新编译针对xtensa架构的版本。

好的，謝謝您的回覆

closed 2024-04-12 by AM0219350