espressif/esp-idf #13529: WiFi EAP (Enterprise) for Soft AP Mode

SmartAppsML, 2024-04-02:
### Is your feature request related to a problem?

I would like to implement a Soft Access Point using WiFi EAP as authentication method. As far as I know, this is only supported in station mode.

### Describe the solution you'd like.

Implement the WiFi EAP protocol for Soft AP mode. Are there any plans on this implementation? Can I implement it myself or is it necessary to modify the pre-compiled WiFi library for that?

kapilkedawat, 2024-04-16:
Uninformative its not that straightforward since we will need changes at multiple levels such as adding correct IEs, allowing enterprise stations, forwarding enterprise packets to sever and get them back and then send them through driver and so on.. 

How do you expect packets to be forwarded to host? If we can understand the usecase more, may be we'll be able suggest some solution.

If you only want to run it as phy, I think better solution would be esp-hosted(https://github.com/espressif/esp-hosted), however it doesn't support AP mode at the moment.

SmartAppsML, 2024-04-16:
We have heavily modified the network stack and are able to directly
exchange packets between the host and a WiFi client using USB or SPI. For
the host, it basically is just a WiFi stick. And the client can basically
directly communicate with the host using NAPT.

Thank you for the quick replies!

kapilkedawat, 2024-04-17:
Hi @SmartAppsML, thanks for the info.. did you get a chance to look at https://github.com/espressif/esp-hosted? It should fulfill most of your requirements(works on SPI/SDIO interface). 

Since esp chips natively doesn't support Ethernet, adding support for external AAA server doesn't make much sense and internal server will cause memory crunch issues(We wouldn't want to put a half baked feature in WiFi stack).

ESP-Hosted was written for these kinda usecases only where we want to use esp as a wifi stick, please take a look.

closed 2024-10-18 by Sherry616