espressif/esp-idf #13530: No outbound bearer found spam receiving a vendor model message

ShahinHaque97-CT, 2024-04-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.2.1 release branch - a322e6b

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

No warning response

### What is the actual behavior?

No outbound bearer found, inbound bearer 0 Spam when receiving a vendor model message, message does receive successfully.

### Steps to reproduce.

The code is integrated into my project unfortunately and so I cannot share. if required after general debugging queries, I will create a new project to replicate this for you. 

In essence, Vendor Model Client and Server model examples. When receiving a vendor model message this message is spammed across the terminal  

### Debug Logs.

```plain
I (199718) LASCOR_MODEL_CALLBACK: Receive publish message 0xe20059

W (200018) BLE_MESH: No outbound bearer found, inbound bearer 0
W (200028) BLE_MESH: No outbound bearer found, inbound bearer 0
W (200178) BLE_MESH: No outbound bearer found, inbound bearer 0
W (200338) BLE_MESH: No outbound bearer found, inbound bearer 0
W (200408) BLE_MESH: No outbound bearer found, inbound bearer 0
W (200468) BLE_MESH: No outbound bearer found, inbound bearer 0
W (210018) BLE_MESH: No outbound bearer found, inbound bearer 0
W (210078) BLE_MESH: No outbound bearer found, inbound bearer 0
```

forx157, 2024-04-08:
@ShahinHaque97-CT 
Please enable the mesh log level to info level and then share the log of what went wrong.
`(Top) → Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL`

pranjalchanda08, 2025-01-02:
I also am facing the same issue v5.3.1 (ESP32-C3)

forx157, 2025-01-03:
Hi, @pranjalchanda08, @block-cha-cha 

When a node subscribes to address A and then publishes a message to address A, the network interface for that message includes the local interface. Messages from the local interface will not be forwarded, hence they cannot find an outbound bearer. Therefore, this warning can be temporarily disregarded. We will update the logic for printing this warning in the future to prevent it from appearing under normal conditions.