espressif/esp-idf #13531: CTE (Constant Tone Extension)-Feature on ESP32-C6, BLE 5.1

Robin24-5, 2024-04-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I haven't found a direct implementation of CTE (Constant Tone Extension) in the current libraries. 
Is the C6-SOC able to send BLE-Advertisements with this CTE-Field in the header?
As it's a BLE 5.1-feature, the C6 having BLE 5.3, should be able to do it, right?

zhaoweiliang2021, 2024-04-08:
The ESP32C6 doesn't support CTE feature, thanks

GeorgeDowning20, 2024-04-08:
> The ESP32C6 doesn't support CTE feature, thanks

how did it get the BLE 5.3 certification? CTE is part of the core specification

zhaoweiliang2021, 2024-04-09:
It can select relevant features for 5.3 certification. The 5.3 certification of ESP32C6 doesn't  include CTE feature.

Indastri, 2024-04-16:
Closing the issue as the issue is "caused by design". If you will have any further issues feel free to reopen this one or create a new issue.

Cheers!

closed 2024-04-16 by Indastri