espressif/esp-idf #13534: FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/blufi_init.c.obj

qianxiaojun1, 2024-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2815-gbe06a6f5ff

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build esp-idf/examples/bluetooth/blufi success

### What is the actual behavior?

Build faild
[1238/1245] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/blufi_init.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/blufi_init.c.obj 

### Steps to reproduce.

1. cd /esp-idf/examples/bluetooth/blufi
2. git pull and git submodule update --init --recusive
3. ctrl+shift+P and chose ESP-IDF:Set Espressif Deivce Target ->chose esp32s3
4. build failed

### Build or installation Logs.

```plain
[5/1245] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1500K,
*******************************************************************************
[404/1245] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/xiaojunqian/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/xiaojunqian/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/xiaojunqian/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /mnt/s1/esp_project/esp-idf/examples/bluetooth/blufi/sdkconfig
Compiler supported targets: xtensa-esp32s3-elf
[... 138 lines cut ...]
Bootloader binary size 0x51c0 bytes. 0x2e40 bytes (36%) free.
[1238/1245] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/blufi_init.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/blufi_init.c.obj 
/home/xiaojunqian/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.1.2-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/mnt/s1/esp_project/esp-idf/examples/bluetooth/blufi/build/config -I/mnt/s1/esp_project/esp-idf/examples/bluetooth/blufi/main -I/home/xiaojunqian/esp/esp-idf/components/newlib/platform_include -I/home/xiaojunqian/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/xiaojunqian/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/xiaojunqian/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/xiaojunqian/esp/esp-idf/components/freertos/esp_additions/include -I/home/xiaojunqian/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/home/xiaojunqian/esp/esp-idf/components/esp_hw_support/include -I/home/xiaojunqian/esp/esp-idf/components/esp_hw_support/include/soc -I/home/xiaojunqian/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/xiaojunqian/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/xiaojunqian/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/xiaojunqian/esp/esp-idf/components/heap/include -I/home/xiaojunqian/esp/esp-idf/components/log/include -I/home/xiaojunqian/esp/esp-idf/components/soc/include -I/home/xiaojunqian/esp/esp-idf/components/soc/esp32s3 -I/home/xiaojunqian/esp/esp-idf/components/soc/esp32s3/include -I/home/xiaojunqian/esp/esp-idf/components/hal/esp32s3/include -I/home/xiaojunqian/esp/esp-idf/components/hal/include -I/home/xiaojunqian/esp/esp-idf/components/hal/platform_port/include -I/home/xiaojunqian/esp/esp-idf/components/esp_rom/include -I/home/xiaojunqian/esp/esp-idf/components/esp_rom/include/esp32s3 -I/home/xiaojunqian/esp/esp-idf/components/esp_rom/esp32s3 -I/home/xiaojunqian/esp/esp-idf/components/esp_common/include -I/home/xiaojunqian/esp/esp-idf/components/esp_system/include -I/home/xiaojunqian/esp/esp-idf/components/esp_system/port/soc -I/home/xiaojunqian/esp/esp-idf/components/esp_system/port/include/private -I/home/xiaojunqian/esp/esp-idf/components/xtensa/include -I/home/xiaojunqian/esp/esp-idf/components/xtensa/esp32s3/include -I/home/xiaojunqian/esp/esp-idf/components/lwip/include -I/home/xiaojunqian/esp/esp-idf/components/lwip/include/apps -I/home/xiaojunqian/esp/esp-idf/components/lwip/include/apps/sntp -I/home/xiaojunqian/esp/esp-idf/components/lwip/lwip/src/include -I/home/xiaojunqian/esp/esp-idf/components/lwip/port/include -I/home/xiaojunqian/esp/esp-idf/components/lwip/port/freertos/include -I/home/xiaojunqian/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/xiaojunqian/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/xiaojunqian/esp/esp-idf/components/esp_ringbuf/include -I/home/xiaojunqian/esp/esp-idf/components/efuse/include -I/home/xiaojunqian/esp/esp-idf/components/efuse/esp32s3/include -I/home/xiaojunqian/esp/esp-idf/components/driver/include -I/home/xiaojunqian/esp/esp-idf/components/driver/deprecated -I/home/xiaojunqian/esp/esp-idf/components/driver/analog_comparator/include -I/home/xiaojunqian/esp/esp-idf/components/driver/dac/include -I/home/xiaojunqian/esp/esp-idf/components/driver/gpio/include -I/home/xiaojunqian/esp/esp-idf/components/driver/gptimer/include -I/home/xiaojunqian/esp/esp-idf/components/driver/i2c/include -I/home/xiaojunqian/esp/esp-idf/components/driver/i2s/include -I/home/xiaojunqian/esp/esp-idf/components/driver/ledc/include -I/home/xiaojunqian/esp/esp-idf/components/driver/mcpwm/include -I/home/xiaojunqian/esp/esp-idf/components/driver/parlio/include -I/home/xiaojunqian/esp/esp-idf/components/driver/pcnt/include -I/home/xiaojunqian/esp/esp-idf/components/driver/rmt/include -I/home/xiaojunqian/esp/esp-idf/components/driver/sdio_slave/include -I/home/xiaojunqian/esp/esp-idf/components/driver/sdmmc/include -I/home/xiaojunqian/esp/esp-idf/components/driver/sigma_delta/include -I/home/xiaojunqian/esp/esp-idf/components/driver/spi/include -I/home/xiaojunqian/esp/esp-idf/components/driver/temperature_sensor/include -I/home/xiaojunqian/esp/esp-idf/components/driver/touch_sensor/include -I/home/xiaojunqian/esp/esp-idf/components/driver/twai/include -I/home/xiaojunqian/esp/esp-idf/components/driver/uart/include -I/home/xiaojunqian/esp/esp-idf/components/driver/usb_serial_jtag/include -I/home/xiaojunqian/esp/esp-idf/components/driver/touch_sensor/esp32s3/include -I/home/xiaojunqian/esp/esp-idf/components/esp_pm/include -I/home/xiaojunqian/esp/esp-idf/components/mbedtls/port/include -I/home/xiaojunqian/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/xiaojunqian/esp/esp-idf/components/mbedtls/mbedtls/library -I/home/xiaojunqian/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/xiaojunqian/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/home/xiaojunqian/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/home/xiaojunqian/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/home/xiaojunqian/esp/esp-idf/components/esp_app_format/include -I/home/xiaojunqian/esp/esp-idf/components/bootloader_support/include -I/home/xiaojunqian/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/xiaojunqian/esp/esp-idf/components/esp_partition/include -I/home/xiaojunqian/esp/esp-idf/components/app_update/include -I/home/xiaojunqian/e [... cut ...]
/mnt/s1/esp_project/esp-idf/examples/bluetooth/blufi/main/blufi_init.c:36:11: error: implicit declaration of function 'esp_bluedroid_init_with_cfg'; did you mean 'esp_bluedroid_init'? [-Werror=implicit-function-declaration]
   36 |     ret = esp_bluedroid_init_with_cfg(&bluedroid_cfg);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |           esp_bluedroid_init
cc1: some warnings being treated as errors
[1239/1245] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/blufi_example_main.c.obj
ninja: build stopped: subcommand failed.
```

### More Information.

I can build example blink project success, but this project failed.

qianxiaojun1, 2024-04-09:
and the branch of project :
xiaojunqian@xiaojun:/mnt/s1/esp_project/esp-idf/examples/bluetooth/blufi$ git branch
* master
xiaojunqian@xiaojun:/mnt/s1/esp_project/esp-idf/examples/bluetooth/blufi$ git submodule
 24c60e243580c7868f4334a1ba3123481fe1aa48 ../../../components/bootloader/subproject/components/micro-ecc/micro-ecc (v1.1)
 79c3d50f699efee7c4a1b80604e96fd2a355311b ../../../components/bt/controller/lib_esp32 (heads/master-5-g79c3d50)
 cb051020d238fb6e2d1e5f4f23a9678912a1fe28 ../../../components/bt/controller/lib_esp32c2/esp32c2-bt-lib (heads/main-1-gcb05102)
 e5c0f7256ecf5b5f8eb28c1793051a6b88f95124 ../../../components/bt/controller/lib_esp32c3_family (heads/master)
 9c0690805642cd9653e4c9acff31de59a670b225 ../../../components/bt/controller/lib_esp32c6/esp32c6-bt-lib (heads/main-2-g9c06908)
 18a00c3fe8e35cdfef280c9465692ebc74e4e1ca ../../../components/bt/controller/lib_esp32h2/esp32h2-bt-lib (heads/master-2-g18a00c3)
 41bf5fc0926fd6d3fb39cb5107e97f2fc6aed7e5 ../../../components/bt/esp_ble_mesh/lib/lib (heads/main)
 3705a21d499cc9b8f584e8468ff7e82cc5452a4c ../../../components/bt/host/nimble/nimble (remotes/origin/master-1574-g3705a21d)
 eeecc49ce8af123cf8ad40efdb9673e37b56230f ../../../components/cmock/CMock (v2.5.2-2-geeecc49)
 2717868407a01878a49d19794538b6da120597ae ../../../components/esp_coex/lib (remotes/origin/HEAD)
 2d319a382336cf0522ea4bb5a3fbd6701a8633c6 ../../../components/esp_phy/lib (heads/master)
 854c76c14a89c7835ed7cf555c027de4b8442067 ../../../components/esp_wifi/lib (heads/master-9-g854c76c1)
 d2e28f872472ffc6a704faae65ddee1f24e2dfba ../../../components/heap/tlsf (remotes/origin/master-25-gd2e28f8)
 87d8f0961a01bf09bef98ff89bae9fdec42181ee ../../../components/json/cJSON (v1.7.15-21-g87d8f09)
 f79221431fa9042b3572d271d687de66da7560c4 ../../../components/lwip/lwip (remotes/origin/2.1.3-esp)
 09bba150d0d822aad2e58d71723f5407da5c21e0 ../../../components/mbedtls/mbedtls (mbedtls-3.5.2-7-g09bba150d)
 e6afdb4025fe018ae0add44e3c45249ea1974774 ../../../components/mqtt/esp-mqtt (ESP-MQTT_FOR_IDF_4.x-227-ge6afdb4)
 fcee30db4b7342de4df9105bdd049a09d2d63187 ../../../components/openthread/lib (fcee30d)
 456c448284486abe2a9118a9fdaa468fe2383fcd ../../../components/openthread/openthread (thread-reference-20230706-340-g456c44828)
 abc67a11c6db271bedbb9f58be85d6f4e2ea8389 ../../../components/protobuf-c/protobuf-c (v1.4.1)
 0dbb3f71c5f6fae3747a9d935372773762baf852 ../../../components/spiffs/spiffs (0.2-255-g0dbb3f7)
 bf560290f6020737eafaa8b5cbd2177c3956c03f ../../../components/unity/unity (v2.6.0_RC1)

zhp0406, 2024-04-09:
zhanghaipeng@FA002426:~/esp/esp-idf$ **git checkout be06a6f5ff**
M       components/bt/controller/lib_esp32
M       components/bt/controller/lib_esp32c2/esp32c2-bt-lib
M       components/bt/controller/lib_esp32c3_family
M       components/bt/controller/lib_esp32c6/esp32c6-bt-lib
M       components/bt/controller/lib_esp32h2/esp32h2-bt-lib
M       components/bt/host/nimble/nimble
M       components/esp_phy/lib
M       components/esp_wifi/lib
M       components/heap/tlsf
M       components/openthread/lib
Note: switching to 'be06a6f5ff'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at be06a6f5ff Merge branch 'bugfix/fix_softap_coex_start_issue' into 'master'
zhanghaipeng@FA002426:~/esp/esp-idf$ **git submodule update** 
Submodule path 'components/bt/controller/lib_esp32': checked out '79c3d50f699efee7c4a1b80604e96fd2a355311b'
Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out 'cb051020d238fb6e2d1e5f4f23a9678912a1fe28'
Submodule path 'components/bt/controller/lib_esp32c3_family': checked out 'e5c0f7256ecf5b5f8eb28c1793051a6b88f95124'
Submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib': checked out '9c0690805642cd9653e4c9acff31de59a670b225'
Submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib': checked out '18a00c3fe8e35cdfef280c9465692ebc74e4e1ca'
Submodule path 'components/bt/host/nimble/nimble': checked out '3705a21d499cc9b8f584e8468ff7e82cc5452a4c'
Submodule path 'components/esp_phy/lib': checked out '2d319a382336cf0522ea4bb5a3fbd6701a8633c6'
Submodule path 'components/esp_wifi/lib': checked out '854c76c14a89c7835ed7cf555c027de4b8442067'
Submodule path 'components/heap/tlsf': checked out 'd2e28f872472ffc6a704faae65ddee1f24e2dfba'
zhanghaipeng@FA002426:~/esp/esp-idf$ **./install.sh** 
Detecting the Python interpreter
Checking "python3" ...
Python 3.11.4
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Updating /home/zhanghaipeng/.espressif/idf-env.json
Selected targets are: esp32c61, esp32c5, esp32s2, esp32c3, esp32, esp32c6, esp32p4, esp32s3, esp32c2, esp32h2
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@12.1_20231023 (already installed)
Skipping riscv32-esp-elf-gdb@12.1_20231023 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20240305 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20240305 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20230921 (already installed)
Skipping esp-rom-elfs@20240305 (already installed)
Installing Python environment and packages
Python 3.11.4
pip 24.0 from /home/zhanghaipeng/.espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
WARNING: Error while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: '/home/zhanghaipeng/.espressif/python_env/idf5.3_py3.11_env/idf_version.txt'
Upgrading pip and setuptools...
Looking in indexes: https://dl.espressif.com/pypi, https://pypi.org/simple
Requirement already satisfied: pip in /home/zhanghaipeng/.espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (24.0)
Requirement already satisfied: setuptools in /home/zhanghaipeng/.espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (69.2.0)
Skipping the download of /home/zhanghaipeng/.espressif/espidf.constraints.v5.3.txt because it was downloaded recently.
Installing Python packages
 Constraint file: /home/zhanghaipeng/.espressif/espidf.constraints.v5.3.txt
 Requirement files:
  - /home/zhanghaipeng/esp/esp-idf/tools/requirements/requirements.docs.txt
[... 271 lines cut ...]
Generated /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/build/blufi_demo.bin
[1297/1297] cd /home/zhanghaipeng/esp/esp-idf/examples/bluetoo...peng/esp/esp-idf/examples/bluetooth/blufi/build/blufi_demo.bin
blufi_demo.bin binary size 0x10fb40 bytes. Smallest app partition is 0x177000 bytes. 0x674c0 bytes (28%) free.

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blufi_demo.bin
or from the "/home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/build" directory
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
zhanghaipeng@FA002426:~/esp/esp-idf/examples/bluetooth/blufi$ gi tlog
gi: command not found
zhanghaipeng@FA002426:~/esp/esp-idf/examples/bluetooth/blufi$ git log
commit be06a6f5ffe36f9554cfc91fe2036e0fc85fea60 (HEAD)
Merge: 0e3673a2bd 81b86f657c
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Thu Mar 21 21:22:54 2024 +0800

    Merge branch 'bugfix/fix_softap_coex_start_issue' into 'master'
    
    fix(wifi): fix softap coex start issue, no wifi request if wifi is not used at idle
    
    See merge request espressif/esp-idf!29772

zhp0406, 2024-04-09:
@qianxiaojun1 
It seems like your environment hasn't been properly installed.
```
~/esp/esp-idf/examples/bluetooth/blufi$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/build
Executing "ninja all"...
[15/1297] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1500K,
*******************************************************************************
[490/1297] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/asio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/zhanghaipeng/esp/esp-idf/components/bootloader /home/zhanghaipeng/esp/esp-idf/components/bootloader_support /home/zhanghaipeng/esp/esp-idf/components/efuse /home/zhanghaipeng/esp/esp-idf/components/esp_app_format /home/zhanghaipeng/esp/esp-idf/components/esp_bootloader_format /home/zhanghaipeng/esp/esp-idf/components/esp_common /home/zhanghaipeng/esp/esp-idf/components/esp_hw_support /home/zhanghaipeng/esp/esp-idf/components/esp_rom /home/zhanghaipeng/esp/esp-idf/components/esp_system /home/zhanghaipeng/esp/esp-idf/components/esptool_py /home/zhanghaipeng/esp/esp-idf/components/freertos /home/zhanghaipeng/esp/esp-idf/components/hal /home/zhanghaipeng/esp/esp-idf/components/log /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/main /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /home/zhanghaipeng/esp/esp-idf/components/newlib /home/zhanghaipeng/esp/esp-idf/components/partition_table /home/zhanghaipeng/esp/esp-idf/components/soc /home/zhanghaipeng/esp/esp-idf/components/spi_flash /home/zhanghaipeng/esp/esp-idf/c [... cut ...]
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blufi_demo.bin
or from the "/home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/build" directory
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
```

zhp0406, 2024-04-09:
Please refer to the following document for installing the IDF environment:
[Please refer to: https://docs.espressif.com/projects/esp-idf/zh_CN/stable/esp32/get-started/linux-macos-setup.html](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/get-started/linux-macos-setup.html)

qianxiaojun1, 2024-04-09:
> Please refer to the following document for installing the IDF environment: [Please refer to: https://docs.espressif.com/projects/esp-idf/zh_CN/stable/esp32/get-started/linux-macos-setup.html](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/get-started/linux-macos-setup.html)

OK，I'll try to uninstall and retry。Maybe I use vscode extension to install and caused this problem。

qianxiaojun1, 2024-04-09:
> Please refer to the following document for installing the IDF environment: [Please refer to: https://docs.espressif.com/projects/esp-idf/zh_CN/stable/esp32/get-started/linux-macos-setup.html](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/get-started/linux-macos-setup.html)

Thank you! I have solve this problem。When I use vscode extesions,I find the environment will cause this problem。

closed 2024-04-09 by qianxiaojun1