espressif/esp-idf #13537: pppos example missing

hazad, 2024-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I pppos_client example is no longer available. Its mentioned in esp modem and other places in documentation. The link is also mentioned in some older tickets, but the link is not longer working. Has it been moved to some other path/repo?

https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client

hazad, 2024-04-03:
Found details of new link in esp modem component.

idf.py create-project-from-example "espressif/esp_modem^1.1.0:pppos_client"

closed 2024-04-03 by hazad

leeebo, 2024-04-03:
@hazad Thanks to the [ESP-IDF Component Manager](https://components.espressif.com/), many components now no longer need to be maintained in ESP-IDF. esp-modem now maintained in https://github.com/espressif/esp-protocols/tree/master/components.