espressif/esp-idf #13538: 5.2版本i2c发送函数返回值一直为0

crl6, 2024-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

产品从4.4升级到5.2发现，i2c_master_transmit函数不管硬件是否连接i2c外设返回都是0也就是成功。

初始化：
void i2c_app_main_init(uint8_t dev_addr, uint32_t freq, uint32_t timeout_ms, bool enable_pullup)
{
    esp_err_t ret = ESP_OK;
    g_i2c_dev_addr = dev_addr;
    g_i2c_dev_clk = freq;
    g_i2c_transmit_timeout = timeout_ms;
    g_i2c_master_config.flags.enable_internal_pullup = enable_pullup;

    ret += i2c_new_master_bus(&g_i2c_master_config, &g_i2c_bus_handle);

    i2c_device_config_t dev_cfg = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = g_i2c_dev_addr,
        .scl_speed_hz = g_i2c_dev_clk,
    };

    uint8_t read_buf[256] = {0x00};
    int detect_count = 0;
    uint8_t id_reg = 0x10;
    uint8_t read_id = 0x00;
    do_i2cdetect_cmd(&read_buf[0], &detect_count);
    ret += i2c_master_bus_add_device(g_i2c_bus_handle, &dev_cfg, &g_dev_handle);
    if(ret == ESP_OK) {
        ESP_LOGI(TAG, "i2c_app_main_init successfully");
    }else {
        ESP_LOGI(TAG, "i2c_app_main_init fail");
    }
    i2c_master_read_slave(dev_addr, &id_reg, 1, &read_id, 1);

}

写寄存器：
esp_err_t i2c_master_write_slave(uint8_t dev_addr, uint8_t* data_wr, size_t size)
{
    esp_err_t ret = ESP_OK;
    ret = i2c_master_transmit(g_dev_handle, data_wr, size, -1);
    return ret;// 不管硬件是否连接i2c外设返回都是0也就是成功。
}

mythbuster5, 2024-04-06:
这个已经在master修掉了，5.2会有一些延时

mythbuster5, 2024-04-06:
It's here 016877b7ce184e8e8e0cdb037c230a887bc8a5dc

closed 2024-04-06 by mythbuster5