espressif/esp-idf #13540: How to connect WiFi and Ethernet at same time using Netif?

pedrohugo-psc, 2024-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Espressif SoC revision.

ESP32-32D

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Devkit_v4

### Power Supply used.

USB

### What is the expected behavior?

I want to connect WiFi and Ethernet at same time using Netif.

### What is the actual behavior?

I am working utilizing WiFi for a mesh network and Ethernet for connecting the ESP32 to another device via ModBus TCP simultaneously. In the Modbus TCP, I am utilizing netif, and my WiFi solution also employs netif. When I try to connect both network, the system rebooting.

### Steps to reproduce.

1. Combining the ModBus TCP example code with code that utilizes WiFi Mesh.
2. Run the code.

### Debug Logs.

```plain
I (27) boot: ESP-IDF v4.4.4-dirty 2nd stage bootloader
I (27) boot: compile time 13:50:51
I (27) boot: chip revision: v3.0
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 1
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (81) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (88) boot:  2 app0             OTA app          00 10 00010000 00180000
I (96) boot:  3 app1             OTA app          00 11 00190000 00180000
I (103) boot:  4 spiffs           Unknown data     01 82 00310000 000e0000
I (111) boot:  5 coredump         Unknown data     01 03 003f0000 00010000
I (118) boot: End of partition table
I (122) boot_comm: chip revision: 3, min. application chip revision: 1
I (130) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=46c88h (289928) map
[... 340 lines cut ...]
D (5588) esp_core_dump_elf: Add stack for task 0x3ffafd24: addr 0x3ffbd640, sz 416
D (5595) esp_core_dump_elf: Add segment size=416, start_off=0x295c
D (5602) esp_core_dump_port: Task (TCB:3ffaf934) EXIT/PC/PS/A0/SP 40083004 4008628f 60430 800862a0 3ffaf810
D (5611) esp_core_dump_elf: Add TCB for task 0x3ffaf934: addr 0x3ffaf934, sz 356
D (5618) esp_core_dump_elf: Add segment size=356, start_off=0x2afc
D (5626) esp_core_dump_elf: Add stack for task 0x3ffaf934: addr 0x3ffaf750, sz 464
D (5632) esp_core_dump_elf: Add segment size=464, start_off=0x2c60
D (5640) esp_core_dump_port: Task (TCB:3ffbe7fc) EXIT/PC/PS/A0/SP 40083004 400862a0 60030 8008b6ae 3ffbe030
D (5648) esp_core_dump_elf: Add TCB for task 0x3ffbe7fc: addr 0x3ffbe7fc, sz 356
D (5656) esp_core_dump_elf: Add segment size=356, start_off=0x2e30
D (5663) esp_core_dump_elf: Add stack for task 0x3ffbe7fc: addr 0x3ffbdf70, sz 2176
D (5670) esp_core_dump_elf: Add segment size=2176, start_off=0x2f94
D (5681) esp_core_dump_elf: Found 0 bad task out of 8
D (5681) esp_core_dump_elf: ================ Processing coredump info ================
D (5689) esp_core_dump_elf: Application SHA256='3f7e2bd03208e9c6', length=17.
D (5697) esp_core_dump_elf: Add note size=104, start_off=0x3814
D (5703) esp_core_dump_elf: Add note size=172, start_off=0x3814
D (5708) esp_core_dump_elf: Add NOTE segment, size=276, start_off=0x3814
D (5715) esp_core_dump_elf: =========== Data written size = 14652 bytes ==========
D (5723) esp_core_dump_checksum: Total length of hashed data: 14656
I (5729) esp_core_dump_flash: Write end offset 0x3944, check sum length 4
I (5737) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jul 29 2019 12:21:46
```

pedrohugo-psc, 2024-04-08:
> There is an official example here. https://github.com/espressif/esp-idf/tree/release/v5.1/examples/ethernet/eth2ap
> 
> This example uses WiFi and Ethernet at the same time. It works correctly.

Hello, nopnop2002,

I already verify this example. However, this example don´t use netif to configure WiFi and Ethernet.

nopnop2002, 2024-04-08:
>However, this example don´t use netif to configure WiFi and Ethernet.

It's definitely what you say.

espressif-abhikroy, 2024-04-09:
@pedrohugo-psc Can you please provide a simple example code to reproduce this issue?

pedrohugo-psc, 2024-05-21:
> @pedrohugo-psc Can you please provide a simple example code to reproduce this issue?

Hello, I found the solution to this issue. I simply commented this function esp_event_loop_create_default().

closed 2024-05-21 by pedrohugo-psc

KaeLL, 2024-05-21:
#8745 This is a silly issue.