espressif/esp-idf #13544: FREERTOS_WATCHPOINT_END_OF_STACK severely slow down the ISR response on C3 and C6 with debugger attached.

andylinpersonal, 2024-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 and release/v5.2

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.3) and ESP32-C6 revision 0.0

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

C3: Custom board, C6: ESP32-C6-DevkitC-N8

### Power Supply used.

USB

### What is the expected behavior?

System should not be clogged even with the debug probe attached and enabled on it.

### What is the actual behavior?

When the system is busy (simulated by two busy printing tasks) and compiled with FREERTOS_WATCHPOINT_END_OF_STACK=y, system will become very slow and the serving of ISRs are severely deferred (use systick as demo). 
BTW, this issue also affect the interrupt-intensive tasks. I have been successfully reproduced on the I2C and I2S over GDMA.

### Steps to reproduce.

1. Enable `FREERTOS_WATCHPOINT_END_OF_STACK=y` in the `sdkconfig.defaults`.
2. Apply the snippets to the systick handler.
3. Build and flash
4. Run `openocd -f 'board/$CHIP-builtin.cfg'`.
5. Run `idf.py monitor -p $PORT --no-reset`.
6. See the console output with and without `FREERTOS_WATCHPOINT_END_OF_STACK`.
   1. Systick has been severely delayed for about 407x. It's an incredible ratio.

### Debug Logs.

```plain
C6 with debugger attached

ESP-ROM:esp32c6-20220919
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x18 (JTAG_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x20000830
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x19b0
load:0x4086c410,len:0xd54
load:0x4086e610,len:0x2e38
entry 0x4086c41a
[0;32m22       boot            I: ESP-IDF v5.2-dev-5051-g9010d945e1-dirty 2nd stage bootloader[0m
[... 353 lines cut ...]
    if (start_measurement) {
        systimer_hal_context_t tmrhal = {.dev = &SYSTIMER};
        if (intr_counter == 0) {
            if (systimer_ll_is_counter_value_valid(&SYSTIMER, SYSTIMER_COUNTER_ESPTIMER)) {
                systick_start = systimer_hal_get_counter_value(&tmrhal, SYSTIMER_COUNTER_ESPTIMER);
                intr_counter++;
            }
        } else if (intr_counter < max_intr_counter) {
            intr_counter++;
        } else {
            start_measurement = false;
            systick_end = systimer_hal_get_counter_value(&tmrhal, SYSTIMER_COUNTER_ESPTIMER);
        }
    }
    // omit
}

```

sdkconfig.defaults
```
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_FREERTOS_HZ=1000
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y # Toggle this one
```

KonstantinKondrashov, 2024-04-09:
Hi @andylinpersonal!
I have tried to reproduce it but still no luck. I am getting 1ms for both w and w/o FREERTOS_WATCHPOINT_END_OF_STACK + w and w/o the debug probe attached. I used the C6 chip with your example.
Does it happen on the master as well?

Your result:
0x7D000 = 512000 ticks,  
512000 / 32 (max_intr_counter = 32) / 16 (systimer division) = 1000us. Which is equal to the used CONFIG_FREERTOS_HZ=1000.

(It seems this is without FREERTOS_WATCHPOINT_END_OF_STACK)
0x00000000095B2A0B = 156969483 ticks
156969483 / 16384 (max_intr_counter = 16384) / 16 (systimer division) = 598us. Which is even less than 1ms. Could you explain this case? Is it for 16384  or not?

KonstantinKondrashov, 2024-04-09:
@andylinpersonal
It seems I have reproduced this case. I slightly changed your example.
This example works well when CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=n or debug is not attached.
The issue occurs when CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y and the debug is attached. I am getting the 32 Systick interrupts not within 32ms as expected (1 tick is 1ms, CONFIG_FREERTOS_HZ=1000) but within 10sec. Which is 312 times longer.
I will check why it works like that.

andylinpersonal, 2024-04-09:
BTW, attaching the OpenOCD to the running ESP32-C3 and ESP32-C6 will occasionally trigger infinite unhandled debug exceptions, which cannot be eliminated without a power cycle. 
It's strange. 😅