espressif/esp-idf #13545: MODBUS broadcast request (slave ID: 0) leaves system in illegal state with the error "receive buffer initialization fail."

BestBot7, 2024-04-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tried to send the modbus broadcast request i.e; slave id = 0, using the sample code of esp-idf modbus.  the master sends the request and enters into an illegal state as shown in the figure and does not allow any further requests until reboot. As per the MODBUS protocol standard, the slave only receives request and processes it but doesn't respond since it is the broadcast request. But the freemodbus is making the master enter into illegal state. What is the exact problem for this error of modbus master entering into an unknown state? I suspect that the master is checking the port for any packet received from slave end. Since there is no packet on the port, it gives this error **"receive buffer initialization fail."** The log images of both modbus master and slave are attached for reference. Let me know the issue and ways to fix it.

MODBUS Master logs with broadcast request:
[image]

MODBUS Slave logs:
[image]

Thanks in advance!

alisitsyn, 2024-04-08:
@BestBot7, Thanks for reporting. This error will be verified and I will give you solution as soon as possible.

BestBot7, 2024-04-08:
@alisitsyn, Thanks for the reply! waiting for the solution.

alisitsyn, 2024-04-12:
@BestBot7 , 

Sorry for the delay. I got some time to check this issue. The issue happens after some update of FSM in the master and confirmed as a bug. Please follow the steps below to fix this:

1. Copy the latest version of the esp-modbus stack into components folder in the root of your project folder. After this you will be able to apply changes.
2. Find the file `/root_of_proj/esp-modbus/freemodbus/modbus/mb_m.c` and make the changes as per diff below:
```
diff --git a/freemodbus/modbus/mb_m.c b/freemodbus/modbus/mb_m.c
index 24c51d6..f0b1402 100644
--- a/freemodbus/modbus/mb_m.c
+++ b/freemodbus/modbus/mb_m.c
@@ -391,6 +391,9 @@ eMBMasterPoll( void )
                 break;
             case EV_MASTER_EXECUTE:
                 if (xCurTransactionId == xEvent.xTransactionId) {
+                    if ( xMBMasterRequestIsBroadcast()    // <<<<<<<<<<<<<<<<< changes to apply
+                         && (( ucMBMasterGetCommMode() == MB_RTU ) || ( ucMBMasterGetCommMode() == MB_ASCII ) ) ) {
+                        ucMBRcvFrame = ucMBSendFrame;
+                    }
                     MB_PORT_CHECK(ucMBRcvFrame, MB_EILLSTATE, "receive buffer initialization fail.");
                     ESP_LOGD(MB_PORT_TAG, "%" PRIu64 ":EV_MASTER_EXECUTE", xEvent.xTransactionId);
                     ucFunctionCode = ucMBRcvFrame[MB_PDU_FUNC_OFF];
```
The diff below allows to report fail by code for broadcast read request.
```
diff --git a/freemodbus/serial_master/modbus_controller/mbc_serial_master.c b/freemodbus/serial_master/modbus_controller/mbc_serial_master.c
index f8769a2..0efde7d 100644
--- a/freemodbus/serial_master/modbus_controller/mbc_serial_master.c
+++ b/freemodbus/serial_master/modbus_controller/mbc_serial_master.c
@@ -358,6 +358,10 @@ static esp_err_t mbc_serial_master_set_request(char* name, mb_param_mode_t mode,
         // Compare the name of parameter with parameter key from table
         int comp_result = memcmp((const void*)name, (const void*)reg_ptr->param_key, (size_t)param_key_len);
         if (comp_result == 0) {
+            if (!reg_ptr->mb_slave_addr && (mode == MB_PARAM_READ)) {
+                 error = ESP_ERR_INVALID_ARG;
+                 break;
+            }
             // The correct line is found in the table and reg_ptr points to the found parameter description
             request->slave_addr = reg_ptr->mb_slave_addr;
             request->reg_start = reg_ptr->mb_reg_start;
```
3. Rebuild the project using `idf.py fullclean; idf.py build". 

This should solve the issue while sending the broadcast request.

Please let me know if you have any issues applying these patches.

BestBot7, 2024-04-15:
Thanks for the reply @alisitsyn! We will use the fix, verify and update the results.

alisitsyn, 2024-05-02:
@BestBot7 , the fix was merged with commit 20c34b20