espressif/esp-idf #13554: esp32c2 need #define SOC_SPI_HD_BOTH_INOUT_SUPPORTED 1，to use SPI Hafl3Wire

xiaguangbo, 2024-04-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.3

### Espressif SoC revision.

esp32c2 v1.0

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

nodemcu-esp32-c2

### Power Supply used.

USB

### What is the expected behavior?

use SPI Hafl3Wire

### What is the actual behavior?

if not fix, can report some err
<https://github.com/esp-rs/esp-idf-hal/issues/401>

### Steps to reproduce.

<https://github.com/esp-rs/esp-idf-hal/issues/401>

### Debug Logs.

```plain
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (32) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (32) boot: compile time Jan 16 2024 14:05:06
I (32) boot: chip revision: v1.0
I (37) boot.esp32c2: MMU Page Size  : 32K
I (44) boot.esp32c2: SPI Speed      : 30MHz
I (51) boot.esp32c2: SPI Mode       : DIO
I (59) boot.esp32c2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
I (91) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[... 199 lines cut ...]
3fcbb470: 0x00000014 0xf11050e4 0x4bd945a5 0x3fcbb458 0x00000000 0x00000005 0x3fcb9454 0x72687470

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382484
0x40382484 - esp_restart_noos
    at /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/v5.1.3/components/esp_system/port/soc/esp32c2/system_internal.c:96
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (36) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (36) boot: compile time Jan 16 2024 14:05:06
```
```

### More Information.

`Hafl3wire` can be used using the modification method in the problem link. If full duplex is used and `miso` and `mosi` are connected together to imitate the behavior of `Hafl3wire`, errors will occur.

wanckl, 2024-04-08:
@xiaguangbo Hi~
`SOC_SPI_HD_BOTH_INOUT_SUPPORTED` is really not supported for chips after esp32s2, it not a BUG but HW design changes.   
But is not means these chips don't support half duplex, as error log reported:   
> SPI half duplex mode is not supported when both MOSI and MISO phases are enabled.   

It just a small changes that you need split tx and rx to **independent** transactions, ensure one transaction have only one direction. I don't know how rust implement 3-wire HD mode, but for esp-idf, can refer to test `test_spi_sio.c`.    

For the modification mentioned in https://github.com/esp-rs/esp-idf-hal/issues/401, it just work by accident, if txleng != rxleng, it not work as design.

xiaguangbo, 2024-04-08:
@wanckl
I don't know what that means `both MOSI and MISO phases are enabled`. `phases enabled`means`CPHA = 1`? or tx and rx are enabled?

I don't understand this code.
.embuild/espressif/esp-idf/v5.1.3/components/driver/spi/gpspi/spi_master.c L790:
```
#if !SOC_SPI_HD_BOTH_INOUT_SUPPORTED
    //On these chips, HW doesn't support using both TX and RX phases when in halfduplex mode
    SPI_CHECK(!is_half_duplex || !tx_enabled || !rx_enabled, "SPI half duplex mode is not supported when both MOSI and MISO phases are enabled.", ESP_ERR_INVALID_ARG);
    SPI_CHECK(!is_half_duplex || !trans_desc->length || !trans_desc->rxlength, "SPI half duplex mode is not supported when both MOSI and MISO phases are enabled.", ESP_ERR_INVALID_ARG);
#endif
```

vscode prompt:
```
#define SPI_CHECK(a,str,ret_val) ESP_RETURN_ON_FALSE_ISR(a, ret_val, SPI_TAG, str)
扩展到:

do { if ((!(!is_half_duplex || !tx_enabled || !rx_enabled))) { ESP_EARLY_LOGE(SPI_TAG, "%s(%d): " "SPI half duplex mode is not supported when both MOSI and MISO phases are enabled.", __FUNCTION__, 792); return 0x102; } } while(0)
```
```
#define SPI_CHECK(a,str,ret_val) ESP_RETURN_ON_FALSE_ISR(a, ret_val, SPI_TAG, str)
扩展到:

do { if ((!(!is_half_duplex || !trans_desc->length || !trans_desc->rxlength))) { ESP_EARLY_LOGE(SPI_TAG, "%s(%d): " "SPI half duplex mode is not supported when both MOSI and MISO phases are enabled.", __FUNCTION__, 793); return 0x102; } } while(0)
```

wanckl, 2024-04-09:
@xiaguangbo ,,Sorry for confusing you,,

The code block you mentioned is a check, it is a hardware limitation    
`phases enabled` not means CPHA    

half duplex means only one direction at a moment, then
> both MOSI and MISO phases are enabled

This limitation is: in one transaction constructed in `spi_transaction_t`, can only contain tx OR rx, if you need one TX followed by one RX, you need construct 2 `spi_transaction_t` then start them independently

xiaguangbo, 2024-04-09:
@wanckl 
Problem is found ! It's a library function parameter problem.
/home/xiaguangbo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.43.1/src/spi.rs L1610:
```
fn spi_read_transactions(
    words: &mut [u8],
    chunk_size: usize,
) -> impl Iterator<Item = spi_transaction_t> + '_ {
    words.chunks_mut(chunk_size).map(|chunk| {
        spi_create_transaction(
            chunk.as_mut_ptr(),
            core::ptr::null(),
            chunk.len(), // problem is here. changed to 0
            chunk.len(),
        )
    })
}

...

fn spi_create_transaction(
    read: *mut u8,
    write: *const u8,
    transaction_length: usize,
    rx_length: usize,
) -> spi_transaction_t {
    spi_transaction_t {
        flags: 0,
        __bindgen_anon_1: spi_transaction_t__bindgen_ty_1 {
            tx_buffer: write as *const _,
        },
        __bindgen_anon_2: spi_transaction_t__bindgen_ty_2 {
            rx_buffer: read as *mut _,
        },
        length: (transaction_length * 8) as _,
        rxlength: (rx_length * 8) as _,
        ..Default::default()
    }
}
```

closed 2024-04-09 by xiaguangbo