espressif/esp-idf #13558: Esp32-s3 master write share buf

Sotung96, 2024-04-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all, I got a problem. Esp keep reboot while I'm trying to use `essl_spi_wrbuf` in code. Here is my way for using the command
`int Period = 0x2;`
`ESP_ERROR_CHECK(essl_spi_wrbuf(spi, (uint8_t *)Period, 10, sizeof(Period), SPI_TRANS_MODE_QIO | SPI_TRANS_MULTILINE_ADDR));`
and the system comes out `Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.`
Do anyone can help me? Thanks

wanckl, 2024-04-08:
@Sotung96 en,,,

`Period` is a variable not a memory pointer, maybe you need add `&` to get it's address ?
```c
ESP_ERROR_CHECK(essl_spi_wrbuf(spi, (uint8_t *)&Period, 10, sizeof(Period), SPI_TRANS_MODE_QIO | SPI_TRANS_MULTILINE_ADDR));
```

wanckl, 2024-04-09:
:+1: If no problem please close this issue, and feel free to open it any time with new issue

closed 2024-04-09 by Sotung96