espressif/esp-idf #13561: build failed with newlib target

Himanshu-4, 2024-04-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

build successfully and created the app.elf file 

### What is the actual behavior?

build failed with link time failure. major errors given by linker is of below type
dangerous relocation: call8: call target out of range: _findenv_r 
major link error give by newlib library 

### Steps to reproduce.

1. Step
2. Step
3. Step
...
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/14904422/CMakeLists.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/14904433/sdkconfig.txt)

### Build or installation Logs.

```plain
`
-- Build files have been written to: D:/stimveda_codebase/V1_OTA/OTA/build
[47/47] Linking C executable src\app\app.elf
FAILED: src/app/app.elf
cmd.exe /C "cd . && C:\PROGRA~2\IDF-TO~1\tools\XTENSA~2\ESP-12~1.0_2\XTENSA~1\bin\XT8994~1.EXE  -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=D:/stimveda_codebase/V1_OTA/OTA/build/src/app/app.map -Wl,--no-warn-rwx-segments -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\app.rsp -o src\app\app.elf  && cd ."
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/bootloader_support/libbootloader_support.a(bootloader_flash_config_esp32.c.obj):(.text.bootloader_init_spi_flash+0x3c): undefined reference to `bootloader_image_hdr'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/crt0.o:(.literal+0x0): undefined reference to `main'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-findfp.o): in function `_cleanup':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:236:(.text+0xd5): dangerous relocation: call8: call target out of range: _cleanup_r
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-findfp.o): in function `__sinit':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:247:(.text+0x11f): dangerous relocation: call8: call target out of range: __sinit_lock_acquire
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:300:(.text+0x17e): dangerous relocation: call8: call target out of range: __sinit_lock_release
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-findfp.o): in function `__sfp':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:153:(.text+0x191): dangerous relocation: call8: call target out of range: __sfp_lock_acquire
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:156:(.text+0x19f): dangerous relocation: call8: call target out of range: __sinit
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:177:(.text+0x1c9): dangerous relocation: call8: call target out of range: __sfp_lock_release
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:163:(.text+0x213): dangerous relocation: call8: call target out of range: __sfmoreglue
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:166:(.text+0x21d): dangerous relocation: call8: call target out of range: __sfp_lock_release
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-findfp.o): in function `__fp_lock_all':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:355:(.text+0x233): dangerous relocation: call8: call target out of range: __sfp_lock_acquire
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-findfp.o): in function `__fp_unlock_all':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/findfp.c:365:(.text+0x25a): dangerous relocation: call8: call target out of range: __sfp_lock_release
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-itoa.o): in function `itoa':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/itoa.c:66:(.text+0x45): dangerous relocation: call8: call target out of range: __itoa
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-qsort.o): in function `qsort':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/search/qsort.c:329:(.text+0x322): dangerous relocation: call8: call target out of range: qsort
[... 31 lines cut ...]
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/bootloader_support/libbootloader_support.a(esp_image_format.c.obj): in function `process_segment':
D:/stimveda_codebase/V1_OTA/components/bootloader_support/src/esp_image_format.c:563: undefined reference to `bootloader_sha256_data'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/bootloader_support/libbootloader_support.a(esp_image_format.c.obj): in function `image_load':
D:/stimveda_codebase/V1_OTA/components/bootloader_support/src/esp_image_format.c:164: undefined reference to `bootloader_sha256_finish'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/stimveda_codebase/V1_OTA/components/bootloader_support/src/esp_image_format.c:168: undefined reference to `bootloader_sha256_finish'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/bootloader_support/libbootloader_support.a(esp_image_format.c.obj): in function `verify_simple_hash':
D:/stimveda_codebase/V1_OTA/components/bootloader_support/src/esp_image_format.c:928: undefined reference to `bootloader_sha256_finish'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/hal/libhal.a(spi_flash_hal.c.obj):(.text.get_flash_clock_divider+0xc): undefined reference to `ceil'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/hal/libhal.a(spi_flash_hal.c.obj):(.text.get_flash_clock_divider+0x10): undefined reference to `floor'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/hal/libhal.a(spi_flash_hal.c.obj): in function `get_flash_clock_divider':
D:/stimveda_codebase/V1_OTA/components/hal/spi_flash_hal.c:33: undefined reference to `ceil'
[... cut ...]
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/wbuf.c:93:(.text+0xbd): dangerous relocation: call8: call target out of range: __swbuf_r
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
PS D:\stimveda_codebase\V1_OTA\OTA\build>
`

i am getting this error (link time failure). don't know the reason. is there is a problem with the compilation flag or linker flag or is anything i am missing.
```

### More Information.

i am using my own cmake scripts , trying to mimic idf behavior. successful in bootloader build but failed in main application

KonstantinKondrashov, 2024-04-09:
Hi @Himanshu-4 ! 
Probably you might need to add `-mlongcalls` flag to your CMAKE_C_FLAGS and CMAKE_CXX_FLAGS. For reference https://github.com/espressif/esp-idf/blob/5a40bb8746633477c07ff9a3e90016c37fa0dc0c/tools/cmake/toolchain-esp32.cmake#L10-L21

Himanshu-4, 2024-04-09:
Hii @KonstantinKondrashov 
I actually did this , as shown below i am using this build flags and there are some other build flags 
`function(__build_set_default_build_flags)

    set(compile_definitions 
                    "ESP_PLATFORM"
                    # "IDF_VER=\\\"idf_ver_git\\\""
                    "SOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE"
                    "_GNU_SOURCE"
                    # @todo this should be set by build_get_idf_git_revision 
                    IDF_VER=\"idf_ver_git\"
                    )
                    
    if(NOT BOOTLOADER_BUILD)
        list(APPEND compile_definitions "_POSIX_READER_WRITER_LOCKS" )

    endif()
    set (compile_flags       
                                    # "-mtext-section-literals"
                                    # ignore multiple enum conversion warnings since gcc 11
                                    # TODO: IDF-5163
                                    "-Wno-enum-conversion"
                                    # Default is dwarf-5 since GCC 11, fallback to dwarf-4 because of binary size
                                    # TODO: IDF-5160
                                    "-gdwarf-4"
                                    # always generate debug symbols (even in release mode, these don't
                                    # go into the final binary so have no impact on size
                                    "-ggdb"
                                    "-mlongcalls"
                                    "-Wno-frame-address"
                                    "-fdiagnostics-color=always"
                                    "-ffunction-sections"
                                    "-fdata-sections"
                                     # warning-related flags
                                    "-Wall"
                                    "-Werror=all"
                                    "-Wextra"
                                    "-Wno-error=unused-function"
                                    "-Wno-error=unused-variable"
                                    "-Wno-error=unused-but-set-variable"
                                    "-Wno-error=deprecated-declarations"
                                    "-Wno-unused-parameter"
                                    "-Wno-sign-compare"
                                    "-Wno-enum-conversion"
                                    "-Wno-old-style-declaration"
                                    "-fstrict-volatile-bitfields"
                                    "-fno-jump-tables"
                                    "-fno-tree-switch-conversion"
                                
                                    )
    # set the compile definations and flags as common 
    # set(COMMON_OPTIONS "${compile_flags}" CACHE STRING "compile flags for the target")
    # set(COMMON_DEFINES "${compile_definitions}" CACHE STRING "common defines for the target")

    # # get the idf components path and set this property there 
    # get_property(build_comps_dir GLOBAL PROPERTY BUILD_COMPONENTS_PATH)
    # if(NOT EXISTS "${build_comps_dir}")
    #     message(FATAL_ERROR "Build component directory is missing and is required by the build process")
    # endif()

    
    # set the compiler flags for all the build compopnents in the directory z
    idf_build_set_property(COMPILE_DEFINITIONS "${compile_definitions}" APPEND)
    idf_build_set_property(COMPILE_OPTIONS "${compile_flags}" APPEND)
    idf_build_set_property(C_COMPILE_OPTIONS "${compile_flags}" APPEND)
    idf_build_set_property(CXX_COMPILE_OPTIONS "${compile_flags}" APPEND)
    idf_build_set_property(ASM_COMPILE_OPTIONS "${compile_flags}" APPEND)
    
endfunction()`

[... 222 lines cut ...]
    list(APPEND c_compile_options "${compile_options}")
    list(APPEND cxx_compile_options "${compile_options}")
    list(APPEND asm_compile_options "${compile_options}")
    
    idf_build_set_property(COMPILE_DEFINITIONS "${compile_definitions}" APPEND)

    idf_build_set_property(COMPILE_OPTIONS "${compile_options}" APPEND)
    idf_build_set_property(C_COMPILE_OPTIONS "${c_compile_options}" APPEND)
    idf_build_set_property(CXX_COMPILE_OPTIONS "${cxx_compile_options}" APPEND)
    idf_build_set_property(ASM_COMPILE_OPTIONS "${asm_compile_options}" APPEND)
    idf_build_set_property(LINK_OPTIONS "${link_options}" APPEND)

    message(STATUS "compile options ${compile_options} ")
    message(STATUS "c compile options ${c_compile_options}")
    message(STATUS "link options ${link_options}")
    message(STATUS "C Known features  ${CMAKE_C_COMPILE_FEATURES}")
    # message(STATUS "CXX known features ${CMAKE_CXX_COMPILE_FEATURES}")

endfunction()

# -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1

`

. if required i can share the whole cmake files that i have used. still the linker gives this error

KonstantinKondrashov, 2024-04-10:
Hi @Himanshu-4!
May I know why you do not use the standard way of building it (idf.py build)?

I am sorry, but we can not help with debugging your own script. We put an effort into making our build system suitable for many cases. Please give information about your use case. We would recommend you follow [the IDF build](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#build-the-project) if possible.

You can always have a look at our build system as a reference. 
Maybe somebody from our community can help you.
Thanks. (We'll leave it open for a while).

Himanshu-4, 2024-04-12:
Hii @KonstantinKondrashov  
the dangerous relocations is resolved newlib target. but i am getting this error now 
`
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/hal/libhal.a(spi_flash_hal.c.obj):(.literal.get_flash_clock_divider+0xc): undefined reference to `ceil'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/hal/libhal.a(spi_flash_hal.c.obj):(.literal.get_flash_clock_divider+0x10): undefined reference to `floor'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: components/hal/libhal.a(spi_flash_hal.c.obj): in function `get_flash_clock_divider':
D:/stimveda_codebase/V1_OTA/components/hal/spi_flash_hal.c:42: undefined reference to `ceil'
c:/progra~2/idf-to~1/tools/xtensa~2/esp-12~1.0_2/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/stimveda_codebase/V1_OTA/components/hal/spi_flash_hal.c:44: undefined reference to `floor'`

i want to have some custom functionality that are dependent on my scripts and to save build times. 
because idf build system build all the components and link it in final elf. also i have to generate some isolated component to flash into particular custom partitions. 
so that's why i have to develop my own scripts. it would be really helpful if you can provide some support.

Himanshu-4, 2024-04-12:
forget to mention that i already include the math library and gcc, c lib as 
idf_component_get_property(newlib newlib COMPONENT_LIB)
target_link_libraries(${COMPONENT_LIB} INTERFACE   m c gcc  "$<TARGET_FILE:${newlib}>")

KonstantinKondrashov, 2024-05-31:
Hi! I am closing this issue due to a lack of activity. Feel free to re-open if there is any new information. Thanks

closed 2024-05-31 by KonstantinKondrashov