espressif/esp-idf #13563: ESP-IDF V4.4.6 - Can't use core dump

FrancoBisciglia, 2024-04-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Greetings.

I'm trying to use the debug functionality of core dump in an ESP32 development board (ESP32 DevKit v1). I enabled the core dump "save in flash" option in the menuconfig, made a simple code that just run a while loop in main without blocking the task, so a reset will trigger, and then tried to read the core dump using the following command:

espcoredump.py -p COM4 dbg_corefile

For the record, I tried the same command but with the corresponding path to the firmware ".elf" that is generated when compiling the project.

But, I get the following error in the ESP-IDF terminal:

```
C:\Espressif\esp-idf\v4.4.6\esp-idf\components\espcoredump\espcoredump.py:2

# ESP-IDF Core Dump Utility
^

SyntaxError: Invalid or unexpected token
    at internalCompileFunction (node:internal/vm:73:18)
    at wrapSafe (node:internal/modules/cjs/loader:1185:20)
    at Module._compile (node:internal/modules/cjs/loader:1227:27)
    at Module._extensions..js (node:internal/modules/cjs/loader:1326:10)
    at Module.load (node:internal/modules/cjs/loader:1126:32)
    at Module._load (node:internal/modules/cjs/loader:967:12)
    at c._load (node:electron/js2c/node_init:2:13672)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:101:12)
    at node:internal/main/run_main_module:23:47

Node.js v18.18.2
```

Any ideas what does this mean? I used a USB to UART converter with CP210x, CH340 and PL2303 chips, because I read in a post that the problem could be realted to the USB to UART converter, but it seems that it isn't.

dobairoland, 2024-04-08:
Duplicate here: https://esp32.com/viewtopic.php?f=13&t=39241

FrancoBisciglia, 2024-04-08:
@dobairoland 

Greetings. As I answered in the other forum, yes, I posted the same topic in both forums, to maximize the chance of getting an answer.