espressif/esp-idf #13567: I2C and I2S Failing after Provisioning Process completes

monkeytronics, 2024-04-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

My application includes one I2S sensor and a number of I2C sensors. They all work fine in general. In my code, the provisioning process is initialised before any sensors are started as it is a process that runs infrequently. When it completes, the system is restarted. The sensors, you would assume should work as before.  Provisioning is initiated by pressing a button which calls a GPIO interrupt. This restarts the system - with a wakeup reason EXT1 (this is recognised as a provisioning event).

### What is the actual behavior?

However, I have found that if provisioning is run, after restart, the I2S and I2C buses no longer operate. I have to switch the power off and on again for it to start working again. It appears that there is some conflict between the resources in use during provisioning and those used by the I2C and I2S buses. The concerning part is that I have not been able to find any way to release these resources. I have tried WDT, esp_restart, deep sleep. However, the only way to cause the system to go back to normal is via power off / on again. Therefore, I can't concoct a code work-around.  

I have also tried a variety of things such as disabling the I2C and I2S buses before provisionig is called. Also I have tried every BT and WiFi deinitialisation function I can find. Nothing I have done seems to catch the conflict. 

### Steps to reproduce.

System is running normally. Sensors working...

Provisining
1. uer presses provisioing button - gpio int causes sleep
2. wakes up and sees ext1 reason. 
3. Starts provisioing process
4. provisioning completes 
5. device resets again.

Restart
1. device comes up again.
2. wifi connects etc. all good.
3. all I2C and I2S sensors are disconnected. The serial i2c bus is not operating. 
4. If I power the board off and on again, everything is fine.
5. It needs to work without manual intervention.

### Build or installation Logs.

```plain
Here is the log when user presses prov button:

I (127700) Monkey Button.c: Entering Pairing Mode
D (127700) i2s_common: i2s rx channel disabled
I (127700) Monkey Sleep.c: Enabling wakeup interrupt for GPIO 15
E (127700) Monkey Sleep.c: error on line 150, ret = 0
I (127710) Monkey Sleep.c: Going to sleep for 0
W (127710) Monkey Sleep.c: wifiMode = 1
I (127720) Monkey Sleep.c: Going to sleep...
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6008
load:0x40078000,len:15364
[... 123 lines cut ...]
I (54130) wifi:Deinit lldesc rx mblock:6
I (54140) monkey_wifi.c: WDT About to go...
I (62140) monkey_wifi.c: WDT did not work!
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6008
load:0x40078000,len:15364
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3804
entry 0x40080634
I (29) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (29) boot: compile time Apr  8 2024 17:53:10

After reboot, all sensors are unable to communicate on the serial buses.
```

### More Information.

I've recently put a lot of effort into reducing the memory usage of the project, and changed a lot of setting in menu config - reducing WiFi buffers and disabling IRAM optimaizations etc. I don't know if this is relevant but including in case it is.

SolidStateLEDLighting, 2024-04-09:
I'm assuming that you're referring to Wifi provisioning?    I'm trying to understand the significance of going to sleep for the provisioning.    I would think that this would be making your objective more difficult.   Can you explain to me why you're entering sleep and waking up again as it relates to the provisioning process?

For your reference -- I have practical experience with AWS IOT provisioning, BluFi Wifi provisioning (the old method), and Universal (BT and Wifi) provisioning (the new methods).    I have done all of these.    However, I have not used either I2C or I2S with any of these provisioning processes.   I do have plenty of I2C experience -- but no I2S experience yet.

monkeytronics, 2024-04-09:
Hi @SolidStateLEDLighting The use of sleep mode is just a way to restart the system such that the wakeup  reason can be parsed and used to determine the button press from the previous session. In the past, I have found that managing Bluetooth and other resources has been problematic. For that reason, I choose to restart the system before and after provisioning to draw a fence around that process and ensure that the resources used are released in normal operation. 

In any case, let's put provisioning to one side. It's not actually related directly the bug. It's just what I was doing at the time. A more concise description of the bug is that I2S and I2C seem not to work when the system wakes up from sleep due to a EXT1 GPIO interrupt. 

I have removed the EXT1 sleep and replaced it with a simple flag stored in NVS and everything now works again. So, I'm good to go, but it does seem to me that there is something happening that should not be.